espressif/esp-idf #10849: blecent crash after fail to allocate ACL buffers

embedded-sm, 2023-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version: v5.0.1
DevKit: ESP32-C3-DevKit

if bleprph send multiple notify to blecent, the blecent crash with error loop:
`NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT`

is possible to reproduce with the example:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/throughput_app

need to setup the example to use NOTIFY instead the READ

zztiger123, 2023-02-27:
Hi @embedded-sm 

Actually this is happening because 
Our sample applications are based on what upstream nimble applications provide and they are more of a demo purpose to show usage of APIs . Since we target embedded device , as a generic solution , we provide a memory configuration that should suffice majority of the use case .it looks like your case needs pushing packets at a rate faster than the speed at which packets are getting freed up . So, to suit your case, you can increase the buffer count such that they are sufficient enough to handle the scenario. 

However, plz note that increasing buffer counts will increase the memory footprint

embedded-sm, 2023-03-01:
unfortunately increasing the buffer count not solve the issue (only postpone a bit).
adding a delay between send on the peripheral can partially solve the problem, but is required big delay, that ruin the performance.
furthermore if the peripheral "flood" the ble central, the message should drop and is not acceptable that it can not recover from that status.
with the example is not easy to reproduce, but in our project when this issue happen it create guru meditation or watchdog trigger.
did you have opportunity to test it?

rahult-github, 2023-03-01:
Hi @embedded-sm  ,
 
  If putting delay also is not helping in your case, then another suggestion i have is that in case buffer is not allocated, instead of processing and going ahead, attempt to re-allocate packet. At some point of time, there will definitely be a freed packet, and hence allocation will happen. This is similar to introducing delay, as it is just trying to find memory , whenever applicable. 
  
In the throughput app, with default memory setting, "NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT" print is  visible. But there is no crash, since in sometime buffer are available and the test case goes ahead. 

In your setup, when above scenario is hit, what is the next processing step being done, that is leading to crash ?

embedded-sm, 2023-03-01:
> If putting delay also is not helping in your case, then another suggestion i have is that in case buffer is not allocated, instead of processing and going ahead, attempt to re-allocate packet. At [...]

your theory is right, but if you try the example you will notice that the behaviors is different
after the first "failed to allocate ACL buffers" if bleprph regular send message, the blecent not receive anymore messages and it never stop the error loop until bleprph stop for long time (>500ms)

rahult-github, 2023-03-01:
Hi @embedded-sm  ,

I tried the throughput_app example for notify.  I see below behaviour:

on blecent side:

> I (88280) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1345000
E (88290) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
I (88290) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1345500
E (88310) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT

and then, once buffers start to deplete, more such prints:

> E (88330) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (88340) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (88350) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (88350) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (88360) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (88380) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (88390) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT

However, on bleprph side:

Initially logs show 

> I (88865) NimBLE: GATT procedure initiated: notify; 
I (88865) NimBLE: att_handle=14
I (88865) NimBLE: GATT procedure initiated: notify; 
I (88875) NimBLE: att_handle=14
I (88875) NimBLE: GATT procedure initiated: notify; 
I (88885) NimBLE: att_handle=14
I (88885) NimBLE: GATT procedure initiated: notify; 
I (88895) NimBLE: att_handle=14
I (88895) NimBLE: GATT procedure initiated: notify; 
I (88905) NimBLE: att_handle=14
 

But then, buffers start to deplete. So, prints shown are:

> E (89215) bleprph_throughput: No MBUFs available from pool, retry..
I (89325) NimBLE: GATT procedure initiated: notify; 
I (89325) NimBLE: att_handle=14
I (89325) NimBLE: GATT procedure initiated: notify; 
I (89325) NimBLE: att_handle=14
I (89325) NimBLE: GATT procedure initiated: notify; 
I (89335) NimBLE: att_handle=14

But , the test keeps running and such prints occur on both central and peripheral side. 

Eventually the test finishes and we see :

On bleprph: 

> I (148365) NimBLE: GATT procedure initiated: notify; 
I (148375) NimBLE: att_handle=14
*********************************
I (148385) bleprph_throughput: Notify throughput = 154866 bps, count = 2323
*********************************

And on blecent side:

> E (148480) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
E (148490) NimBLE: ble_hci_rx_acl failed to allocate ACL buffers; increase ACL_BUF_COUNT
I (88320) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1346000
I (148510) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1346500
I (148520) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1347000
I (148530) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1347500
I (148540) blecent_throughput: received notification; conn_handle = 1 attr_handle = 14 attr_len = 500 ; Total length = 1348000

The above behaviour doesn't match to the observation you have mentioned : 

> if you try the example you will notice that the behaviors is different
after the first "failed to allocate ACL buffers" if bleprph regular send message, the blecent not receive anymore messages and it never stop the error loop until bleprph stop for long time (>500ms)

If the application behaviour at your end needs to be debugged, can you please let me know:

1. The steps you are using while running the application where above behaviour is seen 
2. The commit id you are currently on ?   " Please share the output of  ` git describe --tags ` "
3. The sdkconfig being used at your end ( for both the blecent / bleprph )

embedded-sm, 2023-03-01:
> The above behaviour doesn't match to the observation you have mentioned :

why you say that not match the my description?
look that at your side the ACL_BUF_COUNT error is repeated for the full duration of 60s of test (except for the first messages )
and blecent recover from that state only when bleprph stop, as here

> 1. The steps you are using while running the application where above behaviour is seen

on blecent i only change:

```
if (!cli_receive_key(test_data)) {
    /* No command supplied, start with reading */
    test_data[0] = NOTIFY_THROUGHPUT;
    test_data[1] = 60;
    ESP_LOGI(tag, "No command received from user, start with READ op"
             " with 60 seconds test time");
}
```

on bleprph i change different delay here:

```
/* Memory not available for mbuf */
ESP_LOGE(tag, "No MBUFs available from pool, retry..");
vTaskDelay(100 / portTICK_PERIOD_MS); //over 500ms, blecent recover from error loop
om = ble_hs_mbuf_from_flat(payload, sizeof(payload));
assert(om != NULL);
```

> 2. The commit id you are currently on ? " Please share the output of git describe --tags "

v5.0.1-3-g886e98a2c1
if need i can back to that V5.0.1 tag

> 3. The sdkconfig being used at your end ( for both the blecent / bleprph )

[sdkconfig_both.zip](https://github.com/espressif/esp-idf/files/10861789/sdkconfig_both.zip)

i notice a details, from your log (i verify it the same here) the timestamp of the log is wrong, when it recovers it has the same timestamp present at the begin of loop error (as some ble thread, take extra resource):
[image]
[image]

embedded-sm, 2023-03-01:
sorry i forgot to remove a extra log (incremental number) exchanged between blecent and bleprph, (without there are the same behaviors)

rahult-github, 2023-03-03:
Hi @embedded-sm ,  can you please check if attached patch helps ? 
[fast_notify.txt](https://github.com/espressif/esp-idf/files/10878982/fast_notify.txt)

The patch needs to be applied at $IDF_PATH.

embedded-sm, 2023-03-03:
thanks, this patch prevent the example to stuck as before.
this is a clue that the issue loop take resource and due to the high priority of the task all the rest of the code have problem, correct or if im wrong?

this delay is it only a test to understand the direction for a final fix, right? did you plan to keep improving this fix properly?

rahult-github, 2023-03-03:
Hi @embedded-sm ,

Thanks for confirming. So, the controller task priority is higher compared to stack and in this use case, the Rx packets are coming at a very fast rate. This is causing no chance to host stack to process and hence free up packets. Will review this change internally and fix this.

closed 2023-03-13 by espressif-bot