espressif/esp-idf #10852: Coredump (elf) data shiftted, causing idf.py coredump-debug not working

wuyuanyi135, 2023-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3636-g66ab22986f

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N8R8

### Power Supply used.

USB

### What is the expected behavior?

-

### What is the actual behavior?

I found occasionally, `idf.py coredump-debug` fails with the following stacktrace:
```
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 767, in <module>
    main()
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 702, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1689, in invoke
    return _process_result(rv)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1626, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 605, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 187, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf_py_actions\debug_ext.py", line 537, in coredump_debug
    espcoredump.dbg_corefile()
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 329, in dbg_corefile
    core_elf_path, target, temp_files = self.get_core_dump_elf(e_machine=exe_elf.e_machine)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 108, in get_core_dump_elf
    loader = ESPCoreDumpFlashLoader(self.off, self.chip, port=self.port, baud=self.baud)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\corefile\loader.py", line 429, in __init__
    self.target = self._load_core_src()
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\corefile\loader.py", line 151, in _load_core_src
    _header = EspCoreDumpV1Header.parse(coredump_bytes)  # first we use V1 format to get version
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 2120, in _parse
    subobj = sc._parsereport(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 2653, in _parse
    return self.subcon._parsereport(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 1041, in _parse
    data = stream_read(stream, self.length, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 91, in stream_read
    raise StreamError("stream read less than specified amount, expected %d, found %d" % (length, len(data)), path=path)
construct.core.StreamError: Error in path (parsing) -> tot_len
stream read less than specified amount, expected 4, found 0
```

Therefore, I dump the [coredump partition](https://github.com/espressif/esp-idf/files/10841085/coredump-original.zip). Supurisingly, it does not starts with the ELF magics (0x7f,E, L, F). See the picture below, the magic bytes in the circle are supposed to be the first bytes.

[image]

I then remove the bytes before the ELF header then debug with ` idf.py coredump-debug --core coredump` and it works without problem!

### Steps to reproduce.

I could not reproduce this error consistently. The coredump partition occasionally broke.

### More Information.

Partition Table:
```csv
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory,,  2500K,
ota_0,    app,  ota_0,, 2500K,
ota_1,    app,  ota_1,, 2500K,
coredump, data, coredump,,        64K
config,   data, nvs,,             256K

```

Attached archive contains the elf symbol file, modified coredump that can be debugged, and unmodified coredump. 

[Uploading coredump.zip…]()

igrr, 2023-02-27:
> Supurisingly, it does not starts with the ELF magics (0x7f,E, L, F). See the picture below, the magic bytes in the circle are supposed to be the first bytes.

That part — that the ELF magic is not at the beginning of the coredump partition — is expected and this should be the case _every time_. In flash, the ELF formatted coredump is prefixed with a header: 
* [writing the header](https://github.com/espressif/esp-idf/blob/57b6be22a7779bbfd194ba254f66bdbe95a18a7c/components/espcoredump/src/core_dump_elf.c#L602-L614)
* [reading the header](https://github.com/espressif/esp-coredump/blob/9b0d29ffda14d7f4d0ee281206ed83aadf1edd83/esp_coredump/corefile/loader.py#L151-L166)

Since the coredump-debug command is failing, it's probably a real bug somewhere, though. Thank you for the attached files, we will try to reproduce the issue and debug it.

wuyuanyi135, 2023-02-27:
@igrr Thanks for your quick response and explanation! 

For the prefixed coredump, I guessed it should be the case since the [docs ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump_internals.html#anatomy-of-core-dump-image) hint so. However, I traced the stacktrace when executing `idf.py coredump-debug -c coredump`. 

https://github.com/espressif/esp-idf/blob/57b6be22a7779bbfd194ba254f66bdbe95a18a7c/tools/idf_py_actions/debug_ext.py#L194

As you may see, it checks the elf magic at the begining of the file. I guess if the coredump was created by `idf.py` commands it will strip the extra starting bytes?

wuyuanyi135, 2023-02-27:
@igrr For your second request, I attach the full stacktrace here. 
```
Executing action: coredump-debug
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-master\components\partition_table\parttool.py", line 358, in <module>
    main()
  File "C:\Espressif\frameworks\esp-idf-master\components\partition_table\parttool.py", line 327, in main
    target = ParttoolTarget(**target_args)
  File "C:\Espressif\frameworks\esp-idf-master\components\partition_table\parttool.py", line 103, in __init__
    self._call_esptool(['read_flash', str(partition_table_offset), str(gen.MAX_PARTITION_LENGTH), temp_file.name])
  File "C:\Espressif\frameworks\esp-idf-master\components\partition_table\parttool.py", line 130, in _call_esptool
    raise e
  File "C:\Espressif\frameworks\esp-idf-master\components\partition_table\parttool.py", line 127, in _call_esptool
    subprocess.check_call(esptool_args, stdout=out, stderr=subprocess.STDOUT)
  File "subprocess.py", line 364, in check_call
subprocess.CalledProcessError: Command '['C:\\Espressif\\python_env\\idf5.1_py3.8_env\\Scripts\\python.exe', 'C:\\Espressif\\frameworks\\esp-idf-master\\components\\esptool_py\\esptool\\esptool.py', '--port', 'COM6', 'read_flash', '32768', '3072', 'C:\\Users\\wuyua\\AppData\\Local\\Temp\\tmp_9qrjqjo']' returned non-zero exit status 2.
ERROR:root:parttool script execution failed with err 1
ERROR:root:Error during the subprocess execution
Traceback (most recent call last):
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 767, in <module>
    main()
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 702, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1689, in invoke
    return _process_result(rv)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 1626, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 605, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf.py", line 187, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "C:\Espressif\frameworks\esp-idf-master\tools\idf_py_actions\debug_ext.py", line 537, in coredump_debug
    espcoredump.dbg_corefile()
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 329, in dbg_corefile
    core_elf_path, target, temp_files = self.get_core_dump_elf(e_machine=exe_elf.e_machine)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\coredump.py", line 108, in get_core_dump_elf
    loader = ESPCoreDumpFlashLoader(self.off, self.chip, port=self.port, baud=self.baud)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\corefile\loader.py", line 429, in __init__
    self.target = self._load_core_src()
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\esp_coredump\corefile\loader.py", line 151, in _load_core_src
    _header = EspCoreDumpV1Header.parse(coredump_bytes)  # first we use V1 format to get version
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 2120, in _parse
    subobj = sc._parsereport(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 2653, in _parse
    return self.subcon._parsereport(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 1041, in _parse
    data = stream_read(stream, self.length, path)
  File "C:\Espressif\python_env\idf5.1_py3.8_env\lib\site-packages\construct\core.py", line 91, in stream_read
    raise StreamError("stream read less than specified amount, expected %d, found %d" % (length, len(data)), path=path)
construct.core.StreamError: Error in path (parsing) -> tot_len
stream read less than specified amount, expected 4, found 0
```

wuyuanyi135, 2023-02-27:
And I do agree with you that might be some serial connection problem. I occasionally experience jammed flashing that stopped at certain percent of uploading and CTRL-C does not stop the program - I have to either unplug or reset the board so that it will dump tons of output suddenly. 

The PCB and electronics should be fine. I used the USB interface for communication in my app and it worked very stable. Only in bootloader did I experienced the interrupted data transfer. IIRC, this happens since a while ago, maybe the bootloader or the flash tool were changed recently?

wuyuanyi135, 2023-02-27:
I also leave my sdkconfig here, in case it provides some insights
<details>
  <summary>sdkconfig</summary>
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_ARBITER_SUPPORTED=y
CONFIG_SOC_ADC_DIG_IIR_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=24
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
[... 2182 lines cut ...]
# CONFIG_PPP_SUPPORT is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# CONFIG_USB_ENABLED is not set
# End of deprecated options
```
</details>

igrr, 2023-02-27:
Ah, i see. So the actual error is here:

```
subprocess.CalledProcessError: Command '['C:\\Espressif\\python_env\\idf5.1_py3.8_env\\Scripts\\python.exe', 'C:\\Espressif\\frameworks\\esp-idf-master\\components\\esptool_py\\esptool\\esptool.py', '--port', 'COM6', 'read_flash', '32768', '3072', 'C:\\Users\\wuyua\\AppData\\Local\\Temp\\tmp_9qrjqjo']' returned non-zero exit status 2.
ERROR:root:parttool script execution failed with err 1
ERROR:root:Error during the subprocess execution
```
Basically, esptool.py has failed to read the coredump partition and failed half-way. Then instead of exiting from the `idf.py coredump-debug` script, we carried on and passed the truncated file to coredump loader, which obviously failed because the file was truncated.

So the issue _you_ are facing is probably with esptool.py read_flash not working reliably. I would suggest running this command manually a few times. If you get it to fail occasionally, please check the troubleshooting instructions in https://github.com/espressif/esptool, then in case you can't figure out what the issue is, open a bug report in esptool repository.

The issue that we can fix in this ticket is that `idf.py coredump-debug` doesn't bail out on the first error and mistakenly proceeds with parsing the coredump.

---

> Also, I noticed you were using another script espcoredump.py instead of idf.py coredump-debug. Could you explain if there is any difference between them?

`idf.py coredump-debug` is correct and is the recommended way to debug the coredump from within an IDF project. `espcoredump.py` is a lower-level tool which needs more arguments to be passed manually, because it doesn't "know" about IDF projects, sdkconfig files, and such. I used the latter because I only had your .elf and coredump files, not the whole project, so `idf.py coredump-debug` wouldn't work for me. But aside from the differences in command line arguments, these tools use the same code internally (https://github.com/espressif/esp-coredump).

wuyuanyi135, 2023-02-27:
@igrr Many thanks for your detailed explanation. 

Yes, I agree that the output should be improved so that the actual problem will not be masked. I will keep monitoring if the bootloader I/O problem persists and open a issue in esptool repository.

wuyuanyi135, 2023-02-27:
@igrr in addition to what you suggest, it also worth noting that the coredump-debug handles the core file differently. When passing the coredump it assumes it is a base64 encoded file rather than raw elf due to magic check. Maybe this should also be documented or covered.

closed 2023-03-01 by espressif-bot

wuyuanyi135, 2023-04-25:
@dizcza it should be available AFAIK. Also, upgrading esptool to 4.6 rather than 4.5 shipped with 5.0 will drastically improve the stability during flashing and coredumping.