espressif/esp-idf #10853: Error "adc_continuous: adc_continuous_stop(446): release pm_lock failed", due to double esp_pm_lock_release

Erlkoenig90, 2023-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

When calling `adc_continuous_stop` with power management enabled, the ADC should stop and no errors should be printed.

### What is the actual behavior?

When calling `adc_continuous_stop` with power management enabled, the message attached is printed to the serial output, coming from [the continous_adc driver](https://github.com/espressif/esp-idf/blob/v5.0.1/components/esp_adc/adc_continuous.c#L446).

The PM lock was already released [here](https://github.com/espressif/esp-idf/blob/v5.0.1/components/esp_adc/adc_continuous.c#L432).

### Steps to reproduce.

1. Compile the application and IDF with power management enabled (`CONFIG_PM_ENABLE`).
2. Start and stop the continuous ADC.

### Debug Logs.

```plain
E (48075) adc_continuous: adc_continuous_stop(446): release pm_lock failed
```

### More Information.

This can be fixed by simply one of the calls to `esp_pm_lock_release`.

Icarus113, 2023-02-28:
@Erlkoenig90  Hi, thanks for the reporting. We'll fix this soon.

closed 2023-03-03 by espressif-bot

Erlkoenig90, 2023-03-08:
Great, thanks!