espressif/esp-idf #10857: Minus symbol stripped from negative numbers by console component

tadejg, 2023-02-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have setup a REPL following the basic console example. For one of my commands, I want to accept negative numbers as the value. The command looks like this: `set <key> <value>`. When I run the command as `set some_key -5` I get the following error `set: invalid option "-?"`. I have tried escaping the `-` which worked, but made the number positive.
```
repl> set some_key \-5
I (81716) repl: Updating value to 5
```

I also tried wrapping the value in quotes and double escaping the `-` symbol with no better results.
```
repl> set some_key \\-5
I (199946) repl: Updating value to \-5
repl> set some_key "-5"
set: invalid option "-?"
Command returned non-zero error code: 0x1 (ERROR)
repl> set some_key "\-5"
I (224686) repl: Updating value to 5
```

The only way I was able to get this working was by following the `cmd_nvs` component from the advanced console example where the value is passed after `-v` or `--value`.
```
repl> set some_key -v -3
I (17196) repl: Updating value to -3
```

This time it worked without having to escape anything. I can use this as a workaround for the time being, but would prefer to drop the `-v` since you're always expected to pass a value to the `set` command anyways.

The only difference in code was with the args definition:
```c++
// Doesn't work
static struct {
    struct arg_str *key = arg_str1(NULL, NULL, "<key>", "Key to update");
    struct arg_str *value = arg_str1(NULL, NULL, "<value>", "Value to set");
    struct arg_end *end = arg_end(2);
} setArgs;

// Works
static struct {
    struct arg_str *key = arg_str1(NULL, NULL, "<key>", "Key to update");
    struct arg_str *value = arg_str1("v", "value", "<value>", "Value to set");  // This line changed
    struct arg_end *end = arg_end(2);
} setArgs;
```

igrr, 2023-02-27:
@tadejg Since the behavior is the same with argtable3 on Linux ([reproducer](https://gist.github.com/igrr/42ced595140f91c7d5dfd8f98a91a8ab)), I would recommend reporting this upstream at https://github.com/argtable/argtable3/issues. If this issue is fixed upstream, we will update argtable3 to the latest release version.

tadejg, 2023-02-27:
Understood, will report it upstream then, thanks.

closed 2023-02-27 by tadejg