espressif/esp-idf #10858: python.exe doesn't support asyncio. The issue can be worked around by re-running idf.py with the "--no-hints" argument.

RJSDevel, 2023-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Executing action: flash
Running ninja in directory C:\Users\Developer\espressif\HA_on_off_light\build
Executing "ninja flash"...
ERROR: D:\Espressif\python_env\idf5.1_py3.7_env\Scripts\python.exe doesn't support asyncio. The issue can be worked around by re-running idf.py with the "--no-hints" argument.

### What is the actual behavior?

Upload

### Steps to reproduce.

try to flash anything

dobairoland, 2023-02-28:
Please use the Windows tool installer from https://dl.espressif.com/dl/esp-idf/ in order to install a working environment. The Python environment the installer offers doesn't have this issue and solves other potential issues as well.

dobairoland, 2023-02-28:
@georgik Just in idea: Maybe the installer needs another sanity check for determining which Python is suitable. We could test if asyncio support is present or not.

There was another report recently (#10712).

georgik, 2023-02-28:
@dobairoland The embedded 3.8 shipped with the installer should be ok. Or are there some missing parts. It is some time after it is EOL, but it is the only remaining version supporting Windows 7. Probably with dropping of 4.1 by the end of February we can drop also 3.8 and move forward to more modern version.

dobairoland, 2023-02-28:
> The embedded 3.8 shipped with the installer should be ok. Or are there some missing parts? 

@georgik Yes, that one is perfectly fine. 

At least at some point there was the possibility to select Pythons which are already installed on the computer. I remember you mentioned some kind of sanity checks these need to pass in order to be eligible to use it with ESP-IDF. Is this no longer supported? If not then you can ignore my suggestion.

RJSDevel, 2023-02-28:
> Please use the Windows tool installer from https://dl.espressif.com/dl/esp-idf/ in order to install a working environment. The Python environment the installer offers doesn't have this issue and solves other potential issues as well.

Yes, that works

closed 2023-03-01 by dobairoland

Yolnit, 2023-06-23:
Answers checklist.
 I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.

 I have checked if the solution shared before works but nothing changed.

Operating System used.
Windows

How did you build your project?
With the toolbar "build" then run but run is not working

If you are using Windows, please specify command line type.
None

Development Kit.
ESP32-C6-WROOM-1

Power Supply used.
USB

What is the expected behavior?
C:\esp-idf-v5.1-rc2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: flash
Running ninja in directory C:\****\****\eclipse-workspace_esp32\test4\build
Executing "ninja flash"...
ERROR: C:\Espressif\python_env\idf5.1_py3.7_env\Scripts\python.exe doesn't support asyncio. The issue can be worked around by re-running idf.py with the "--no-hints" argument.

What is the actual behavior?
return error

Steps to reproduce.
try to flash anything

Debug Logs.
No response

More Information.
No response