espressif/esp-idf #10859: esp_timer_is_active() should have IRAM_ATTR as documentation requires to check before stop or start

acninja, 2023-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Relates to: 4.4.4 up to master
Use case: ISR and bootloader during flash erase and write operations.

**esp_timer** functions used in runtime are all placed in **IRAM** except **esp_timer_is_active()** which looks like an oversight.
The documentation **requires** us to check if the timer is active before stopping and starting the timer.

Currently the lack of **IRAM_ATTR** on **esp_timer_is_active()** causes my bootloader to WDTreset in communication ISR.
Please add **IRAM_ATTR** to **esp_timer_is_active()** function.

AxelLin, 2023-02-28:
https://github.com/espressif/esp-idf/issues/10522

AxelLin, 2023-04-22:
@KonstantinKondrashov 
The commit 
745cac7 subject says "esp_timer: Adds IRAM_ATTR for esp_timer_restart and esp_timer_is_active" but it does not change esp_timer_restart at all.

igrr, 2023-04-22:
In this case we have decided to keep the message the same between branches to make the change easier to track.

KonstantinKondrashov, 2023-04-22:
@AxelLin this commit is backport for release/v4.4, this branch does not have this esp_timer_restart func.
 The commit for the master was merged too but will appear later on GitHub.

AxelLin, 2023-05-06:
@KonstantinKondrashov 
Would you backport the fix for v4.3 as well?