espressif/esp-idf #10860: If run esp-idf-master/install.ps1 then the script install again idf5.1_py3.7_env

RJSDevel, 2023-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I've installed the esp-idf-master via Online Installer 2.20, and it was ok, but when I run esp-idf-master/install.ps1 it installs second python smaller version idf5.1_py3.7_env

### What is the actual behavior?

Do nothing maybe

### Steps to reproduce.

run esp-idf-master/install.ps1

dobairoland, 2023-02-28:
> I've installed the esp-idf-master via Online Installer 2.20, and it was ok, but when I run esp-idf-master/install.ps1 it installs second python smaller version idf5.1_py3.7_env

Have you opened the ESP-IDF Powershell environment? The shortcut was installed by the installer.

If you have run the script from a regular Powershell windows then the above behavior is expected. Your system `python.exe` is linked to the 3.7 version instead of the embedded 3.8 installed by the installer. That is why it creates a new environment.

RJSDevel, 2023-02-28:
> Have you opened the ESP-IDF Powershell environment? The shortcut was installed by the installer.
>
> If you have run the script from a regular Powershell windows then the above behavior is expected. [...]

No, I haven't. I've just clicked in context menu on the install.ps1 and executed it

dobairoland, 2023-03-01:
All ESP-IDF commands should be run in ESP-IDF environment. Shortcuts are installed by the installer. 

I'm closing this as it should be solved.

closed 2023-03-01 by dobairoland