espressif/esp-idf #10861: esptool not working with latest bitstring version

simon4web, 2023-02-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

espefuse.py -p xxxx summary

### What is the actual behavior?

Python Import error
from bitstring import BitString
Apperently V4.0.0 of bitstring is missing BitString
Fixed by adding <4.0.0 in requirements.txt

### Steps to reproduce.

1. Step
2. Step
3. Step
...
...

igrr, 2023-02-28:
This was fixed in commit https://github.com/espressif/esp-idf/commit/7f732720f0381a290d00a5027409710e5a6a73ff, part of IDF v4.4.4 release.

dobairoland, 2023-03-01:
I'm closing this. Please let us know if upgrading to v4.4.4 didn't help.

closed 2023-03-01 by dobairoland