espressif/esp-idf #10864: Continuous ADC mode should allow overrunning buffer

felixcollins, 2023-02-28:
### Is your feature request related to a problem?

I have a system where I need to capture ADC data for a given number of samples before and after an asynchronous external event. Outside of this window I do not care about the data.

### Describe the solution you'd like.

I have implemented this on STM32 using the ADC/DMA in continuous circular mode. I would like to use the continuous mode driver on ESP32 to do this. 

The driver needs:
1. A configuration flag to allow overwriting old data instead of halting with an error.
2. A function to read back the data from the entire buffer starting with either the most recent or the oldest sample.

This is how I would use it:
1. Set ADC/DMA going on a large circular buffer (with my specified number of samples) . The DMA simply overwrites the oldest samples rather than setting a pool overflow event and stopping.
3. I can at some arbitrary point halt the ADC.
4. At my leisure I can read the samples out from the buffer one by one.

https://www.esp32.com/viewtopic.php?f=13&t=28608

### Describe alternatives you've considered.

Using the continuous mode driver,  I have to copy the data from the dma buffer into another circular buffer that I maintain. Most of this data will be thrown away so the copy overhead is wasted.

Icarus113, 2023-03-01:
@felixcollins  Hi, thanks for the request. We're considering this as well, will let you know when there's update.

Icarus113, 2023-04-11:
> Set ADC/DMA going on a large circular buffer (with my specified number of samples) . The DMA simply overwrites the oldest samples rather than setting a pool overflow event and stopping.

Sorry for late updates, would you help check if the [callback usage](https://github.com/espressif/esp-idf/issues/10875#issuecomment-1451183615) meets this requirement? @felixcollins

felixcollins, 2023-04-11:
@Icarus113 great! Thanks. The main change required is to allow a setting that stops the overflow behavior.

felixcollins, 2023-05-12:
@Icarus113 Any update on this? I'm hoping it will also fix #10612

Icarus113, 2023-05-12:
@felixcollins  It's in progress, though interrupted by some stuffs. Will let you know the update soon~

Icarus113, 2023-05-15:
[adc_new_data.txt](https://github.com/espressif/esp-idf/files/11474102/adc_new_data.txt)

Hi @felixcollins , would you try with this patch? It's based on the latest master. (I renamed the file to .txt)

felixcollins, 2023-05-16:
Hi @Icarus113. 

I note that the new flag is named "flush_pool" this makes it sound like the whole buffer will be cleared on overrun.  This is not what I need, I just want to overwrite the oldest individual samples. The comment "* Replace by ringbuffer reset API when this API is ready." also indicates you plan to clear the old data completely.  The code looks like it is copying from the DMA buffer to a ring buffer. I want to avoid that step (copy overhead).

This is more what I'm after - dma/adc running continuously with no copying to ring buffers etc until stopped.
https://github.com/espressif/esp-idf/issues/10804#issuecomment-1549409748
https://github.com/espressif/esp-idf/pull/11411

Icarus113, 2023-05-17:
@felixcollins Thanks for your feedback~

> I just want to overwrite the oldest individual samples

Actually there are few ways to do this
1. pop out the oldest ringbuffer item, adding a ringbuffer send to front feature and send the new data to the front of the ringbuffer
2. pop out the oldest ringbuffer item, and send to the back of the ringbuffer
3. flush all the ringbuffer, and send with new data

The patch is adding the third way.

## WRT updating the DMA buffer, 

The DMA keeps filling new data to the DMA buffers. 

If you want a way (let's say way 4) that, you provide the DMA buffers, and the driver help you load the buffers to the DMA, and DMA keeps filling them with new data. We possibly won't do this, and you might need to implement this in your own. The risk for this is, if you're accessing the DMA buffer, while the DMA is filling them, we cannot guarantee the data consistency. 

Your requirement to me is like, you need a pool to save the newest data, and at leisure time you read them back. Current ringbuffer is for this usage. We can consider adding the three ways I listed above, but not the way 4.

## WRT the pull request

It updates the way how the driver internal DMA buffers usage, but the way that:
- conv_done callback
- filling data to ringbuffer

are not changed. I'll review that one later.

felixcollins, 2023-05-17:
I want 4. Using a ringbuffer and copying the data just to then throw it away is very inefficient (more power consumption for a battery powered device). 

Imagine a device that behaves a bit like a triggering oscilloscope. No data is accessed in the dma buffers until the capture is stopped (by an external event trigger). After stopping the DMA/ADC we can review the captured data from before and after the trigger. This is a very common use case for ADC. If the IDF will not support this with an API could you at least provide a good, clearly commented, sample that has the correct calls to the HAL? The technical reference manual and HAL api is very sparse on programming details.  

WRT to the PR I linked, I realise that it does still use the ringbuffer and is just a bug fix for another critical outstanding issue, however he does manage to make the DMA run in a circular mode which is what I wish to do. 

Thanks for looking into this for me. If there is no IDF support coming you may as well close this ticket.

closed 2023-06-01 by Icarus113

felixcollins, 2023-06-09:
@Erlkoenig90 how hard would it be to start with your continuous ADC changes and implement my feature request here? Basically one buffer that is overwritten in circular fashion, until a user request to stop. Sort of like a digital sampling oscilloscope with trigger. The key point is that there is no copying of data (except the DMA read) while the run is going.

Erlkoenig90, 2023-06-15:
> @Erlkoenig90 how hard would it be to start with your continuous ADC changes and implement my feature request here? 

I think this should be doable, you'd have to implement your own ADC driver for that.