espressif/esp-idf #10868: Problems when two RTC-GPIOs are used as wake-up pin from deep-sleep

tonyVopo, 2023-03-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-8684-MINI-1

### Power Supply used.

USB

### What is the expected behavior?

I want to use GPIO_ NUM_2 and GPIO_ NUM_3 is used as the wake-up pin for deep sleep, and sometimes I will only use one of them as the wake-up pin, and hope that the other cannot wake up the system。

### What is the actual behavior?

Once when I only configured GPIO_ NUM_ 3 as a wake-up pin, I found GPIO_ NUM_ 2 It can also wake up the system.

### Steps to reproduce.

1. Enable GPIO_NUM_3 and GPIO_NUM_2 as the wake up pin.  
2. Enter to deep-sleep 
3. Use GPIO_ NUM_ 2 Wake up system
4. Enable just GPIO_NUM_3 as the wake up pin
5.  ### Use GPIO_ NUM_ 2 Wake up system successfully
...

### Debug Logs.

```plain
Enable GPIO_NUM_3 and GPIO_NUM_2 as the wake up pin
log:
  gpio: GPIO[2]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
  gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0  

Enable just GPIO_NUM_3 as the wake up pin
log：
  gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
```

### More Information.

API instructions：
    esp_deep_sleep_enable_gpio_wakeup( BIT(GPIO_NUM_2), ESP_GPIO_WAKEUP_GPIO_HIGH );
    esp_deep_sleep_enable_gpio_wakeup( BIT(GPIO_NUM_3), ESP_GPIO_WAKEUP_GPIO_HIGH );
     esp_deep_sleep_start();

tadejg, 2023-03-01:
I had a similar problem in on of my projects. What I found was that I had to explicitly disable GPIOs I don't want to use for wakeup. Also, my project was for ESP32c3, so I'm not sure if this will carry over to the ESP-8684-MINI-1 you're using. Not sure if this is a feature or a bug, but hopefully this helps you solve your issue.

```c
if(use both to wakeup) {
  esp_deep_sleep_enable_gpio_wakeup( BIT(GPIO_NUM_2), ESP_GPIO_WAKEUP_GPIO_HIGH );
  esp_deep_sleep_enable_gpio_wakeup( BIT(GPIO_NUM_3), ESP_GPIO_WAKEUP_GPIO_HIGH );
} else {
  gpio_deep_sleep_wakeup_disable(GPIO_NUM_2); // Note that you need to pass the pin number and not a bit mast when disabling
  esp_deep_sleep_enable_gpio_wakeup( BIT(GPIO_NUM_3), ESP_GPIO_WAKEUP_GPIO_HIGH );
}
esp_deep_sleep_start();
```

tonyVopo, 2023-03-02:
Thx, I solved the problem according to your opinion

closed 2023-03-02 by tonyVopo