espressif/esp-idf #10873: IDF 5.0.1 BLE scan never ends

DirkGmailMeintjies, 2023-03-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When calling esp_ble_gap_start_scanning(10) then current increases by ~70mA (ESP32-C3) and I get gap callback for:

ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
ESP_GAP_BLE_SCAN_RESULT_EVT

And then ESP_GAP_BLE_SCAN_RESULT_EVT repeats for every advertisement message. After 10 seconds I get no more callbacks for ESP_GAP_BLE_SCAN_RESULT_EVT, but also no ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT callback to say scanning ended... Previous versions I did get the ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT callback after 10 seconds. Also the current stays >70mA so internally the BLE radio is still switched on. 

If I then call esp_ble_gap_stop_scanning() then I see error output:

(278895) BT_BTM: BTM_BleScan scan not active
(278895) BT_APPL:  bta_dm_ble_scan stop scan failed, status=0x6

and I get gap callback ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT with error parameter.

But if I call esp_ble_gap_stop_scanning() before the 10 seconds pass (i.e. while still scanning) then I get successful gap callback ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT, and also the current drops again by 70mA - so the radio is switched off.

The problem is: Previous versions switched the radio off itself when scanning time is over and calls gap callback event ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT. New v5.0.1 does not call callback event and does not turn off radio. I cannot call esp_ble_gap_stop_scanning() after scan time because I get error message and current stays high (BT radio stays switched on).

ilovebt9, 2023-09-05:
i have the same issue, 
disabling "Report adv data and scan response individually when BLE active scan" solved the problem for me,
i have activated this option previously but not really know what is doing, reponses seems to be triggered one by one by bluedroid, btw

zhp0406, 2023-09-19:
@ilovebt9 @DirkGmailMeintjies 
Thank you for reporting the problem. Could you please provide the complete log to help us analyze the issue?

zhp0406, 2024-07-02:
ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT is only triggered when esp_ble_gap_stop_scanning is called. If esp_ble_gap_start_scanning(10) is called, scanning will stop after 10 seconds, which means ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT event will not be generated, only ESP_GAP_BLE_SCAN_RESULT_EVT event will be generated.

zhp0406, 2024-07-02:
When modem sleep is activated, the device will enter sleep mode if there is no TX or RX activity.
(Top) → Component config → Bluetooth → Bluetooth → Bluetooth controller → MODEM SLEEP Options
[*]     Bluetooth Modem sleep Mode 1

zhp0406, 2024-07-02:
@mrx23dot 
When a scan is already in progress, calling esp_ble_gap_start_scanning again will result in an error, which is expected behavior.

closed 2025-04-10 by Alvin1Zhang