espressif/esp-idf #10874: WebSocket -> httpd_ws_recv_frame: WS frame is not properly masked

Mowglli, 2023-03-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.3-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I have a designed PCB with the ESP32-S3 and a USB-C, I have implemented it so when i plug it in to my computer, it pop up as a RNDIS-device, it gets a IP:2.2.2.2 and a MAC-adress. There is also implemted a webserver/websocket, where i send data to the ESP via the USB(ethernet).
When i send a data array[512] through the websocket, the data is handled in the ESP for further use. The data array is send as a binary array and is translated on the ESP.

### What is the actual behavior?

When i send data through the websocket, most of the data is received correctly but sometimes it prints out WS frame is not properly masked.
Then it opens a new connection, and it runs fine for a while and then the same as above.
The data array is sent every 40 ms like the following: 
```
function sendToWs(sliderValues) {
               dmx_data[0] = 10;
               for (let i = 1; i < sliderValues.length; i++) {
                  dmx_data[i] = parseInt(sliderValues[i]);
               }
               const buffer = new ArrayBuffer(dmx_data.length); // create an ArrayBuffer with the same length as the array
               const view = new DataView(buffer); // create a DataView to write the data to the buffer
               for (let i = 0; i < dmx_data.length; i++) {
                  view.setUint8(i, dmx_data[i]); // write each element of the array to the DataView as an unsigned 8-bit integer
               }
               if (ws.readyState) {
               ws.send(buffer); // send the buffer via WebSocket
               console.log(view);
               }
            }
```

### Steps to reproduce.

1. In app_main - > The usb_service is first initialized -> then the HTTP server -> then the HTTP Task(which at the moment does nothing)
2. HTTP-server 
```
static esp_err_t ws_handler(httpd_req_t *req) {
	uint64_t tStart = esp_timer_get_time();
	uint8_t *buf = NULL;
	if (req->method == HTTP_GET) {
		ESP_LOGI(TAG, "Handshake done, the new connection was opened");
		return ESP_OK;
	}
	/* Set max_len = 0 to get the frame len */
	esp_err_t ret;	// = httpd_ws_recv_frame(req, &ws_pkt, 2000);
	httpd_ws_frame_t ws_pkt;
	memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
	ws_pkt.payload = (uint8_t*) readBufWS;
	/* Set max_len = 0 to get the frame len */
	ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
	if(ws_pkt.len){
		buf = calloc(1,ws_pkt.len +1);
		if(buf == NULL){
			ESP_LOGE(TAG, "Failed to calloc memory for buf");
			free(buf);
			return ESP_ERR_NO_MEM;
		}
		ws_pkt.payload = buf;
		ret = httpd_ws_recv_frame(req, &ws_pkt,ws_pkt.len);
		if(ret != ESP_OK){
			free(buf);
			return ret;
		}

		
	}
```
3. After the recieved frame the following is done for the passing af the array to furher use:
```
if (ws_pkt.type == HTTPD_WS_TYPE_BINARY) {
		    if (memcmp(ws_pkt.payload, "\x0a", 1) == 0) {
		        dmx_packet_t dPack;
		        dPack.MsgSource = (void *)1;
		        uint8_t *dmx_data = dPack.DMX;
		        uint8_t *payload_data = ws_pkt.payload + 1;  // skip the "0a" prefix
		        int payload_len = ws_pkt.len - 1;
		        if (payload_len > 513) {
		            // Payload is too long, ignore it
		        	ESP_LOGI(TAG,"Ignoring payload to big");
		        }
		        memcpy(dmx_data, payload_data, payload_len);
		        if (xQueue_dmx) {
		            xQueueSend(xQueue_dmx, &dPack, 0);
		        }
		    }
		}
```

### Debug Logs.

```plain
I (509) TinyUSB: TinyUSB Driver installed
I (513) system_api: Base MAC address is not set
I (518) system_api: read default base MAC address from EFUSE
I (525) USB_service: mac address: 70 4 1d a1 94 XX
I (532) ./main/http_service.c: Registering URI handlers
I (536) ./main/http_service.c: Registering URI /
I (542) ./main/http_service.c: Registering URI /c.css
I (547) ./main/http_service.c: Registering URI /jquery-3.6.0.min.js
I (554) ./main/http_service.c: Registering URI /progress.json
I (561) ./main/http_service.c: Registering URI /reboot
I (566) ./main/http_service.c: Registering URI /update
I (572) ./main/http_service.c: Registering URI /upload
I (578) ./main/http_service.c: Registering URI /ws
I (2162) USB_service: tud_network_init_cb.
I (4231) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 2.2.2.3
I (12305) ./main/http_service.c: Handshake done, the new connection was opened
I (22495) ./main/http_service.c: Handshake done, the new connection was opened
W (22512) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now
W (23292) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
W (23294) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
W (23297) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
I (23324) ./main/http_service.c: Handshake done, the new connection was opened
W (23332) httpd_ws: httpd_ws_get_frame_type: Failed to read header byte (socket FD invalid), closing socket now
I (24344) ./main/http_service.c: Handshake done, the new connection was opened
W (28582) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
I (29624) ./main/http_service.c: Handshake done, the new connection was opened
W (56064) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
W (56097) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
I (57138) ./main/http_service.c: Handshake done, the new connection was opened
W (61080) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
I (62122) ./main/http_service.c: Handshake done, the new connection was opened
W (81400) httpd_ws: httpd_ws_recv_frame: WS frame is not properly masked.
```

### More Information.

[image]

ESP-YJM, 2023-03-06:
@Mowglli From the log, it seems the websocket packet from client is  not masked and the websocket server reject it. Please help check whether the websocket packets from client masked. 
Also you can share the whole wireshark packet attached in this issue.

Mowglli, 2023-03-06:
@ESP-YJM  I have checked on wireshark and it looks like all of the packages sent from client is masked. I have included the wireshark capture down below. 

[http.zip](https://github.com/espressif/esp-idf/files/10896177/http.zip)

ESP-YJM, 2023-03-06:
@Mowglli Could you please help use the following `ws_handler` to reproduce the issue. I have removed your application logic code and add some print.
```
static esp_err_t echo_handler(httpd_req_t *req)
{
    if (req->method == HTTP_GET) {
        ESP_LOGI(TAG, "Handshake done, the new connection was opened");
        return ESP_OK;
    }
    httpd_ws_frame_t ws_pkt;
    uint8_t *buf = NULL;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    /* Set max_len = 0 to get the frame len */
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 0);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_recv_frame failed to get frame len with %d", ret);
        return ret;
    }
    ESP_LOGI(TAG, "frame len is %d", ws_pkt.len);
    if (ws_pkt.len) {
        /* ws_pkt.len + 1 is for NULL termination as we are expecting a string */
        buf = calloc(1, ws_pkt.len + 1);
        if (buf == NULL) {
            ESP_LOGE(TAG, "Failed to calloc memory for buf");
            return ESP_ERR_NO_MEM;
        }
        ws_pkt.payload = buf;
        /* Set max_len = ws_pkt.len to get the frame payload */
        ret = httpd_ws_recv_frame(req, &ws_pkt, ws_pkt.len);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "httpd_ws_recv_frame failed with %d", ret);
            free(buf);
            return ret;
        }
    }
    ESP_LOGI(TAG, "Packet type: %d, len is %d", ws_pkt.type, ws_pkt.len);
    free(buf);
    return ret;
}
```

Mowglli, 2023-03-06:
I have used the above mentioned code in the ws_handler. From the client im still sending the same packet every 40 ms.
[image]

Mowglli, 2023-03-06:
@ESP-YJM 
I have tried to make a simple websocket client in python, sending the binary array with 0's:
```
#!/usr/bin/python

import struct
import time
from websocket import create_connection

# Create a WebSocket connection
ws = create_connection('ws://2.2.2.2:80/ws')

# Create a binary array of 512 zeros
data = bytearray(513)

# Pack the binary data into a bytes object
packed_data = struct.pack('{}B'.format(len(data)), *data)

# Send the binary data every 40ms
while True:
    ws.send_binary(packed_data)
    time.sleep(0.02)  # sleep for 20ms

# Close the WebSocket connection (this line will never be executed in this example)
ws.close()
```
I am here sending every 20 ms, where i also see the ws_frame not properly masked. On wireshark i see the following now. 

[image]

Here is the full capture: 
[http2.zip](https://github.com/espressif/esp-idf/files/10897340/http2.zip)

ESP-YJM, 2023-03-06:
Hi @Mowglli , i use your python code and `ws_handler` i posted before, and i can not reproduce the issue. It works normal for about 10 minutes(maybe longer, due to i stop it). I use the Wi-Fi interface. I doubt the data sent from client maybe broken. The data maybe broken in ESP side, but i could not reproduce it. So could you please try WiFi interface and tell me about your used interface.

Mowglli, 2023-03-07:
I got the same results as you with wifi.
Right now it is set to be an RNDIS device, using TinyUSB library.
Could it be a timing issue in the usb_service ?

ESP-YJM, 2023-03-07:
@Mowglli I‘m not familiar with USB. But i think you can print the data received from USB. The issue is that the packet has been broken. So you can compare the wireshark packet data with the USB data you print.

Mowglli, 2023-03-07:
@ESP-YJM 
I have just tried what you mentioned. It dosent seem to me corrupted. But i'll do more test.
Based on your knowledge around http, how would you recommend the task priorities. Here is the following right now:
LWIP = 8
USB = 7
HOMEPAGE TASK = 5 (dosent do much at the moment)
SENDER = 5

ESP-YJM, 2023-03-08:
@Mowglli A larger value indicates a higher priority. I think it is OK for those priorities.

Mowglli, 2023-03-12:
Thanks, I have have done some more testing, and it seems to still happen, with the ws frame not properly masked

ESP-YJM, 2023-03-13:
@Mowglli Do with wifi still happen this? Or only happen in USB?

Mowglli, 2023-03-13:
I harpens with USB. Could it be an isser regarding the implementation of the USB? Maybe a pbug pt something overflowing/not free’d?

Venlig hilsen
Jesper Egsgaard Nielsen

Mowglli, 2023-03-15:
@ESP-YJM 

`static void task(void *arg){
	esp_netif_t *netif = (esp_netif_t *) arg;
	size_t hs;

	while(1){
		/* Try and "take" the semaphore. This task will "sleep" until either the
		           semaphore is "given" by another task/ISR or the timeout (e.g. 10000ms) elapses. */
		if ( xSemaphoreTake(tud_data_recd_sem, 10000 / portTICK_PERIOD_MS) == pdTRUE ) {
			// Data was received!
			if (tud_rx_len > 0){
				esp_netif_receive(netif, tud_rx_buf, tud_rx_len, NULL);
				//ESP_LOGI(TAG,"%x, %x, %x, %x, %x, %x, %x, %x",tud_rx_buf[0],tud_rx_buf[1],tud_rx_buf[2],tud_rx_buf[3],tud_rx_buf[4], tud_rx_buf[5], tud_rx_buf[6], tud_rx_buf[7]);
				tud_network_recv_renew();
				tud_rx_len = 0; //indicate buf is now read
				hs = xPortGetFreeHeapSize();
				//ESP_LOGI(TAG, "rx heap check (%u).", hs);
			}
		} else {
			// Semaphore take timed out after 10000ms, i.e. no data received within 10000ms. Do we care?
			ESP_LOGI(TAG, "no data received in the last 10s, will continue to wait.");
		}
	}
}`

Here is the transmit

`static esp_err_t netsuite_io_transmit(void *h, void *buffer, size_t len) {
	for (;;) {
		if (!tud_ready()) {
			return ESP_FAIL;
		}
		if (tud_network_can_xmit()) {
			struct pbuf *p = pbuf_alloc(PBUF_RAW, len, PBUF_POOL);
			if (p) {
				memcpy(p->payload, buffer, len);
				tud_network_xmit(p, 0);
				pbuf_free(p);
				return ESP_OK;
			} else {
				return ESP_FAIL;
			}
		}
	}
}`

Here is the recieve:

`bool tud_network_recv_cb(const uint8_t *src, uint16_t size) {
	if (src == NULL) {
		ESP_LOGE(TAG, "Invalid source pointer");
		return false;
	}
	if (size == 0) {
		ESP_LOGE(TAG, "Invalid data size: %d", size);
		return false;
	}
	if (tud_rx_len < 1) {
		memcpy(tud_rx_buf, src, size);
		tud_rx_len = size; //set tud_rx_len ONLY after finish copying
		// "Give" the semaphore, signalling that data was received:
		xSemaphoreGive(tud_data_recd_sem);
		return true;
	}else{
		//unable to receive, busy
		ESP_LOGW(TAG, "tud_network_recv_cb busy.");
		return false;
	}
}`

ESP-YJM, 2023-03-16:
@Mowglli I think you can add print in function `tud_network_recv_cb` and print `src` Byte by Byte, and add print before `esp_netif_receive` to print `tud_rx_buf` Byte by Byte. Compare with those two data and check whether is the same. If the same, and check the data with `httpd_ws` received. If still the same, the problem may occur in low-layer USB receive.

ESP-YJM, 2023-03-16:
@Mowglli Besides, i think you can refer to https://github.com/espressif/esp-iot-bridge about USB part.

Mowglli, 2023-03-16:
@ESP-YJM Thank you for your time. I appreciate it. 
I'll take a look at the link you send me. Maybe i can rewrite some stuff of mine.

closed 2023-03-16 by Mowglli