espressif/esp-idf #10876: settimeofday() sets the local time instead of the UTC time

mbastida123, 2023-03-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3

### Power Supply used.

External 3.3V

### What is the expected behavior?

We have a custom board with an external RTCC to keep the time even if the power goes out.

The time is set via SNTP when there's internet available. 

Each time the ESP32 wakes up from deep sleep the ESP32 reads the time from the external RTC and sets the ESP32's internal RTC with that time. This way we ensure that the ESP32 has the correct time even if its internal RTC has been shut down.

So, assuming the time has been successfully set we:
```
setenv("TZ", "CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00", 1);
tzset();
getTimeFromExternalRtc()
settimeofday(&tset, NULL);
```

The RTC always keeps the time in UTC format. We are sure of that because when we sync the time with the internet we also update the external RTC via:
```
time(&now);
timeinfo = gmtime(&now);
pcf8523_update_time(timeinfo); //the function to set the external RTC time
```

### What is the actual behavior?

settimeofday() interprets that the hour being set is in local time. This means that both UTC and local time are incorrect after the ESP32 wakes up from deep sleep.

### Steps to reproduce.

1. Set the tz, this is the one for Spain:
```
        setenv("TZ", "CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00", 1);
        tzset();
```
3. Use settimeofday() to set an hour, this hour is in UTC (seconds since epoch)
4. Now check the UTC time and the local time, the UTC time should be equal (with a small second difference) to the hour set in the step before.
You can use this:

```
        char strftime_buf[64];
        time_t now;

        time(&now);
        struct tm *tt = gmtime(&now);
        strftime(strftime_buf, sizeof(strftime_buf), "%c", tt);
        ESP_LOGI(TAG, "The current UTC date/time according to ESP32: %s", strftime_buf);

        localtime_r(&now, tt);
        strftime(strftime_buf, sizeof(strftime_buf), "%c", tt);
        ESP_LOGI(TAG, "The current local date/time according to ESP32: %s\n", strftime_buf);
```

igrr, 2023-03-01:
@mbastida123 If I understood the case you are describing correctly, it produces what looks to me like correct behavior.
```
I (322) main: Setting time to 1677683977
I (322) main: The current UTC date/time according to ESP32: Wed Mar  1 15:19:37 2023
I (332) main: The current local date/time according to ESP32: Wed Mar  1 16:19:37 2023
```
(Code is in https://gist.github.com/igrr/d7db8a78170bf6981f2e606b42c4361c)

And checking against Python implementation, same timestamp value converted to UTC:
```
$ python -c 'import datetime; print(datetime.datetime.utcfromtimestamp(1677683977).strftime("%c"))'                
Wed Mar  1 15:19:37 2023
```
which matches the UTC time printed by ESP.

IDF also has a [test case](https://github.com/espressif/esp-idf/blob/94c87a91060ff6e5962ec19e7152fc402e6c704c/components/newlib/test/test_newlib.c#L71-L96) to verify this, and as far as I can tell it's passing okay.

Could you please elaborate this step in more detail?

> Use settimeofday() to set an hour, this hour is in UTC (seconds since epoch)

Since settimeofday is not taking hours but the time since epoch, you must be doing some conversion here. Can you show how you are doing it?

mbastida123, 2023-03-01:
> @mbastida123 If I understood the case you are describing correctly, it produces what looks to me like correct behavior.
>
> ```
> I (322) main: Setting time to 1677683977
> I (322) main: The current UTC [...]

Hi @igrr,

This is the code I use every time I wake up from deep sleep:
```
    struct tm time_var;
    time_var = pcf8523_get_time();

    time_t tt = mktime(&time_var);
    struct timeval tset = {.tv_sec = tt};

    settimeofday(&tset, NULL);
```

I have found that if I set the tz *after* I settimeofday() everything seems to work fine. If I set it before it doesn't

I will however try to reproduce the error with your gist.

igrr, 2023-03-01:
Ah, the behavior you observe makes sense then. The thing is that mktime assumes that the input is the local time, not UTC time. So either you need to set TZ to UTC time when calling mktime, or you need to adjust the returned value by the timezone offset.

Unfortunately the C standard library doesn't provide a UTC equivalent of mktime.

mbastida123, 2023-03-02:
> Ah, the behavior you observe makes sense then. The thing is that mktime assumes that the input is the local time, not UTC time. So either you need to set TZ to UTC time when calling mktime, or you [...]

Then how can I set UTC, call mktime and reset the correct tz?, like this?

```
    struct tm time_var;
    time_var = pcf8523_get_time();

    setenv("TZ", "GMT", 1);
    tzset();
    time_t tt = mktime(&time_var);
    struct timeval tset = {.tv_sec = tt};

    settimeofday(&tset, NULL);
    setenv("TZ", "CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00", 1);
    tzset();
```

Because it doesn't work

mbastida123, 2023-03-02:
@igrr I've done a little research:

The c function I would need is `time_t timegm(struct tm *tm);` but, as I think you mentioned, is not available in ESP-IDF.

When we read the time from the external RTC we obtain the time in a struct tm. This variable is in UTC format. So the error comes when we later make mktime(). 

It seems like all the solutions I have found are a little bit sketchy....

igrr, 2023-03-02:
You are right, timegm is not yet supported in Newlib C library used by IDF. There was an attempt to add it, but the patch wasn't accepted yet: https://sourceware.org/legacy-ml/newlib/2018/msg00811.html.

However the approach with setting timezone to GMT works as expected on my end, I've updated the example: https://gist.github.com/igrr/d7db8a78170bf6981f2e606b42c4361c, please give it a try.

mbastida123, 2023-03-02:
@igrr your example doesn't work. I mean, it works, but it won't work in real life.

To demonstrate what I'm saying add the following two lines between lines 12 and 13.
```
setenv("TZ", "CEST-1CET,M3.2.0/2:00:00,M11.1.0/2:00:00", 1);
tzset();
```

igrr, 2023-03-02:
I'm sorry, of course you are right. Here is the updated one which works: https://gist.github.com/igrr/d7db8a78170bf6981f2e606b42c4361c

To switch to UTC time we have to use `GMT0` or any other `XXX0` timezone string, to be compliant with the format expected by the C library: https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html.

Specifying simply `GMT` works with glibc library but not with newlib, since newlib doesn't have a timezone database to convert the well-known names into timezone rules. When we specify `GMT0`, then newlib can get the `0` offset from the TZ rule.

mbastida123, 2023-03-02:
Thanks @igrr !!!

Here's another workaround if it's useful to anyone else: https://stackoverflow.com/questions/48566198/how-to-convert-utc-date-to-time-t

closed 2023-03-02 by mbastida123