espressif/esp-idf #10882: PWM not generate in ESP32S3

CC-MASTER05, 2023-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Able to Measure Voltage on GPIO as PWM value change.

### What is the actual behavior?

Able to get PWM cycle but didn't get any voltage in pin.

### Steps to reproduce.

1. clone esp-rainmaker parallel to esp-idf 
2. go to example/ led_light/ 
3. change GPIO pin (by-default 48) to GPIO 18 using idf.py menuconfig -> Component config -> WS2812 RGB LED
4. idf.py build
5. idf.py -p (port) flash monitor
6. use DMM to mesure voltage on PIN 18

### Build or installation Logs.

```plain
I (32) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (32) boot: compile time 10:51:56
I (33) boot: chip revision: v0.1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : SLOW READ
I (51) boot.esp32s3: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (80) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (87) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (95) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (102) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (110) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (117) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (125) boot: End of partition table
I (129) boot: No factory image, trying OTA 0
I (134) boot_comm: chip revision: 1, min. application chip revision: 0
I (141) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=3b914h (243988) map
I (208) esp_image: segment 1: paddr=0005b93c vaddr=3fc9a730 size=046dch ( 18140) load
I (213) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=e2460h (926816) map
I (434) esp_image: segment 3: paddr=00142488 vaddr=3fc9ee0c size=00664h (  1636) load
I (434) esp_image: segment 4: paddr=00142af4 vaddr=40374000 size=16730h ( 91952) load
I (474) boot: Loaded app from partition at offset 0x20000
I (499) boot: Set actual ota_seq=1 in otadata[0]
I (499) boot: Disabling RNG early entropy source...
I (510) cpu_start: Pro cpu up.
I (510) cpu_start: Starting app cpu, entry point is 0x403754c4
I (0) cpu_start: App cpu up.
I (524) cpu_start: Pro cpu start user code
I (524) cpu_start: cpu freq: 160000000
I (524) cpu_start: Application information:
I (527) cpu_start: Project name:     led_light
I (532) cpu_start: App version:      1.0
I (536) cpu_start: Compile time:     Mar  2 2023 10:56:31
I (542) cpu_start: ELF file SHA256:  637e8dc7bba5481c...
I (548) cpu_start: ESP-IDF:          v4.4.4
I (553) heap_init: Initializing. RAM available for dynamic allocation:
I (560) heap_init: At 3FCA5648 len 000440C8 (272 KiB): D/IRAM
I (567) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (574) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (580) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (586) spi_flash: detected chip: mxic (opi)
I (591) spi_flash: flash io: opi_str
W (595) spi_flash: Detected size(32768k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (609) sleep: Configure to isolate all GPIO pins in sleep state
I (615) sleep: Enable automatic switching of GPIO sleep configuration
I (622) coexist: coexist rom version e7ae62f
I (628) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (668) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (708) pp: pp rom version: e7ae62f
I (708) net80211: net80211 rom version: e7ae62f
I (718) wifi:wifi driver task: 3fcedbe8, prio:23, stack:6656, core=0
I (718) system_api: Base MAC address is not set
I (718) system_api: read default base MAC address from EFUSE
I (728) wifi:wifi firmware version: 6567a16
I (728) wifi:wifi certification version: v7.0
I (728) wifi:config NVS flash: enabled
I (738) wifi:config nano formating: disabled
I (738) wifi:Init data frame dynamic rx buffer num: 32
I (738) wifi:Init management frame dynamic rx buffer num: 32
I (748) wifi:Init management short buffer num: 32
I (748) wifi:Init dynamic tx buffer num: 32
I (758) wifi:Init static tx FG buffer num: 2
I (758) wifi:Init static rx buffer size: 1600
I (768) wifi:Init static rx buffer num: 10
I (768) wifi:Init dynamic rx buffer num: 32
I (768) wifi_init: rx ba win: 6
I (778) wifi_init: tcpip mbox: 32
I (778) wifi_init: udp mbox: 6
I (778) wifi_init: tcp mbox: 6
I (788) wifi_init: tcp tx win: 5744
I (788) wifi_init: tcp rx win: 5744
I (798) wifi_init: tcp mss: 1440
I (798) wifi_init: WiFi IRAM OP enabled
I (808) wifi_init: WiFi RX IRAM OP enabled
I (818) esp_rmaker_work_queue: Work Queue created.
I (838) esp_mqtt_glue: AWS PPI: ?Platform=APN3|A0|RM|EX00|RMDev|1x0|68B6B33CD3D8
I (838) esp_mqtt_glue: Initialising MQTT
I (838) esp_rmaker_mqtt_budget: MQTT Budgeting initialised. Default: 100, Max: 1024, Revive count: 1, Revive period: 5
I (848) esp_rmaker_node: Node ID ----- 68B6B33CD3D8
I (858) esp_rmaker_ota_using_topics: OTA enabled with Topics
I (858) esp_rmaker_ota: OTA state = 2
I (868) esp_rmaker_time_service: Time service enabled
I (868) esp_rmaker_time: Initializing SNTP. Using the SNTP server: pool.ntp.org
I (878) app_insights: Enable CONFIG_ESP_INSIGHTS_ENABLED to get Insights.
I (888) esp_rmaker_core: Starting RainMaker Work Queue task
I (888) esp_rmaker_work_queue: RainMaker Work Queue task started.
I (888) wifi_prov_scheme_ble: BT memory released
I (908) app_wifi: Already provisioned, starting Wi-Fi STA
I (908) wifi_prov_scheme_ble: BTDM memory released
I (918) esp_rmaker_local: Event 6
I (918) phy_init: phy_version 540,a5d905b,Oct 20 2022,19:36:11
I (968) wifi:mode : sta (68:b6:b3:3c:d3:d8)
I (968) wifi:enable tsf
I (968) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (988) wifi:state: init -> auth (b0)
I (988) wifi:state: auth -> assoc (0)
I (998) wifi:state: assoc -> run (10)
I (1068) wifi:connected with NetweeN, aid = 2, channel 11, BW20, bssid = 90:8d:78:76:27:00
I (1068) wifi:security: WPA2-PSK, phy: bgn, rssi: -52
I (1078) wifi:pm start, type: 1

I (1078) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (1088) wifi:<ba-add>idx:0 (ifx:0, 90:8d:78:76:27:00), tid:0, ssn:2, winSize:64
I (1138) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2088) app_wifi: Connected with IP Address:192.168.0.4
I (2088) esp_netif_handlers: sta ip: 192.168.0.4, mask: 255.255.255.0, gw: 192.168.0.1
I (2088) esp_rmaker_local: Starting ESP Local control with HTTP Transport and security version: 1
I (2118) esp_https_server: Starting server
I (2118) esp_https_server: Server listening on port 8080
I (2118) esp_rmaker_local: esp_local_ctrl service started with name : 68B6B33CD3D8
I (2118) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (2128) esp_rmaker_core: Waiting for MQTT connection
I (3168) esp_rmaker_time: SNTP Synchronised.
I (3168) wifi:<ba-add>idx:1 (ifx:0, 90:8d:78:76:27:00), tid:1, ssn:0, winSize:64
I (3168) esp_rmaker_time: The current time is: Thu Mar  2 11:54:56 2023 +0530[IST], DST: No.
I (3818) esp-x509-crt-bundle: Certificate validated
I (5648) esp_mqtt_glue: MQTT Connected
I (5648) esp_rmaker_node_config: Reporting Node Configuration of length 1903 bytes.
I (5668) esp_rmaker_param: Params MQTT Init done.
I (5668) esp_rmaker_param: Reporting params (init): {"Light":{"Name":"Light","Power":true,"Brightness":25,"Hue":180,"Saturation":100},"Time":{"TZ":"Asia/Kolkata","TZ-POSIX":"IST-5:30"},"Schedule":{"Schedules":[]},"Scenes":{"Scenes":[]},"Local Control":{"POP":"abf70837","Type":1}}
I (5688) esp_rmaker_cmd_resp: Enabling Command-Response Module.
I (5688) esp_rmaker_ota_using_topics: Subscribing to: node/68B6B33CD3D8/otaurl
I (10698) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
```

### More Information.

build the application and push into Board  as we have changed the pin 48 to 18 so as per our setup we need to measure some voltage on pin 18.

As per Oscilloscope i get PWM cycle but not able to measure voltage throw Multi-meter.

Provide voltage : 5 V through USB cable

suda-morris, 2023-03-02:
If your WS2812 LED can light up and give you the correct color, then, no issue at all. WRT why you can't see the voltage change by your multi-meter, you may need to read WS2812 datasheet first. The IO protocol is not a simple **PWM** at all.

CC-MASTER05, 2023-03-02:
Thanks for the answer but actually i am able to measure voltage in default pin. actually issue is there when i try to change my pin at a time led doesn't blink as well as those pin not generate any voltage i have measure voltage on pin using multi-meter but didn't get any voltage.

CC-MASTER05, 2023-03-02:
I have taken someone else code 
https://github.com/nopnop2002/esp-idf-pwm-slider/blob/main/led-control/main/

and test with the pin 18 it working fine but in our rainmaker sdk example in not able to work example when i change GPIO

suda-morris, 2023-03-02:
do you mean gpio18 is not working on your board? maybe you need to check if GPIO18 is used by another purpose in your project.

suda-morris, 2023-03-02:
or you can try the [blink example](https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink) to narrow down the issue is with your board or with your rainmaker or with the idf driver.

CC-MASTER05, 2023-03-03:
Hi,
can you help to resolve?

CC-MASTER05, 2023-04-08:
@suda-morris  issue is resolve after changing SDK version of ESp-IDF.

Thanks for the support!!!

Alvin1Zhang, 2023-04-24:
Thanks for the updates, feel free to reopen.

closed 2023-04-24 by Alvin1Zhang