espressif/esp-idf #10883: ESP32 hangs on Connecting....... when trying to flash the device

krupis, 2023-03-02:
Hello. We have designed our product with ESP32 chip and we have been manufacturing our devices for a while now. After all components are soldered, the devices goes through testing where it is being flashed with the production firmware and hardware is tested. Everything has been going well and we didint have any problems since recently.

We have ordered a new batch of ESP32 chips (same chips just different batch), we have noticed that something is not right when flashing those devices. Initial device flash takes about 1 minute or even longer as the device is getting stuck on Connecting....... After about 40seconds - 1minute after being stuck in Connecting..., the device wakes up and starts flashing..
[image]

Once the initial flash, any subsequent flashes are working as expected (flashing immediately).

We are trying to understand what could be the reason the first flashing takes so long. Maybe you have any ideas?

We didint have any simillar problems with the last batch of our esp32 chips.

krupis, 2023-03-02:
1.  We are using esp modules.
2.  Yes which is why it is so strange. The old batch of devices (I think we already tested more than a few thousand did not have any issues  with flashing and design hasnt change since then).
My initial idea was that maybe there was something in the CPU memory (some program) that does not allow the enable/boot pins to function correctly. Because once the flash is erased (with idf.py erase-flash or some different program is flashed even "hello world"). Any subsequent flashes are instant.

I will try using the updated esptool

krupis, 2023-03-03:
We have tried multiple different PC's and a couple of different CP (USB->Serial bridges). But all have identical issues on those modules. We have a feeling that there are differences in modules from the old batch and the new batch even though markings are identical on both.

 Is there any way to debug this further? Is there anything we can measure/read to help figure out what could be issue? I can even send over a couple of those "faulty" modules over if that helps since I really want to get to the bottom of this issue.

radimkarnis, 2023-03-06:
@krupis thank you for trying my suggestions.
We can try comparing the modules. Please, for both the working and faulty ones:

1) Run and post results of the `esptool.py flash_id` command
2) Run and post results of the `espefuse.py -p PORT summary` command 
3) Dump the flash of the faulty module: `esptool.py -b 460800 read_flash 0 0x200000 dump.bin` - to read 2MB (you can increase the baud rate to read faster). Send the bin dump here, I will try to reproduce the issue.

krupis, 2023-03-06:
@radimkarnis  Hello. I did tests as you described. github does not allow me to send .bin files here hence I am sending wetransfer link where you can download dump files.

I am sending raw text as well as .txt files of the results

**GOOD MODULE**

```
PS C:\Users\gabrielaitis.li\esp\esp-idf\examples\get-started\hello_world> C:\Users\gabrielaitis.li\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\gabrielaitis.li\esp\esp-idf\components\esptool_py\esptool\esptool.py 
-p COM18 flash_id
>>
esptool.py v3.2-dev
Serial port COM18
Connecting.........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting..........
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 90:38:0c:5d:e7:e4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4017
Detected flash size: 8MB
Hard resetting via RTS pin...                                             C:\Users\gabrielaitis.li\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\gabrielaitis.li\esp\esp-idf\components\esptool_py\esptool\espefuse.py -p COM18 summaryelaitis.li\esp\esp-idf\examples\get-started\hello_world>
Connecting...................
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.........
Detecting chip type... ESP32
espefuse.py v3.2-dev
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)   
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1121 R/W (0b00011)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)   
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)   
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)    
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 51 R/W (0x33)     
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)   
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 0 R/W (0x0000)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 0 R/W (0x0)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address
   = 90:38:0c:5d:e7:e4 (CRC 0xe1 OK) R/W
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 225 R/W (0xe1)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                             Reserved for future chip versions                  = 2 R/W (0b10)
image     = False R/W (0b0)                                                                           = 1 R/W (0b001)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = 0 R/W (0x00)
          = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
[... 144 lines cut ...]
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
2097152 (100 %)
2097152 (100 %)
Read 2097152 bytes at 0x0 in 49.0 seconds (342.1 kbit/s)...
Hard resetting via RTS pin...
PS C:\Users\gabrielaitis.li\esp\esp-idf\examples\get-started\hello_world>
```

The dump.bin of bad module can be found here:
https://we.tl/t-C690z9fDBL

**IMPORTANT**

When we execute these 3 commands on a "good" module, the response is almost immediate. The whole test (3 commands) took no longer than a minute probably.

[... cut ...]  the same way it does during initial flashing.  The whole test (3 commands) probably took more than 5-6 minutes. It almost feels like in order to "fix" the module from being stuck on "Connecting", is to flash or erase the whole flash at least once.

Please let me know if I can help any further!

[ESP32_normal_flash.txt](https://github.com/espressif/esp-idf/files/10896483/ESP32_normal_flash.txt)
[ESP32_delayed_flash.txt](https://github.com/espressif/esp-idf/files/10896484/ESP32_delayed_flash.txt)

radimkarnis, 2023-03-06:
Thanks for the information and the dumps! 

Could you please take a faulty module and:
1) Verify the issue happens,
2) erase the flash and verify the issue is gone,
3) reflash the original contents (`esptool.py write_flash 0 dump2.bin`) and see if the issue starts happening again?

Thank you!

krupis, 2023-03-06:
@radimkarnis 

Hi again. I have performed steps as you described above:

1. Picked up faulty module
2. Erased the flash completely. The first erase_flash hanged on "Conneting..." and it took about 2 minutes to continue. After erasing flash once, I have tried to erase it again, 2nd time was immediate so I can verify that the issue is gone.
3. Reflashed the original contents with the command that you have suggested:
```
**esptool.py write_flash 0 dump2.bin**
```

I can confirm that the issue started happening again. I tried to erase flash and it hanged on "Connecting....".

Does that mean that flash contents (whatever has been flashed to the modules before they have been sent to us from the distributor) somehow have an effect? How can this be?

radimkarnis, 2023-03-06:
> Does that mean that flash contents (whatever has been flashed to the modules before they have been sent to us from the distributor) somehow have an effect? How can this be?

That's what I wanted to verify and now we have a confirmation. So yes, the flash contents are what's causing your troubles.
I have checked `dump2.bin` and it looks like a simple app that just prints `OHAI` to the console and resets. This might be some kind of a factory test or integrity-check.

I have no idea how this is possible or why it would influence the reset process, though.

> Did you try to flash this dump2.bin to some random ESP32 device or devboard to check if you are able to reproduce?

Yes, I have tried it on several ESP32 devices, but can't reproduce it. Even with `dump2.bin` flashed into the flash memory, everything works fine here.

krupis, 2023-03-06:
@radimkarnis 
First of all, thank you very much for your assitance!
I tried to flash ```dump2.bin``` to esp32 devkit and as you said - everything seems to work fine.

We will be further investigating our hardware, try to swap around components between new batch and old batch and try to figure out what could be an issue. 
If we are not able to reproduce the issue with the devboards, I can only assume that the only thing that is left to blame is our hardware.

I will post an update here.

radimkarnis, 2023-03-06:
You can try to flash `dump2.bin` to one of the old batch modules and see if the issue appears. If it does - we can rule out any HW differences between the old and new ESP32 modules. Then it's probably a combination of the binary and your hardware design.

Please keep me informed! This is a very interesting issue.

krupis, 2023-03-09:
@radimkarnis 
We found out what was causing the delayed flashing issue, it was a hardware issue. Although we don't understand how it effects the flashing..

The following pins are used in our hardware:

GPIO0 is floating, connected to external UART/USB bridge is controlled LOW/HIGH;
GPIO2 is pulled LOW, connected to NPN transistor controlling HEAT relay;
GPIO4 is pulled LOW, connected to NPN transistor controlling COMP relay;
GPIO5 is pulled HIGH, connected to external EEPROM Chip Select pin;
GPIO12 is pulled LOW, connected to voltage divider USB +5V. Detecting when the USB is plugged in or not;
GPIO15 is pulled HIGH, connected to NPN transistor controlling FAN relay;

[image]

The old batch had these **MUN5214T1G** transistors with internal resistors: 10k on base (input), and 47k on emitter (ground).

The new batch uses **MUN5235T1G** transistors with internal resistors: 2.2k on base (input), and 47k on emitter (ground). 

Somehow, the new batch transistor **MUN5235T1G** that is controlling the GPIO 15 relay is causing the issues. If we replace this transistor with the old batch **MUN5214T1G** , the problem goes away. 

Maybe you have some ideas why would this transistor cause delayed flashing issues? The only difference seems to internal base resistance.

radimkarnis, 2023-03-10:
@krupis that's interesting. So there also was a change in your design, which together with the pre-loaded factory binary resulted in this issue. Therefore, if you load the `dump.bin` into the old batch (with the `MUN5214T1G` transistor), the issue doesn't happen?

GPIO15 is used to [silence boot messages printed by the ROM bootloader](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#other-pins), but that shouldn't make any difference in this case.

I am sorry, but am not the right person to discuss the issue on a hardware level. I can recommend you our [free Circuit Schematic & PCB Design Review service](https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review). An experienced team of HW engineers should be able to help if you provide enough information!

krupis, 2023-03-10:
@radimkarnis
Since this was almost identical component, we were not considering this as a change in a design at first.

> Therefore, if you load the `dump.bin` into the old batch (with the `MUN5214T1G` transistor), the issue doesn't happen?

That is correct. We are not able to represent the issue with **MUN5214T1G** at all. We tried to erase flash multiple times, try with dump.bin and dump2.bin but the device flashing is instant.

As soon as we desolder this component and add **MUN5235T1G**, the issue starts happening. 

Thank you very much for recommending PCB design review service. We will try and contact them regarding this

radimkarnis, 2023-03-10:
@krupis 
Thanks for your patience and all the information you've provided.

I think there's not much more I can do for you, can we close this ticket now? Feel free to reopen it if you get some more info from the PCB review service or if you figure out something new yourself.

krupis, 2023-03-10:
@radimkarnis 
Thanks for your assistance. I will close this for time being then.

krupis, 2023-03-10:
@radimkarnis  We havent had a chance to contact the PCB design review service that you recommended yet. We did some more troubleshooting and found something that we thought was interesting:

[image]

The old batch transistor had a base resistance of 10kOhm.GPIO 15 is pulled up via 10kOhm resistor as you can see from the image above.

Since the new batch transistor base resistance is 2.2kOhm, we have decided to replace the GPIO 15 pull up resistor to match the transistor base resistance and put a 2.2kOhm resistor instead of 10kOhm. Strangely enough, this fixed the issue. After replacing the resistor, the device is flashing properly even with the new transistor **MUN5235T1G**

srinietc, 2025-02-06:
> Thanks for the information and the dumps!
> 
> Could you please take a faulty module and:
> 
> 1. Verify the issue happens,
> 2. erase the flash and verify the issue is gone,
> 3. reflash the original contents (`esptool.py write_flash 0 dump2.bin`) and see if the issue starts happening again?
> 
> Thank you!

Thank you @radimkarnis. I had the same issue, pulling my hair out for the past 3 days trying to put my EPS32-C3 into download mode. No matter what it just wouldn't. Per your post, I erased the flash, bingo... Arduino IDE flashed it like a charm.

Thanks a lot man!