espressif/esp-idf #10885: 'clock_gettime()' function doesn't return the time with us accuracy

marco711, 2023-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I want to get an Epoch timestamp by using `clock_gettime()` function. Nevertheless, when I print the result I only get a second accuracy:

```
[2023-03-02_11:01:33:33.865073] time_us: 21000000
[2023-03-02_11:01:34:34.165139] time_us: 21000000
[2023-03-02_11:01:34:34.465992] time_us: 22000000
[2023-03-02_11:01:34:34.765256] time_us: 22000000
[2023-03-02_11:01:35:35.065876] time_us: 22000000
[2023-03-02_11:01:35:35.365351] time_us: 23000000
[2023-03-02_11:01:35:35.665258] time_us: 23000000
[2023-03-02_11:01:35:35.966376] time_us: 23000000
[2023-03-02_11:01:36:36.265882] time_us: 24000000
[2023-03-02_11:01:36:36.565201] time_us: 24000000
[2023-03-02_11:01:36:36.865079] time_us: 24000000
[2023-03-02_11:01:37:37.165179] time_us: 24000000
[2023-03-02_11:01:37:37.465581] time_us: 25000000
```

This is the code I'm using:

```
clock_gettime(CLOCK_REALTIME, &tv_now);
int64_t time_us = (int64_t)tv_now.tv_sec * 1000000L + (int64_t)tv_now.tv_usec;
printf("time_us: %ld\n", time_us);
```
As clock source I'm using the internal 136 kHz oscillator. Could this be the reason why I get this accuracy?

igrr, 2023-03-02:
Could you please provide the rest of the information from the bug report issue template: IDF version (commit) ID, sdkconfig file?

The code snippet you have posted looks odd, because:
* clock_gettime function accepts a pointer to `struct timespec`
* but `struct timespec` doesn't have a member `tv_usec`. `tv_usec` is a memeber of a different structure, `struct timeval`. On the other hand, `struct timespec` has `tv_nsec`.

Please try the following code instead:
```c++
        struct timespec tv_now;
        clock_gettime(CLOCK_REALTIME, &tv_now);
        int64_t time_us = (int64_t)tv_now.tv_sec * 1000000L + (int64_t)tv_now.tv_nsec / 1000;
        printf("time_us: %lld\n", time_us);
```

marco711, 2023-03-02:
I have modified the first post to provide the rest of the information.
I have modified the code according to your suggestion (the type of `tv_now` was wrong as you indicated). Now it seems to work fine:
```
[2023-03-02_12:00:55:55.769797] time_us: 26273548
[2023-03-02_12:00:56:56.069833] time_us: 26573393
[2023-03-02_12:00:56:56.369436] time_us: 26873552
[2023-03-02_12:00:56:56.669293] time_us: 27173421
[2023-03-02_12:00:56:56.969435] time_us: 27473478
[2023-03-02_12:00:57:57.269677] time_us: 27773443
[2023-03-02_12:00:57:57.569212] time_us: 28073313
[2023-03-02_12:00:57:57.869938] time_us: 28373304
[2023-03-02_12:00:58:58.169406] time_us: 28673298
[2023-03-02_12:00:58:58.469217] time_us: 28973315
[2023-03-02_12:00:58:58.769280] time_us: 29273314
[2023-03-02_12:00:59:59.069151] time_us: 29573279
[2023-03-02_12:00:59:59.369256] time_us: 29873308
[2023-03-02_12:00:59:59.670175] time_us: 30173355
[2023-03-02_12:00:59:59.969411] time_us: 30473490
[2023-03-02_12:01:00:00.269352] time_us: 30773500
[2023-03-02_12:01:00:00.569446] time_us: 31073321
[2023-03-02_12:01:00:00.869888] time_us: 31373565
[2023-03-02_12:01:01:01.169309] time_us: 31673428
[2023-03-02_12:01:01:01.469293] time_us: 31973452
[2023-03-02_12:01:01:01.769922] time_us: 32273460
[2023-03-02_12:01:02:02.069148] time_us: 32573307
```

Nevertheless, sometimes the microcontroller stops working right after the first `time_us` print.

igrr, 2023-03-02:
Good to know that you now get the expected result from clock_gettime!

> Nevertheless, sometimes the microcontroller stops working right after the first time_us print.

You might be running into an issue fixed by this PR: https://github.com/espressif/esp-idf/pull/10649. The version you are using (02605f1a31) doesn't include that fix yet.

marco711, 2023-03-02:
How do I update the IDF version? I'm following this procedure (https://docs.espressif.com/projects/esp-idf/en/release-v3.2/get-started/index.html#updating-esp-idf) but I get the following result:
```
$ git clone -b release/v5.1 --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
fatal: Remote branch release/v5.1 not found in upstream origin
```

igrr, 2023-03-02:
There is no release/v5.1 branch in IDF yet, therefore you get the error.

It looks like you have cloned the `master` branch of esp-idf at some point. You can go into your IDF project and try to pull the latest `master` branch.

The docs for updating IDF are here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-master. Just be aware, if you have done local changes in IDF repo, you need to `git stash` them first.

marco711, 2023-03-02:
Is this the latest version?
ESP-IDF v5.1-dev-3710-gacac972f70

david-cermak, 2023-03-02:
@marco711 There's also a note about it in the [v5.1 Migration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.1/networking.html#sntp)

Quick fix is to include `esp_sntp.h` and use `esp_` prefixed API:
```patch
-#include "lwip/apps/sntp.h"
+#include "esp_sntp.h"

...

-sntp_stop();
+esp_sntp_stop();
```

Alternatively (preferably) you can use esp_netif's [SNTP API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#esp-netif-sntp-api)

marco711, 2023-03-02:
Thank you. I have updated to ESP-IDF v5.1-dev-3710-gacac972f70 but the microcontroller still blocks sometimes after th  first `time_us` print.

igrr, 2023-03-02:
@marco711 Could you please attach your `sdkconfig` file, as well as try to isolate the piece of code which reproduces the blocking issue and which you can share with us?

marco711, 2023-03-03:
Attached you will find the `sdkconfig` file:
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/10880112/sdkconfig.zip)

Below you will find the log with the line (in bold) where the microcontroller blocks:

```
$ bash monitor_esp32s3.sh 
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[2023-03-03_09:58:04:04.785507] ESP-ROM:esp32s3-20210327
[2023-03-03_09:58:04:04.792231] Build:Mar 27 2021
[2023-03-03_09:58:04:04.797241] rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
[2023-03-03_09:58:04:04.797367] SPIWP:0xee
[2023-03-03_09:58:04:04.797420] mode:DIO, clock div:1
[2023-03-03_09:58:04:04.802823] load:0x3fce3818,len:0x1678
[2023-03-03_09:58:04:04.802940] load:0x403c9700,len:0x4
[2023-03-03_09:58:04:04.802990] load:0x403c9704,len:0xbe8
[2023-03-03_09:58:04:04.808759] load:0x403cc700,len:0x2e8c
[2023-03-03_09:58:04:04.808842] entry 0x403c9904
[2023-03-03_09:58:04:04.814168] I (27) boot: ESP-IDF v5.1-dev-3710-gacac972f70 2nd stage bootloader
[2023-03-03_09:58:04:04.819764] I (27) boot: compile time Mar  2 2023 13:55:02
[2023-03-03_09:58:04:04.825871] I (28) boot: chip revision: v0.1
[2023-03-03_09:58:04:04.831421] I (31) boot.esp32s3: Boot SPI Speed : 80MHz
[2023-03-03_09:58:04:04.836592] I (36) boot.esp32s3: SPI Mode       : DIO
[2023-03-03_09:58:04:04.842124] I (41) boot.esp32s3: SPI Flash Size : 2MB
[2023-03-03_09:58:04:04.842206] I (46) boot: Enabling RNG early entropy source...
[2023-03-03_09:58:04:04.847527] I (51) boot: Partition Table:
[2023-03-03_09:58:04:04.853782] I (55) boot: ## Label            Usage          Type ST Offset   Length
[2023-03-03_09:58:04:04.864168] I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
[2023-03-03_09:58:04:04.870075] I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[2023-03-03_09:58:04:04.875237] I (77) boot:  2 factory          factory app      00 00 00010000 00177000
[2023-03-03_09:58:04:04.880732] I (84) boot: End of partition table
[2023-03-03_09:58:04:04.889884] I (89) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=3b1c4h (242116) map
[2023-03-03_09:58:04:04.931331] I (140) esp_image: segment 1: paddr=0004b1ec vaddr=3fc9a500 size=04bb0h ( 19376) load
[2023-03-03_09:58:04:04.942496] I (145) esp_image: segment 2: paddr=0004fda4 vaddr=40374000 size=00274h (   628) load
[2023-03-03_09:58:04:04.947976] I (146) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=d88c8h (886984) map
[2023-03-03_09:58:05:05.103937] I (314) esp_image: segment 4: paddr=001288f0 vaddr=40374274 size=161c4h ( 90564) load
[2023-03-03_09:58:05:05.136147] I (343) boot: Loaded app from partition at offset 0x10000
[2023-03-03_09:58:05:05.140999] I (344) boot: Disabling RNG early entropy source...
[2023-03-03_09:58:05:05.141240] I (356) cpu_start: Pro cpu up.
[2023-03-03_09:58:05:05.146184] I (356) cpu_start: Starting app cpu, entry point is 0x403755c4
0x403755c4: call_start_cpu1 at /home/msurca/esp/esp-idf/components/esp_system/port/cpu_start.c:152

[2023-03-03_09:58:05:05.156566] I (0) cpu_start: App cpu up.
[2023-03-03_09:58:05:05.157709] I (370) cpu_start: Pro cpu start user code
[2023-03-03_09:58:05:05.164002] I (370) cpu_start: cpu freq: 160000000 Hz
[2023-03-03_09:58:05:05.169145] I (370) cpu_start: Application information:
[2023-03-03_09:58:05:05.174599] I (373) cpu_start: Project name:     iot_lift
[2023-03-03_09:58:05:05.174673] I (378) cpu_start: App version:      1
[2023-03-03_09:58:05:05.179899] I (383) cpu_start: Compile time:     Mar  2 2023 13:54:52
[2023-03-03_09:58:05:05.191308] I (389) cpu_start: ELF file SHA256:  99afe6536c4f9349...
[2023-03-03_09:58:05:05.231568] I (395) cpu_start: ESP-IDF:          v5.1-dev-3710-gacac972f70
[2023-03-03_09:58:05:05.231909] I (401) cpu_start: Min chip rev:     v0.0
[2023-03-03_09:58:05:05.231945] I (406) cpu_start: Max chip rev:     v0.99 
[2023-03-03_09:58:05:05.232242] I (411) cpu_start: Chip rev:         v0.1
[2023-03-03_09:58:05:05.232750] I (416) heap_init: Initializing. RAM available for dynamic allocation:
[2023-03-03_09:58:05:05.233010] I (423) heap_init: At 3FCBF798 len 00029F78 (167 KiB): DRAM
[2023-03-03_09:58:05:05.233267] I (429) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[2023-03-03_09:58:05:05.235359] I (436) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
[2023-03-03_09:58:05:05.241128] I (443) spi_flash: detected chip: generic
[2023-03-03_09:58:05:05.247263] I (446) spi_flash: flash io: dio
[2023-03-03_09:58:05:05.257866] W (450) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
[2023-03-03_09:58:05:05.263757] I (464) sleep: Configure to isolate all GPIO pins in sleep state
[2023-03-03_09:58:05:05.269121] I (470) sleep: Enable automatic switching of GPIO sleep configuration
[2023-03-03_09:58:05:05.274699] I (478) coexist: coex firmware version: 9747c21
[2023-03-03_09:58:05:05.279914] I (483) coexist: coexist rom version e7ae62f
[2023-03-03_09:58:05:05.285433] I (488) app_start: Starting scheduler on CPU0
[2023-03-03_09:58:05:05.291162] I (493) app_start: Starting scheduler on CPU1
[2023-03-03_09:58:05:05.297247] I (493) main_task: Started on CPU0
[2023-03-03_09:58:05:05.298774] I (503) main_task: Calling app_main()
[... 216 lines cut ...]
						messagepack_data_offline_packet_flag[sent_message_failed_count] = 1;
					}
					else
					{
[... cut ...]
							memset(messagepack_data_offline_packet_flag, 0, sizeof(messagepack_data_offline_packet_flag));
						}
						gettimeofday(&test_time_read, NULL);

						int clk_result = clock_gettime(CLOCK_REALTIME, &tv_now);
						int64_t time_us = (int64_t)tv_now.tv_sec * 1000000L + (int64_t)tv_now.tv_nsec / 1000;
						printf("time_us: %lld\n", time_us); /* Sometimes this line is called only once and the microcontroller blocks after the **MQTT_EVENT_UNSUBSCRIBED**. */
					}
				}
			}
		}
		vTaskDelay(100 / portTICK_PERIOD_MS);
	}
}
```

igrr, 2023-03-03:
Thank you for the extra details.
Unfortuntely the piece of code you have shared seems to have dependencies on some other parts of your application, so I couldn't compile and run it.
Could you try pausing the program in the debugger when it blocks? Then you should be able to observe the backtrace.

marco711, 2023-03-03:
I cannot connect with the debugger:

```
$ xtensa-esp32-elf-gdb -x gdbinit build/iot_lift.elf 
GNU gdb (esp-gdb) 12.1_20221002
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/iot_lift.elf...
gdbinit:1: Error in sourced command file:
:3333: Connection timed out.
(gdb) 
```

Below my gdbinit file:

```
target remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
```

Is there any port other than 3333 that I can try?

igrr, 2023-03-03:
Is OpenOCD running when you are starting GDB? It looks like not, because GDB isn't able to connect to it.

If you are working from the command line, you can start it like this:
1. in one console/terminal, run `idf.py flash openocd monitor` — OpenOCD and serial monitor will start. You can also launch it without `monitor`, it's there just so that you can see serial output as well.
2. in another console, run `idf.py gdb` — this will launch GDB, which will connect to OpenOCD.

marco711, 2023-03-03:
I cannot launch OpenOCD:
```
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.1 seconds (effective 262.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32s3-builtin.cfg"
OpenOCD started as a background task 466673
Executing action: post_debug
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!

"openocd" exited with 1
HINT: Please check the wire connection to debugging device or access rights to a serial port.
Done
```
I'm trying to run OpenOCD with `openocd -f board/esp32s3-builtin.cfg` but the port is not found:
```
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
```
Is there any configuration I'm missing?

igrr, 2023-03-03:
Could you please check, what does `lsusb` say? Which device appears in the system when you plug in the board?

Maybe you aren't connecting the board over the built-in USB port and are using some USB-Serial converter, instead. Which development board are you using, by the way? On the ESP32-S3-DevKit-C there are two ports, you need the one which has "USB" written next to it ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html#getting-started)).

marco711, 2023-03-03:
```
f$ lsusb
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 04f2:b449 Chicony Electronics Co., Ltd Integrated Camera
Bus 002 Device 010: ID 03f0:134a HP, Inc Optical Mouse
Bus 002 Device 009: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 002 Device 008: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Yes, I'm connecting over the port marked as "USB" and not "UART".

igrr, 2023-03-03:
That looks pretty odd! Which of those devices disappears when you unplug the board?
The built-in USB_SERIAL_JTAG should show up with VID:PID 0x303a:0x1001. If you don't see it, maybe there is some useful output in dmesg when you plug the device in?
What's odd is that you are clearly able to "flash" and "monitor" the device, which means that USB device _should_ be present in the system...

marco711, 2023-03-03:
Right after I connect the USB port the following device appears:
`Bus 002 Device 070: ID 303a:1001 Espressif USB JTAG/serial debug unit`

Below the complete log:
```
$ lsusb
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 04f2:b449 Chicony Electronics Co., Ltd Integrated Camera
Bus 002 Device 070: ID 303a:1001 Espressif USB JTAG/serial debug unit
Bus 002 Device 010: ID 03f0:134a HP, Inc Optical Mouse
Bus 002 Device 009: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 002 Device 008: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Nevertheless, if I retry running `lsusb` after 1 or 2 minutes the device disappears:
```
$ lsusb
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 04f2:b449 Chicony Electronics Co., Ltd Integrated Camera
Bus 002 Device 010: ID 03f0:134a HP, Inc Optical Mouse
Bus 002 Device 009: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 002 Device 008: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Below the result of dmesg

```
[128857.832600] usb 2-3.4: new full-speed USB device number 75 using xhci_hcd
[128857.957894] usb 2-3.4: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[128857.957898] usb 2-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[128857.957900] usb 2-3.4: Product: USB JTAG/serial debug unit
[128857.957901] usb 2-3.4: Manufacturer: Espressif
[128857.957903] usb 2-3.4: SerialNumber: 60:55:F9:F5:29:D0
[128857.970066] cdc_acm 2-3.4:1.0: ttyACM0: USB ACM device
[128873.715286] usb 2-3-port4: disabled by hub (EMI?), re-enabling...
[128873.715506] usb 2-3.4: USB disconnect, device number 75
[128874.829212] usb 2-3-port4: Cannot enable. Maybe the USB cable is bad?
```
When I "flash" and "monitor" the device I use the UART port. Just for launching OpenOCD I have connected the "USB" port. I am using the same USB cable I use for flashing and I have also tried with another one. The result is the same when connecting to the "USB" port.

igrr, 2023-03-03:
Alright... seems like this troubleshooting direction is only getting us further away from the original problem.

> Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
> Error: esp_usb_jtag: could not find or open device!

This often indicates that there is a permission issue accessing the device. Assigning the usb device to `plugdev` group using udev rules to give yourself access to it typically solves this on Linux, IIRC.

---

Another way to debug the case when a task gets blocked is to subscribe the task to the [Task Watchdog](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#task-watchdog-timer-twdt) (`ESP_ERROR_CHECK(esp_task_wdt_add(NULL));`) and enable CONFIG_ESP_TASK_WDT_PANIC option. Then when the task gets blocked for some time, task watchdog will abort the program. This will result in the backtrace being shown, which might then give a hint about the cause of the issue.

A few other troubleshooting ideas:
* Check if the MQTT stack is overrunning the stack, enabling CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y option. If it is, you might see a panic handler message about stack watchpoint being triggered.
* Check if switching log output format to CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM changes the behavior?