espressif/esp-idf #10888: protocomm_nimble proto-ver does not work

poengAlex, 2023-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC 

### Power Supply used.

External 3.3V

### What is the expected behavior?

With Bluedroid and protocomm_set_version: 

`protocomm_set_version(pc_instance, "proto-ver", "WHATEVER");`

Sending a blank write to the characteristic "proto-ver" (0xFF53) should return the string "WHATEVER"

### What is the actual behavior?

We are trying to switch from Bluedroid to NimBLE for provisioning. Everything works as intended except for the proto-ver (0xFF53) characteristic through protocomm_set_version.

`protocomm_set_version(pc_instance, "proto-ver", "WHATEVER");`

Sending a blank write used to return the string, but with NimBLE I get this error:

`E (294712) protocomm_nimble: Outbuf with handle 12 not found
E (294712) protocomm_nimble: Failed to read characteristic with attr_handle = 12`

followed by a pairing request on the client.

### Steps to reproduce.

1. Configure Bluetooth to Nimble
2. Normal protobuf procedure
3. Read the version string
...

### More Information.

Its a run time error when trying to read the version. This works totally fine with Bluedroid, but fails with NimBLE.

IshaESP, 2023-03-06:
Hi @poengAlex,

I tested the code on my end. In wifi_prov_mgr example, the endpoints we add are registered as descriptors and not characteristics. So if you have to perform read/write, send commands on descriptors instead.

Using nrF application, read on characteristics returns 0 (for bludroid) and NULL(for nimble) .Whereas read on descriptors returns the string.

poengAlex, 2023-03-07:
@IshaESP Thanks for the reply. I have applied your patch, but the errors are still there. The ideal situation NimBLE and Bluedroid should work the same. Do you not agree?

I test it against code (not sharable) that is almost identical to the examples/provisioning/legacy/ble_prov with NimBLE

IshaESP, 2023-03-08:
Hi @poengAlex 
The errors seen are just prints. After this patch NimBLE and Bluedroid works the same. Both return 0 bytes when a characteristic is read. This [example](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) is tested using nrF application.

poengAlex, 2023-03-09:
@IshaESP : You are right. It does work, my mistake. Then the faulty behavior is only the console error prints, not the actual behavior, which your patch has solved. Thanks.

rahult-github, 2023-04-17:
Fixed with e72d223737edd89b370f80aab09167fe0954b64a . Closing the issue

closed 2023-04-17 by rahult-github