espressif/esp-idf #10889: Coredump missing information present on the backtrace

KaeLL, 2023-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF: release/v4.4

<details>
<summary>defconfig</summary>

```
CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_DETECT is not set
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_ESPTOOLPY_MONITOR_BAUD_CONSOLE=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_ota_encrypted.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=1
# CONFIG_BTDM_CTRL_MODEM_SLEEP is not set
# CONFIG_BTDM_BLE_SCAN_DUPL is not set
# CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP is not set
# CONFIG_BT_BLE_SMP_ENABLE is not set
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_ACL_CONNECTIONS=1
# CONFIG_BT_MULTI_CONNECTION_ENBALE is not set
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_BT_BLE_HOST_QUEUE_CONG_CHECK=y
CONFIG_SPI_MASTER_IN_IRAM=y
# CONFIG_SPI_SLAVE_ISR_IN_IRAM is not set
CONFIG_ESP32_REV_MIN_1=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_ULP_COPROC_ENABLED=y
CONFIG_ESP32_ULP_COPROC_RESERVE_MEM=128
# CONFIG_ESP32_DEBUG_OCDAWARE is not set
CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y
# CONFIG_ESP_ERR_TO_NAME_LOOKUP is not set
CONFIG_ETH_DMA_BUFFER_SIZE=256
CONFIG_ETH_DMA_RX_BUFFER_NUM=3
CONFIG_ETH_DMA_TX_BUFFER_NUM=3
# CONFIG_ETH_USE_SPI_ETHERNET is not set
# CONFIG_ESP_EVENT_POST_FROM_ISR is not set
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
# CONFIG_ESP_HTTP_CLIENT_ENABLE_DIGEST_AUTH is not set
CONFIG_HTTPD_MAX_REQ_HDR_LEN=2048
CONFIG_ESP_IPC_TASK_STACK_SIZE=2048
# CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE is not set
CONFIG_ESP_MAIN_TASK_STACK_SIZE=10240
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART_BAUDRATE=921600
CONFIG_ESP_INT_WDT_TIMEOUT_MS=30
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=2
CONFIG_ESP_PANIC_HANDLER_IRAM=y
# CONFIG_ESP32_WIFI_NVS_ENABLED is not set
# CONFIG_ESP32_WIFI_IRAM_OPT is not set
# CONFIG_ESP32_WIFI_RX_IRAM_OPT is not set
CONFIG_ESP_COREDUMP_ENABLE_TO_UART=y
CONFIG_ESP_COREDUMP_CHECKSUM_SHA256=y
CONFIG_ESP_COREDUMP_MAX_TASKS_NUM=30
CONFIG_ESP_COREDUMP_STACK_SIZE=1024
CONFIG_FREERTOS_CORETIMER_1=y
CONFIG_FREERTOS_HZ=200
# CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION is not set
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM=y
CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
# CONFIG_MQTT_TRANSPORT_WEBSOCKET is not set
CONFIG_MQTT_SKIP_PUBLISH_IF_DISCONNECTED=y
CONFIG_MQTT_TASK_CORE_SELECTION_ENABLED=y
CONFIG_NVS_ASSERT_ERROR_CHECK=y
CONFIG_EMAC_RECV_TASK_CORE=0
```

</details>

Backtrace
```
E (14035) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (14035) task_wdt:  - IDLE (CPU 0)
E (14035) task_wdt: Tasks currently running:
E (14035) task_wdt: CPU 0: ipc0
E (14035) task_wdt: CPU 1: IDLE
E (14035) task_wdt: Aborting.

abort() was called at PC 0x40130610 on core 0
0x40130610: task_wdt_isr at /esp-idf/components/esp_system/task_wdt.c:176 (discriminator 3)

Backtrace: 0x400821ca:0x3ffbea90 0x40093315:0x3ffbeab0 0x4009bfee:0x3ffbead0 0x40130610:0x3ffbeb40 0x40083509:0x3ffbeb70 0x4000bfed:0x3ffb74b0 0x4009861e:0x3ffb74c0 0x40082be3:0x3ffb74e0 0x4008d01a:0x3ffb7500 0x4008b402:0x3ffb7520
0x400821ca: panic_abort at /esp-idf/components/esp_system/panic.c:408

0x40093315: esp_system_abort at /esp-idf/components/esp_system/esp_system.c:137

0x4009bfee: abort at /esp-idf/components/newlib/abort.c:46

0x40130610: task_wdt_isr at /esp-idf/components/esp_system/task_wdt.c:176 (discriminator 3)

0x40083509: _xt_lowint1 at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x4009861e: vPortClearInterruptMaskFromISR at /esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at /esp-idf/components/freertos/port/xtensa/port.c:332

0x40082be3: vPortExitCriticalSafe at /esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:605
 (inlined by) esp_intr_noniram_enable at /esp-idf/components/esp_hw_support/intr_alloc.c:809

0x4008d01a: spi_flash_op_block_func at /esp-idf/components/spi_flash/cache_utils.c:130

0x4008b402: ipc_task at /esp-idf/components/esp_ipc/src/esp_ipc.c:77
```

Resulting coredump
```
==================== THREAD 1 (TCB: 0x3ffb75bc, name: 'ipc0') =====================
#0  0x400821cd in panic_abort (details=0x3ffbead0 \"abort() was called at PC 0x40130610 on core 0\") at /esp-idf/components/esp_system/panic.c:408
#1  0x40093318 in esp_system_abort (details=0x3ffbead0 \"abort() was called at PC 0x40130610 on core 0\") at /esp-idf/components/esp_system/esp_system.c:137
#2  0x4009bff1 in abort () at /esp-idf/components/newlib/abort.c:46
#3  0x40130613 in task_wdt_isr (arg=<optimized out>) at /esp-idf/components/esp_system/task_wdt.c:176
#4  0x4008350c in _xt_lowint1 () at /esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1114
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
```

KaeLL, 2023-03-02:
`-O2` is enabled, which I guess renders the issue invalid? If not, anyone from Espressif is free to reopen it.

KaeLL, 2023-03-02:
Never mind, I'm getting the same thing even with `-Og`

igrr, 2023-03-03:
@KaeLL would it be possible for you to attach the core dump binary and the corresponding ELF file?

KaeLL, 2023-03-03:
Yes, but not publicly. Suggestions?

KaeLL, 2023-03-14:
@igrr ?