espressif/esp-idf #10892: Single SSID Multi AP Environment Connect on SSID and Password to best RSSI

StudentSA, 2023-03-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Consider a Multi AP environment like a large home or office where AP's all share the same SSID name and Password. This is typically done to extend WiFi range and allow for "seemless" mobility within the area. 

I have found that when configuring the ESP32 to connect to an AP only providing the SSID a Password. I am not guaranteed to connect to the AP closest to me showing the best RSSI. instead many times I am connected to an AP further away with weak RSSI. The ESP will remain on this weak AP even though a stronger RSSI AP is closer. 

Does IDF cater for 
1. connecting to the AP with stronger RSSI when only SSID and Password are provided
2. re-establishing connection when a stronger RSSI is detected on AP with same SSID

What is the recommended way to handle this scenario?

kriegste, 2023-03-02:
`wifi_sta_config_t` has a member named `wifi_scan_method_t`.
Default is `WIFI_FAST_SCAN` and matches your problem description. Try `WIFI_ALL_CHANNEL_SCAN`. The only drawback is the longer delay during connection (2 or 3 seconds longer).
https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/network/esp_wifi.html#_CPPv418wifi_scan_method_t

kriegste, 2023-03-02:
But the IDF will not actively disconnect when there is a better AP available. You would have to do this by yourself (scan + compare + disconnect + connect).

vik-gokhale, 2023-03-06:
Hi!
This feature (connecting to best available RSSI) is now available on v4.4 and v4.3 in addition to v5.0 versions of the release.

**Usage**

wifi_config_t wifi_config = {
    .sta = {
        .ssid = <intended SSID>
        .password = <corresponding Password>
        .failure_retry_cnt = 3,
        .scan_method = WIFI_ALL_CHANNEL_SCAN,
    },
};

Please let us know in case of any more problems.

Thanks,
Vikram.