espressif/esp-idf #10893: (HELP!)esp_tls with tcp_server

Inverter-Jo, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is it possible to apply tls on tcp server?

TCP SERVER is currently implemented.
And the message does not seem to be encrypted even though it is in progress as shown in the code

listen_sock = socket(addr_family, SOCK_STREAM, ip_protocol);

esp_tls_cfg_server_t tls_cfg = {
    .servercert_buf = (const unsigned char *)test_cacert_pem,
    .servercert_bytes = strlen(test_cacert_pem) + 1,
    .serverkey_buf = (const unsigned char *)test_prvtkey_pem,
    .serverkey_bytes = strlen(test_prvtkey_pem) + 1,
};

esp_tls_server_session_create(&tls_cfg, listen_sock, tcp_tls);

  if (listen_sock < 0) {
      return ESP_ERR_INVALID_STATE;
  }

setsockopt(listen_sock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));

  int err = bind(listen_sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
  if (err != 0) {
      return ESP_ERR_INVALID_STATE;
  }
  err = listen(listen_sock, 1);
..
sock = accept(listen_sock, (struct sockaddr *)&source_addr, &addr_len);
....
recv(sock, rx_buffer, UART_PACKET_SIZE, 0);

What's wrong with it?
And are there any CONFIGs that need to be changed when applying TLS in TCP_SERVER?

Please help me.

Inverter-Jo, 2023-03-03:
Below is why I think communication encryption has not been achieved as a result of reading through WIRESHARK.
[image]

mahavirj, 2023-03-06:
@Inverter-Jo 

Please check [ESP HTTPS Server](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_https_server.html) guide on how to securely setup a local server. This guide also present a example that you may use as your starting point.

Inverter-Jo, 2023-03-06:
@mahavirj 

thanks for answer.

https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/examples/protocols/https_server/simple/main/main.c
If you think about it using TCP SERVER based on the link above,

I thought I could think that the code could be changed as below.

httpd_ssl_config_t conf = HTTPD_SSL_CONFIG_DEFAULT(); -> esp_tls_init();

&conf  -> esp_tls_cfg_server_t conf

httpd_ssl_start(&server, &config->httpd); -> esp_tls_server_session_create(esp_tls_cfg_server_t *cfg, int sockfd, esp_tls_t *tls)

is that right?

Inverter-Jo, 2023-03-08:
When I connected tls after tcp connection, I was able to check client hello and server hello through wireshark.
However, seeing that the above message is frequently invoked, I think there is a problem with my code.
In addition, the client side called the following error message.
esp-tls: couldn't get hostname for :: getaddrinfo() returns 202

There are two questions.
1. Is it right to connect tls after tcp connection?
2. How do I set the hostname of the esp_tls_conn_new_sync() function? Should I write the local netif hostname of the server if it is a client terminal?

Below is a summary of my codes.

client//
      sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
      if (sock < 0) {
	      ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
	      vTaskDelay(250);
      }
      struct esp_tls *tcp_tls = esp_tls_init();
      
      const char *host_name = "server";
      esp_tls_cfg_t tls_cfg = {
	      .clientcert_buf = (const unsigned char *)test_cacert_pem,//servercert_start,
	      .clientcert_bytes = strlen(test_cacert_pem) + 1,//servercert_end - servercert_start + 1
	      .clientkey_buf = (const unsigned char *)test_public_pem,//prvtkey_pem_start,
	      .clientkey_bytes = strlen(test_public_pem) + 1,//prvtkey_pem_start - prvtkey_pem_end + 1, 
      };		
      int err = connect(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
      if (err != 0) { 
	      ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);            
	      vTaskDelay(250);  
      }
      else {
	      ESP_LOGI(TAG, "Successfully connected");	
	      int tls_conn = esp_tls_conn_new_async(NULL,0,443,&tls_cfg, tcp_tls);
	      if(tls_conn>0){
	        esp_tls_conn_write(tcp_tls, wifi_disconnect_noti, UART_PACKET_SIZE);
	        vTaskResume(tcp_handle);
	      }
	      ESP_LOG_BUFFER_CHAR(TAG,wifi_config.sta.ssid,32);         
	      vTaskDelete(NULL);
      }

server//
      listen_sock = socket(addr_family, SOCK_STREAM, ip_protocol);
      ...
      int opt = 1;
      setsockopt(listen_sock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));//
      int err = bind(listen_sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
      ...
      ESP_LOGI(TAG, "Socket bound, port %d", PORT);	
      err = listen(listen_sock, 1);
      ...
      ...
      socklen_t addr_len = sizeof(source_addr);
      sock = accept(listen_sock, (struct sockaddr *)&source_addr, &addr_len);
      ....	
      setsockopt(sock, SOL_SOCKET, SO_KEEPALIVE, &keepAlive, sizeof(int));
      ...
      ...
      ...
      
      if (source_addr.ss_family == PF_INET) {
	      inet_ntoa_r(((struct sockaddr_in *)&source_addr)->sin_addr, addr_str, sizeof(addr_str) - 1);
      }				
      struct esp_tls *temp_tcp_tls = esp_tls_init();				
      tcp_tls = temp_tcp_tls;
      esp_tls_cfg_server_t tls_cfg = {
	      .servercert_buf = (const unsigned char *)test_cacert_pem,
	      .servercert_bytes = strlen(test_cacert_pem) + 1,
	      .serverkey_buf = (const unsigned char *)test_prvtkey_pem,
	      .serverkey_bytes = strlen(test_prvtkey_pem) + 1,
      };
      esp_tls_server_session_create(&tls_cfg,sock, tcp_tls);

Inverter-Jo, 2023-03-13:
Done!
It was my mistake!

Thanks!

AdityaHPatwardhan, 2023-03-13:
Hi @Inverter-Jo I am closing this issue since it seems to be fixed. Please feel free to re-open in case of any query.
Thanks,
Aditya

closed 2023-03-13 by AdityaHPatwardhan