espressif/esp-idf #10896: Compiling the ULP RISC-V Code

osmond2023, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

stable (v5.01)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Compiled

### What is the actual behavior?

Compilation failed

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
[os@fedora gpio_interrupt]$ idf.py build
Executing action: all (aliases: build)
Running make in directory /home/os/esp/gpio_interrupt/build
Executing "make -j 4 all"...
[  0%] Generating ../../partition_table/partition-table.bin
[  0%] Creating directories for 'ulp_main'
[  0%] Generating memory.ld linker script...
[  0%] Built target custom_bundle
[  0%] Built target memory_ld
[  0%] No download step for 'ulp_main'
[  0%] Generating project_elf_src_esp32s2.c
Partition table binary generated. Contents:
*******************************************************************************
[  0%] Built target _project_elf_src
[  0%] No update step for 'ulp_main'
[  0%] Building C object esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp_common/ulp_common.c.obj
[  1%] Building C object esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp_riscv/ulp_riscv.c.obj
[  1%] No patch step for 'ulp_main'
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
[  2%] Performing configure step for 'ulp_main'
-- Found Git: /usr/bin/git (found version "2.39.2") 
*******************************************************************************
[  2%] Built target partition_table_bin
[  2%] Building C object esp-idf/ulp/CMakeFiles/__idf_ulp.dir/ulp_riscv/ulp_riscv_adc.c.obj
[  2%] Creating directories for 'bootloader'
[  2%] Linking C static library libulp.a
[... 211 lines cut ...]
[ 98%] Built target bootloader.elf
[100%] Generating binary image from built executable
esptool.py v4.5.1
Creating esp32s2 image...
Merged 1 ELF section
Successfully created esp32s2 image.
Generated /home/os/esp/gpio_interrupt/build/bootloader/bootloader.bin
[100%] Built target gen_project_binary
Bootloader binary size 0x4c30 bytes. 0x23d0 bytes (32%) free.
[100%] Built target bootloader_check_size
[100%] Built target app
[  6%] No install step for 'bootloader'
[  6%] Completed 'bootloader'
[  6%] Built target bootloader
make: *** [Makefile:136: all] Error 2
make failed with exit code 2, output of the command is in the /home/os/esp/gpio_interrupt/build/log/idf_py_stderr_output_132622 and /home/os/esp/gpio_interrupt/build/log/idf_py_stdout_output_132622
```

### More Information.

Hi, I copied the example from Application Examples, ULP RISC-V Coprocessor polls GPIO while main CPU is in deep sleep: system/ulp_riscv/gpio. When I try to run idf.py build, it doesn't complete successfully. What should I do to fix it? Thank you.

hp notebook
windows 10x
Fedora 37

ESP-Marius, 2023-03-06:
@osmond2023 I see a `cmake: Permission denied` so this is probably related to your setup somehow.

1. Can you successfully build the project if you build it inside the `esp-idf/examples` folder without copying it out first?
2. Does other examples also give the same error?

osmond2023, 2023-03-07:
Hi Marius,
I reinstalled ESP-IDF and built it inside the examples folder. It builds successfully. Thanks.

ESP-Marius, 2023-03-08:
That's good, seems like there just a permission problem with the folder when you copy it out then.

If you run into it again then please check that the permission are set correctly (how to do this would depend on your operating system).

Closing this one for now then.

closed 2023-03-08 by ESP-Marius

asiot-dkolbe, 2024-05-09:
Ran into this problem on a relatively(?) clean but older version (IDF 4.4.2). Just in case anyone runs into the problem and hits a dead end:

The problem seems to be caused in /home/david/esp/esp-idf-v4.4.2.clean/examples/system/ulp_riscv/gpio/build/esp-idf/main/ulp_main/CMakeFiles/ulp_main.dir/build.make

This file contains three rules that look like:
```
/home/david/esp/esp-idf-v4.4.2.clean/examples/system/ulp_riscv/gpio/main/ulp/main.c: esp32s3.ulp.riscv.ld
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --blue --bold --progress-dir=/home/david/esp/esp-idf-v4.4.2.clean/examples/system/ulp_riscv/gpio/build/esp-idf/main/ulp_main/CMakeFiles --progress-num=$(CMAKE_PROGRESS_1) "Generating /home/david/esp/esp-idf-v4.4.2.clean/examples/system/ulp_riscv/gpio/main/ulp/main.c"
	cmake -E echo
```

"cmake" in "cmake -E echo" resolves to a directory name, presumably because the parent directory is in the path. The directory in question is "/home/david/esp/esp-idf-v4.4.2.clean/tools/cmake". execve of course can't execute directories, and produces EACCESS.
Replacing all three instances of "cmake" by "@$(CMAKE_COMMAND)" fixes the problem.