espressif/esp-idf #10897: Run idf.py monitor without rebuilding project

ligantx, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I've build my project in another remote machine and create a merged flash file. 
I receive the flash file to my local machine and flash it with esptool.py sucessfully.
Now, i try to monitor, but it starting to rebuild the project locally. Can i bypass the building and just monitor the device?
I don't find any esptool doing exactly that.

### What is the actual behavior?

idf.py monitor, rebuilds the project

### Steps to reproduce.

1. create a merged-flash.bin
2. use esptool.py to flash it
3. try to idf.py monitor the board

dobairoland, 2023-03-06:
> Can i bypass the building and just monitor the device?

Yes, this is already possible. I've checked that it should be working for v5.0.1. Are you sure that this is the right version you are using on that computer where you are running `idf.py monitor`? This behavior was introduced in ESP-IDF v5.0 and it is not available in earlier versions.

`idf.py monitor` can even be invoked in a clean project immediately after `idf.py fullclean` without triggering building.

ligantx, 2023-03-06:
Thanks for answering.
I just checked and it looks i run the latest version.
This is what i do:
`C:\\Espressif\\frameworks\\esp-idf-v5.0.1\\export.bat`
`idf.py --version` 
returns `ESP-IDF v5.0.1`

`idf.py fullclean`
Returns
```
Build directory '..\myappfolder\build' not found. Nothing to clean.
Executing action: remove_managed_components
Done
```

`idf.py monitor`
Although it looks it's not rebuilding as usual, it takes some seconds, creates the build folder and ends up at the command:
```
Executing "C:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v5.0.1\tools/idf_monitor.py -p COM12 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 c:\users\username\tutorial\blink\build\blink.elf --force-color -m 'C:\Espressif\python_env\idf5.0_py3.11_env\Scripts\python.exe' 'C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf.py'"...
```
and start the usual monitoring.
If i rerun `idf.py monitor` it goes straight to the monitoring.
Can i bypass the creation of building folder and go straight to the monitoring (as this happens the second time i run the `idf.py monitor`)? (maybe provide the .elf file from my remote machine). 
Also, the build/blink.elf file doesnt exist.

dobairoland, 2023-03-06:
> Although it looks it's not rebuilding as usual, it takes some seconds, creates the build folder and ends up at the command:

That is normal because cmake is still invoked in order to create a basic build structure. No build should be done in that case. This happens only after `fullclean` and once.

> If i rerun idf.py monitor it goes straight to the monitoring.

That means it works as it supposed to be.

> Can i bypass the creation of building folder and go straight to the monitoring (as this happens the second time i run the idf.py monitor)? 

See my previous comments. 

> (maybe provide the .elf file from my remote machine).

Yes, you can copy it to the path you see in the output. And it should be used by `idf.py monitor` without re-writing it. But it is not necessary if you don't need the addr2line outputs (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-monitor.html#automatic-address-decoding).

> Also, the build/blink.elf file doesnt exist.

That is also normal in this case.

dobairoland, 2023-03-20:
One more note. It was pointed out to me by @peterdragun that maybe you look more for invoking IDF Monitor directly.

The following commands invoke IDF Monitor directly without creating a build directory. See the help how to use it (or the previously linked documentation page).

Run this in ESP-IDF v5.0.x (or previous versions):
```
python tools/idf_monitor.py --help
```

Run this in v5.1 (or newer):
```
python -m esp_idf_monitor --help
```

ligantx, 2023-03-23:
> One more note. It was pointed out to me by @peterdragun that maybe you look more for invoking IDF Monitor directly.
>
> The following commands invoke IDF Monitor directly without creating a build [...]

i have v5.0.1 and `python %IDF_PATH%/tools/idf_monitor.py remoteapp.elf` returns:

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2 (_run_outer):
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Lib\threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Lib\threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf_monitor_base\stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "C:\Espressif\frameworks\esp-idf-v5.0.1\tools\idf_monitor_base\serial_reader.py", line 52, in run
    self.serial.open()
  File "C:\Espressif\python_env\idf5.0_py3.11_env\Lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/ttyUSB0': FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)
```

I guess it doesn't recognize i'm using Windows..can i choose monitor port manually?

ligantx, 2023-03-23:
For windows and idf v5.0.x i had to set COM port and choose the `.elf` file (after setting the environment with `export.bat`):

`%IDF_PATH%/tools/idf_monitor.py -p %ESPTOOL_PORT% build/myapp.elf`

Now it works fine..