espressif/esp-idf #10899: btm_rrm_task not stopped after calling esp_wifi_deinit()

kenozz-igs, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-541-g885e501d99

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

esp_wifi_deinit() should stop the task **btm_rrm_task** running. 

### What is the actual behavior?

esp_wifi_deinit() is not stopping the task **btm_rrm_task**. 

### Steps to reproduce.

Project configured with CONFIG_WPA_11KV_SUPPORT and CONFIG_WPA_11R_SUPPORT enabled 

Start Wifi and confirm running, then stop and de-initialise the WiFi via esp_wifi_disconnect(), esp_wifi_stop() and esp_wifi_deinit() 

Check running tasks and confirm that **btm_rrm_task** is still running.  

Repeat steps and there will now be 2 instances of **btm_rrm_task** 

### More Information.

Defect appears to be in function esp_supplicant_common_deinit() in file /components/wpa_supplicant/esp_supplicant/src/esp_common.c. 

The terminate signal **SIG_SUPPLICANT_DEL_TASK** is not sent as logic is incorrect on following line. 
if (!s_supplicant_task_hdl && esp_supplicant_post_evt(SIG_SUPPLICANT_DEL_TASK, 0) != 0) {

Because **s_supplicant_task_hdl** is not NULL since task active, then **esp_supplicant_post_evt()** is not called as fails on first condition check(!s_supplicant_task_hdl).  Believe the && operator should be || since it is an error catch for handle being NULL or the post failing.

Alvin1Zhang, 2023-03-04:
Thanks for reporting.

AxelLin, 2023-04-29:
@jgujarathi Any update for this fix?

AxelLin, 2023-08-20:
How is the status now? @jgujarathi @Alvin1Zhang

Alvin1Zhang, 2023-08-21:
Thanks for reporting and tracking, the fix has been available and fix on release/5.0 is at https://github.com/espressif/esp-idf/commit/bb89a5769b9f0fb6a07359ac2e258e1841fd1b14, feel free to reopen.

closed 2023-08-21 by Alvin1Zhang