espressif/esp-idf #10900: esp_http_client connection timeouts with ESP_ERR_HTTP_FETCH_HEADER

mbastida123, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have an endpoint which we now that is a little bit slower than average (it can take up to 2-3 seconds to respond). And it seems like this makes the esp_http_client error on ESP_ERR_HTTP_FETCH_HEADER. It doesn't fail always. Sometimes it works on the first try, others on the 10th and others it never works.

Knowing that in this case ESP_ERR_HTTP_FETCH_HEADER --> timeout issue I have set `.timeout_ms=10000` in `esp_http_client_config_t` to try fixing the issue. But it doesn't seem to be working as expected. Is there something else I have to do?

The code I use:
```
esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    static char *output_buffer; // Buffer to store response of http request from event handler
    static int output_len;      // Stores number of bytes read
    switch (evt->event_id)
    {
    case HTTP_EVENT_ERROR:
        ESP_LOGD(TAG, "HTTP_EVENT_ERROR: %s", (char *)evt->data);

        break;
    case HTTP_EVENT_ON_CONNECTED:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_CONNECTED");
        break;
    case HTTP_EVENT_HEADER_SENT:
        ESP_LOGD(TAG, "HTTP_EVENT_HEADER_SENT");
        break;
    case HTTP_EVENT_ON_HEADER:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);
        break;
    case HTTP_EVENT_ON_DATA:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_DATA, len=%d", evt->data_len);
        /*
         *  Check for chunked encoding is added as the URL for chunked encoding used in this example returns binary data.
         *  However, event handler can also be used in case chunked encoding is used.
         */
        if (!esp_http_client_is_chunked_response(evt->client))
        {
            // If user_data buffer is configured, copy the response into the buffer
            if (evt->user_data)
            {
                memcpy(evt->user_data + output_len, evt->data, evt->data_len);
            }
            else
            {
                if (output_buffer == NULL)
                {
                    output_buffer = (char *)malloc(esp_http_client_get_content_length(evt->client));
                    output_len = 0;
                    if (output_buffer == NULL)
                    {
                        ESP_LOGE(TAG, "Failed to allocate memory for output buffer");
                        return ESP_FAIL;
                    }
                }
                memcpy(output_buffer + output_len, evt->data, evt->data_len);
            }
            output_len += evt->data_len;
        }
        else
        {
            ESP_LOGV(TAG, "strlen user_data: %d", strlen((char *)evt->user_data));

            memcpy(evt->user_data + output_len, evt->data, evt->data_len);
            // memcpy(evt->user_data + strlen((char*)evt->user_data), evt->data, evt->data_len);
            ESP_LOGV(TAG, "Chunked data: %s", (char *)evt->data); // HERE!
            ESP_LOGD(TAG, "Partial user_data: %s", (char *)evt->user_data);
            output_len += evt->data_len;
        }

        break;
    case HTTP_EVENT_ON_FINISH:
        ESP_LOGD(TAG, "HTTP_EVENT_ON_FINISH");
        if (output_buffer != NULL)
        {
            // Response is accumulated in output_buffer. Uncomment the below line to print the accumulated response
            // ESP_LOG_BUFFER_HEX(TAG, output_buffer, output_len);
            free(output_buffer);
            output_buffer = NULL;
        }
        output_len = 0;
        break;
    case HTTP_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "HTTP_EVENT_DISCONNECTED");
        int mbedtls_err = 0;
        esp_err_t err = esp_tls_get_and_clear_last_error((esp_tls_error_handle_t)evt->data, &mbedtls_err, NULL);
        if (err != 0)
        {
            ESP_LOGI(TAG, "Last esp error code: 0x%x", err);
            ESP_LOGI(TAG, "Last mbedtls failure: 0x%x", mbedtls_err);
        }
        if (output_buffer != NULL)
        {
            free(output_buffer);
            output_buffer = NULL;
        }
        output_len = 0;
        break;
    }
    return ESP_OK;
}
uint16_t https_execute(esp_http_client_handle_t client)
{
    int statusCodeTemp = 0;

    esp_err_t err;
    ESP_LOGV(TAG, "Executing HTTPS call");

    err = esp_http_client_perform(client);

    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));

        esp_http_client_close(client);
        esp_http_client_cleanup(client);

        return -1;
    }
    else if (err == ESP_OK)
    {
        ESP_LOGI(TAG, "HTTP POST Status = %d, content_length = %d",
                 esp_http_client_get_status_code(client),
                 esp_http_client_get_content_length(client));

        // respR.buffResp = (char*)malloc(MAX_HTTP_RECV_BUFFER+1);
        // sprintf(respR.buffResp, "%s", local_response_buffer);
    }

    statusCodeTemp = esp_http_client_get_status_code(client);

    esp_http_client_close(client);
    esp_http_client_cleanup(client);

    return statusCodeTemp;
}
```

The https_execute() is called thrugh:
```
esp_http_client_config_t config = {
        .url = url,
        .timeout_ms=10000,
        .event_handler = _http_event_handler,
        .user_data = response, // Pass address of local buffer to get response
        .crt_bundle_attach = esp_crt_bundle_attach,
    };

    uint8_t retries=0;
    if(authHeader!=NULL)
        free(authHeader);
    authHeader=(char*)malloc(1100);

    while(retries<50){
        esp_http_client_handle_t client = esp_http_client_init(&config);

        esp_http_client_set_method(client, HTTP_METHOD_POST);
        esp_http_client_set_header(client, "Content-Type", "application/x-www-form-urlencoded");
        esp_http_client_set_header(client, "accept", "application/json");
        esp_http_client_set_post_field(client, identificationParams, strlen(identificationParams));

        uint16_t code= https_execute(client);

        if(code == 200)
             break;
}
```

hmalpani, 2023-03-08:
Hello @mbastida123 

I tried to reproduce this but seems like it will not be possible to replicate the server you are using. Setting the timeout in config is a good option in the cases where server response is slow. You are setting `timeout` to 10s but it is still failing which means that the server is taking more time to send the response. Can you make sure that the server always responds within the time you stated? You can also try increasing the timeout value in config.

hmalpani, 2023-03-15:
Hello @mbastida123 
Any update on this issue?

mbastida123, 2023-03-18:
It appears to be working fine now. i don't know if it's the server or that I was configuring something wrongly on the http_client.

If the issue reapears I will create another issue and I will try to give you the server address so you can reproduce it yourself.

Thanks!

closed 2023-03-18 by mbastida123