espressif/esp-idf #10902: mbedtls failure with unknown error codes

Kampi, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I try to use the ESP HTTP client for HTTPS POST requests by using this code and IDF version `5.0.1` on a ESP32.

```
esp_err_t Error;
HTTP_Response_t Response;
 esp_http_client_config_t Config;
esp_http_client_handle_t Client;

memset(&Config, 0, sizeof(Config));

Config.timeout_ms = CONFIG_API_TIMEOUT;
Config.cert_pem = _CA_Server_Start;
Config.transport_type = HTTP_TRANSPORT_OVER_SSL,
Config.method = HTTP_METHOD_POST;
Config.path = "/";
Config.client_cert_pem = NULL;
Config.client_cert_len = 0;
Config.user_agent = CONFIG_GENERAL_DEFAULT_NAME;
Config.event_handler = _HTTP_On_Event_Handler,
Config.user_data = &Response;
Config.url = _HTTP_URL.c_str();
Config.port = CONFIG_API_PORT;

ESP_LOGI(TAG, "Opening connection...");
ESP_LOGI(TAG, "    Time: %u", static_cast<unsigned int>(Timestamp));
ESP_LOGI(TAG, "    Host: %s", Config.url);
ESP_LOGI(TAG, "    Endpoint: %s", Config.path);
ESP_LOGI(TAG, "    Port: %u", Config.port);

Client = esp_http_client_init(&Config);
if(Client == NULL)
{
  Status_SetError(ERROR_WIFI_CONNECTION_ERROR);
  
  return ESP_ERR_HTTP_CONNECTION_CLOSED;
}

if((Response.StatusCode == 200) || (Response.StatusCode == 201))
{
    // TODO
}

Error = esp_http_client_close(Client);
Error |= esp_http_client_cleanup(Client);
``` 

But I get the following error messages

```
E (12308) HTTP: Last ESP error code: 0x103
E (12308) HTTP: Last mbedtls failure: 0x800ff50e
E (12308) HTTP: Last ESP error code: 0x103
E (12318) HTTP: Last mbedtls failure: 0x800d8509
```

What is the reason for this messages?

mahavirj, 2023-03-08:
@Kampi 

Error code 0x103 refers to `ESP_ERR_INVALID_STATE`. Probably its an issue with the API usage in your sample application and in that scenario mbedTLS code could be incorrect. Can you please enable debug/verbose log level and check your application for this error code once? Please share small application that recreates this problem along with your sdkconfig.

Kampi, 2023-03-09:
Hi @mahavirj,

I have increased the debug level to `Verbose`, but I don´t get any additional information:

```
I (12802) HTTP: Opening connection...
I (12812) HTTP:     Time: 0
I (12812) HTTP:     Host: https://...
I (12822) HTTP:     Endpoint: /
I (12822) HTTP:     Port: 443
D (12832) HTTP: HTTP_EVENT_DISCONNECTED
E (12832) HTTP: Last ESP error code: 0x103
E (12842) HTTP: Last mbedtls failure: 0x800edc5e
D (12842) HTTP: HTTP_EVENT_DISCONNECTED
E (12842) HTTP: Last ESP error code: 0x103
E (12852) HTTP: Last mbedtls failure: 0x800d4785
```

Kampi, 2023-03-09:
Figured out the issue. I forgot to use `esp_http_client_open`. I have tested the code with the HTTP example from Espressif and they have used different functions which called `esp_http_client_open` internally.

closed 2023-03-09 by Kampi