espressif/esp-idf #10903: Recreating msc_example_main in an existing project

skanky-dev, 2023-03-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have the sample project msc_example_main successfully accessing a USB memory stick. I now need to add the functionality to my existing ESP32S3 project.
I run idf.py add-dependency "espressif/usb_host_msc^1.0.3" as instructed at 
   https://components.espressif.com/components/espressif/usb_host_msc
and it says it installs ok.
I start porting the sample project code into my project, but when I run it this function call causes a Panic abort:
   ESP_ERROR_CHECK( usb_host_install(&host_config) );

This is the full output of the error:
`SP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4201c5e5
0x4201c5e5: twai_driver_install at C:/Espressif/esp/esp-idf/components/driver/twai.c:449 (discriminator 1)

file: "/IDF/components/driver/twai.c" line 449
func: twai_driver_install
expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)

abort() was called at PC 0x4037af7b on core 0
0x4037af7b: _esp_error_check_failed at C:/Espressif/esp/esp-idf/components/esp_system/esp_err.c:47

Backtrace: 0x40375df2:0x3fcf3900 0x4037af85:0x3fcf3920 0x40381156:0x3fcf3940 0x4037af7b:0x3fcf39b0 0x4201c5e5:0x3fcf39e0 0x4200cc72:0x3fcf3a00 0x42008eb4:0x3fcf3a60 0x42009333:0x3fcf3ae0 0x42034fcf:0x3fcf3b00 0x4037dd8d:0x3fcf3b30   
0x40375df2: panic_abort at C:/Espressif/esp/esp-idf/components/esp_system/panic.c:412

0x4037af85: esp_system_abort at C:/Espressif/esp/esp-idf/components/esp_system/esp_system.c:135

0x40381156: abort at C:/Espressif/esp/esp-idf/components/newlib/abort.c:38

0x4037af7b: _esp_error_check_failed at C:/Espressif/esp/esp-idf/components/esp_system/esp_err.c:47

0x4201c5e5: twai_driver_install at C:/Espressif/esp/esp-idf/components/driver/twai.c:449 (discriminator 1)

0x4200cc72: CanbusInitialise at C:/_Dropbox/Dropbox/_PROGRAMMING/_GIT_LIVE_MASTER/Dyena-VimmsLite-MainPcb/VimmsLite_main/main/canbus.c:51

0x42008eb4: Initialise at C:/_Dropbox/Dropbox/_PROGRAMMING/_GIT_LIVE_MASTER/Dyena-VimmsLite-MainPcb/VimmsLite_main/main/ap-main.c:189

0x42009333: app_main at C:/_Dropbox/Dropbox/_PROGRAMMING/_GIT_LIVE_MASTER/Dyena-VimmsLite-MainPcb/VimmsLite_main/main/ap-main.c:294

0x42034fcf: main_task at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x4037dd8d: vPortTaskWrapper at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: 7e628190de84f461

Rebooting...`

I notice that the sample project has this in its CMakeLists.txt:
   REQUIRES usb vfs

So I add it to mine, but now when I build I now get build errors such as:
   fatal error: esp_adc/adc_oneshot.h: No such file or directory
   fatal error: esp_timer.h: No such file or directory

These are include files I'm using in my project to access the timer, AtoD, etc

So it seems adding in USB and vfs is causing some sort of dramatic change to my build. I've searched for an hour and can't find any explanation or solution to this. Can anyone advise what might be going on? My assumption from what I have found is maybe its casing a downgrade of the API version or something, but honestly I've no idea why or what to do about it.
There is no documentation or help I can find on using the USB MSC driver in a project to aid me

igrr, 2023-03-03:
Regarding this part,

> So I add it to mine, but now when I build I now get build errors such as:
> fatal error: esp_adc/adc_oneshot.h: No such file or directory
> fatal error: esp_timer.h: No such file or directory

You need to add `esp_adc` and `esp_timer` components to the REQUIRES list. Before you declared the dependencies on `usb` and `vfs` you didn't have to do this due to the implict dependency behavior of `main` component ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#main-component-requirements)).

If you cange that line to `REQUIRES usb vfs esp_adc esp_timer`, then it should work. (You may have to add more dependencies if your `main` component uses more features.)

Regarding this part,

> ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4201c5e5
> file: "/IDF/components/driver/twai.c" line 449
> func: twai_driver_install
> expression: esp_intr_alloc(ETS_TWAI_INTR_SOURCE, g_config->intr_flags, twai_intr_handler_main, NULL, &p_twai_obj->isr_handle)

Here the installation of the TWAI driver failed because the system ran out of the CPU interrupt inputs. This can happen when several drivers or subsystems are initialized. Something along the lines of this suggestion might help: https://github.com/espressif/esp-idf/issues/9880#issuecomment-1264334885.

igrr, 2023-03-06:
Basically, interrupt inputs are a limited resource on the ESP32, and initializing certain features requires interrupt inputs to be attached to the peripherals. Just like other resources (such as RAM), the program can run out of the interrupt inputs if it uses multiple features at once. The good part is that it should still be possible to make everything work by utilizing higer level (level 2 or 3) interrupts and possibly by usign shared interrupts.

To debug the interrupt allocations, you can uncomment this line, and also set "log level" option in menuconfig to "Debug".

https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/components/esp_hw_support/intr_alloc.c#L39-L44

(sorry that there is no Kconfig option or another way to enable this without changing IDF code, yet!)

Afterwards, run your program again, you should now get a bunch of lines with `intr_alloc` log tag. Please copy the full log and attach it here. These will give us an idea which interrupts are being allocated.

---

> If I change i2c_driver_install to use ESP_INTR_FLAG_LOWMED it doesn't solve the issue
> If I change usb_host_install to use ESP_INTR_FLAG_LOWMED it doesn't solve the issue

Note, if the application is lacking more than one interrupt, it's possible that you need _both_ of these changes (and perhaps some more) to solve the problem.

> twai_driver_install() doesn't have a intr_alloc_flags argument as far as I can see.

It is part of `twai_general_config_t` structure, the member is called `intr_flags`. You can modify it after initializing the structure with TWAI_GENERAL_CONFIG_DEFAULT(...).

> I am using SPI, UART and JTAG ports also, but none have a "_driver_install" I am using.

For SPI master and slave drivers, this can be set by `intr_flags` member of `spi_bus_config_t`, which is used in the call to `spi_bus_initialize`.

For UART (if you are using the UART driver API, driver/uart.h), it can be set using the last argument of `uart_driver_install`, `intr_alloc_flags`. If you are using UART only for console output without calling `uart_driver_install`, then it doesn't occupy an interrupt.

skanky-dev, 2023-03-08:
Oh my, this is severely complex for someone non-expert with this aspect of the ESP32...!

`
I (0) cpu_start: App cpu up.
D (221) clk: RTC_SLOW_CLK calibration value: 3742477
I (232) cpu_start: Pro cpu start user code
I (232) cpu_start: cpu freq: 160000000 Hz
I (232) cpu_start: Application information:
I (235) cpu_start: Project name:     VimmsLite_main
I (240) cpu_start: App version:      4d11ae7
I (245) cpu_start: Compile time:     Mar  8 2023 12:59:56
I (251) cpu_start: ELF file SHA256:  bdc110a06473b3ff...
I (257) cpu_start: ESP-IDF:          v5.0-dirty
D (262) memory_layout: Checking 5 reserved memory ranges:
D (268) memory_layout: Reserved memory range 0x3fc84000 - 0x3fc93400
D (274) memory_layout: Reserved memory range 0x3fc93400 - 0x3fc988a8
D (281) memory_layout: Reserved memory range 0x3fceee34 - 0x3fcf0000
D (287) memory_layout: Reserved memory range 0x40374000 - 0x40383400
0x40374000: _WindowOverflow4 at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1742

D (294) memory_layout: Reserved memory range 0x600fe000 - 0x600fe010
D (300) memory_layout: Building list of available memory regions:
D (306) memory_layout: Available memory region 0x3fc988a8 - 0x3fca0000
D (313) memory_layout: Available memory region 0x3fca0000 - 0x3fcb0000
D (319) memory_layout: Available memory region 0x3fcb0000 - 0x3fcc0000
D (326) memory_layout: Available memory region 0x3fcc0000 - 0x3fcd0000
D (333) memory_layout: Available memory region 0x3fcd0000 - 0x3fce0000
D (339) memory_layout: Available memory region 0x3fce0000 - 0x3fce9710
D (346) memory_layout: Available memory region 0x3fce9710 - 0x3fceee34
D (352) memory_layout: Available memory region 0x3fcf0000 - 0x3fcf8000
D (359) memory_layout: Available memory region 0x600fe010 - 0x60100000
I (366) heap_init: Initializing. RAM available for dynamic allocation:
D (373) heap_init: New heap initialised at 0x3fc988a8
I (378) heap_init: At 3FC988A8 len 00050E68 (323 KiB): D/IRAM
I (384) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
D (391) heap_init: New heap initialised at 0x3fcf0000
I (396) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
D (402) heap_init: New heap initialised at 0x600fe010
I (407) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
D (414) intr_alloc: get_available_int: try to find existing. Cpu: 0, Source: 39
D (421) intr_alloc: get_free_int: start looking. Current cpu: 0
D (427) intr_alloc: Int 0 reserved 2 priority 1 LEVEL hasIsr 0
D (433) intr_alloc: ....Unusable: reserved
D (437) intr_alloc: Int 1 reserved 2 priority 1 LEVEL hasIsr 0
D (443) intr_alloc: ....Unusable: reserved
D (447) intr_alloc: Int 2 reserved 0 priority 1 LEVEL hasIsr 0
D (453) intr_alloc: Int 3 reserved 0 priority 1 LEVEL hasIsr 0
D (459) intr_alloc: ...worse than int 2
D (463) intr_alloc: Int 4 reserved 2 priority 1 LEVEL hasIsr 0
D (469) intr_alloc: ....Unusable: reserved
D (473) intr_alloc: Int 5 reserved 2 priority 1 LEVEL hasIsr 0
D (479) intr_alloc: ....Unusable: reserved
D (483) intr_alloc: Int 6 reserved 0 priority 1 EDGE hasIsr 0
D (489) intr_alloc: ....Unusable: special-purpose int
D (494) intr_alloc: Int 7 reserved 0 priority 1 EDGE hasIsr 0
D (500) intr_alloc: ....Unusable: special-purpose int
D (505) intr_alloc: Int 8 reserved 2 priority 1 LEVEL hasIsr 0
D (511) intr_alloc: ....Unusable: reserved
D (515) intr_alloc: Int 9 reserved 0 priority 1 LEVEL hasIsr 0
D (521) intr_alloc: ...worse than int 2
D (525) intr_alloc: Int 10 reserved 0 priority 1 EDGE hasIsr 0
D (531) intr_alloc: ....Unusable: incompatible trigger type
D (536) intr_alloc: Int 11 reserved 0 priority 3 EDGE hasIsr 0
D (542) intr_alloc: ....Unusable: special-purpose int
D (547) intr_alloc: Int 12 reserved 0 priority 1 LEVEL hasIsr 0
D (553) intr_alloc: ...worse than int 2
D (557) intr_alloc: Int 13 reserved 0 priority 1 LEVEL hasIsr 0
D (563) intr_alloc: ...worse than int 2
D (567) intr_alloc: Int 14 reserved 2 priority 7 LEVEL hasIsr 0
D (573) intr_alloc: ....Unusable: reserved
[... 523 lines cut ...]

Backtrace: 0x40375e1e:0x3fcf38d0 0x4037afc9:0x3fcf38f0 0x40381302:0x3fcf3910 0x4037afbf:0x3fcf3980 0x420214bd:0x3fcf39b0 0x4200d9ae:0x3fcf39d0 0x4200991c:0x3fcf3a30 0x42009ddf:0x3fcf3ab0 0x4203a0ff:0x3fcf3b00 0x4037df39:0x3fcf3b30     
0x40375e1e: panic_abort at C:/Espressif/esp/esp-idf/components/esp_system/panic.c:412

0x4037afc9: esp_system_abort at C:/Espressif/esp/esp-idf/components/esp_system/esp_system.c:135

0x40381302: abort at C:/Espressif/esp/esp-idf/components/newlib/abort.c:38

0x4037afbf: _esp_error_check_failed at C:/Espressif/esp/esp-idf/components/esp_system/esp_err.c:47

0x420214bd: twai_driver_install at C:/Espressif/esp/esp-idf/components/driver/twai.c:449 (discriminator 1)

0x4200d9ae: CanbusInitialise at C:/_Dropbox/Dropbox/_PROGRAMMING/_GIT_LIVE_MASTER/Dyena-VimmsLite-MainPcb/VimmsLite_main/main/canbus.c:51

0x4200991c: Initialise at C:/_Dropbox/Dropbox/_PROGRAMMING/_GIT_LIVE_MASTER/Dyena-VimmsLite-MainPcb/VimmsLite_main/main/ap-main.c:189

0x42009ddf: app_main at C:/_Dropbox/Dropbox/_PROGRAMMING/_GIT_LIVE_MASTER/Dyena-VimmsLite-MainPcb/VimmsLite_main/main/ap-main.c:294

0x4203a0ff: main_task at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_common.c:131 (discriminator 2)

0x4037df39: vPortTaskWrapper at C:/Espressif/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151

ELF file SHA256: bdc110a06473b3ff

Rebooting...`

skanky-dev, 2023-03-08:
OK so I've taken your advice and used this to change the can irq

`	twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(GPIO_CANBUS_TX, GPIO_CANBUS_RX, TWAI_MODE_NORMAL);
	g_config.intr_flags = ESP_INTR_FLAG_LOWMED;		//Move canbus irq to free up the default level 1 IRQ it will take up`

Once I've done that the resetting issue is fixed, so that is great thank you.

However I'm really worried about hitting this issue again, as we want to use more peripherals in the future such as Bluetooth and WiFi for instance. Are there any resources or explanations anywhere that provide a simple way to understand the intr_alloc log output?  A simpler way to visually see what IRQs are being used? If I pick a line at random it is completely non-obvious what its telling me:
`D (2618) intr_alloc: Int 17 reserved 0 priority 1 LEVEL hasIsr 0`
What on earth is hasIsr? A google for it produces nothing useful. Searching docs.espressif.com gives me no results. What does the value after it mean? How does anyone at a normal level of experience with the ESP32 work out how to solve issues such as this without needing to come to someone as expert as you?

skanky-dev, 2023-03-08:
OK, I now see that hasIsr is not spelt with an L but its actually an i. Once I realise its an i its obvious what the word is, but I'm still struggling to see how this lets me properly work out what is consuming what irq. If I count the following lines:
`intr_alloc: Connected`
I assume if I go and try to find what src39, src52 is maybe I see some irqs being allocated. But there are only 7 of those lines. What's used up all of the other irqs? 
Is there a way to have debug output the actual irq allocation in a more friendly way, maybe by adding in code to my application to do it? 
Thanks
Adam