espressif/esp-idf #10905: NimBLE Examples: Bond lost when RPA enabled peer changes address.

braiden, 2023-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3778-g370e250072

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

My  es32c3 (peripheral) bonds with my phone (central). The bond should be persist.

### What is the actual behavior?

After my phone generates a new RPA address the bond no longer works and I must repair.

### Steps to reproduce.

1. Using example `examples/bluetooth/nimble/bleprph` with the following config value enabled: `EXAMPLE_BONDING`, `BT_NIMBLE_SM_SC`, `EXAMPLE_ENCRYPTION`. 
2. Pair the phone and observer the logged `peer_ota_addr` and `peer_id_addr`.
3. Disconnected from the peripheral.
4. Reconnected until logged `peer_ota_addr` has chaged.
5. The bond is no longer valid, and trying to read the encrypted characteristic will force paring.

I'm using nrf Connect app on my phone, but i assume any app will be the same?

### More Information.

Work-around (or is it a fix?):

Make sure identity address is exchanged:

```
    ble_hs_cfg.sm_our_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
    ble_hs_cfg.sm_their_key_dist = BLE_SM_PAIR_KEY_DIST_ENC | BLE_SM_PAIR_KEY_DIST_ID;
```

Then NimBLE must be able to use that to track the bond information. 

Could we please NimBLE examples to work correctly with RPA peers? Took me a long time to figure out what was going on, and I'd expect this to impact anyone else using the examples to connect to a phone.

IshaESP, 2023-03-20:
Hi @braiden,
A fix has been raised for the same. 
BLE_SM_PAIR_KEY_DIST_ID flag is used for exchanging IRK(a key used for resolving address). Everytime phone was changing it's address, esp32c3 without this flag was treating it as a new device thus causing forced pairing when encrypted characteristic was being read.

rahult-github, 2023-04-07:
Changes committed in master [tree](https://github.com/espressif/esp-idf/commit/720af16ff699db0d4dd4b0ee69c92a8a10c0130b). Closing.

closed 2023-04-07 by rahult-github