espressif/esp-idf #10906: espefuse broke my S3

chegewara, 2023-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have burn `USB_PHY_SEL` before on some of my S3 and it worked fine, but now one of my devices has been bricked.
Yes, its partially my fault, because i ignored the fact `BLOCK0` is also being changed, but i cant see how it passed espressif tests.
```
espefuse.py -p /dev/ttyACM0 burn_efuse USB_PHY_SEL
espefuse.py v4.5
Connecting...
Detecting chip type... ESP32-S3

=== Run "burn_efuse" command ===
The efuses to burn:
  from BLOCK0
     - USB_PHY_SEL

Burning efuses:

    - 'USB_PHY_SEL' (Select internal/external PHY for USB OTGand usb_serial_jtag) 0b0 -> 0b1

Check all blocks for burn...
idx, BLOCK_NAME,          Conclusion
[00] BLOCK0               is not empty
	(written ): 0x0000000080000100000000000000d1f50000000000000000
	(to write): 0x000000000000000002000000000000000000000000000000
	(coding scheme = NONE)
. 
This is an irreversible operation!
Type 'BURN' (all capitals) to continue.
```

Now my device is no longer visible as any port, as i expected (old device with burned that efuse is recognized).
I have flashed some code with tinyusb on bricked device and i can see that port, so USB is still working.
I hope i still can use UART to flash it, but DFU is no longer an option on that device.

_If its my fault, or i missed something then apologize._

Thanks

igrr, 2023-03-04:
Could you please attach the output of `espefuse.py summary`?

chegewara, 2023-03-04:
Hi @igrr 
i cant, because linux does not see this device anymore (windows too), when i am trying to enable bootloader mode.
I can only see tinyUSB app, but unlycky to me its MSC + winUSB (no CDC).

igrr, 2023-03-04:
Do you still have access to UART pins?

I think the issue you are running into is that USB CDC download mode was disabled on this chip by eFuse.

There is more information available here: https://www.espressif.com/sites/default/files/advisory_downloads/AR2022-004%20Security%20Advisory%20for%20USB_OTG%20%26%20USB_Serial_JTAG%20Download%20Functions%20of%20ESP32-S3%20Series%20Products%20EN.pdf

> For ESP32-S3 series chips manufactured on and after Date Code 2219 and modules and development boards with the PW No. of and after PW-2022-06-XXXX, the bit (BLK0 B19[7]) will be open for users to program since it will not be programmed by default. This will enable the USB_OTG Download function.

That means that for chips and modules until that date, USB_CDC download mode was disabled.

Burning the USB_PHY_SEL eFuse bit has switched the USB peripheral from USB_SERIAL_JTAG to USB_CDC so now unfortunately you can't access the download mode over USB.

I'm afraid espefuse.py has no way of knowing whether this final state is intentional or not. The same eFuse settings could be desirable if you were locking down the product for security reasons, to prevent further access to USB download mode.

chegewara, 2023-03-04:
Thanks for answer.
I thought you are asking about current state of efuses. I have efuses from that device from before burning USB_PHY_SEL and you are correct. I missed it:
```
>python -m espefuse  -p com3 summary
espefuse.py v4.5.1
Connecting...
Failed to get PID of a device on com3, using standard reset sequence.

Detecting chip type... ESP32-S3

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                           ??? Temperature calibration                        = 9.700000000000001 R/W (0b001100001)
ADC1_MODE0_D2 (BLOCK2)                             ??? ADC1 calibration 1                             = -292 R/W (0xc9)
ADC1_MODE1_D2 (BLOCK2)                             ??? ADC1 calibration 2                             = 212 R/W (0x35)
ADC1_MODE2_D2 (BLOCK2)                             ??? ADC1 calibration 3                             = -208 R/W (0xb4)
ADC1_MODE3_D2 (BLOCK2)                             ??? ADC1 calibration 4                             = 232 R/W (0x3a)
ADC2_MODE0_D2 (BLOCK2)                             ??? ADC2 calibration 5                             = -312 R/W (0xce)
ADC2_MODE1_D2 (BLOCK2)                             ??? ADC2 calibration 6                             = -320 R/W (0xd0)
ADC2_MODE2_D2 (BLOCK2)                             ??? ADC2 calibration 7                             = -324 R/W (0xd1)
ADC2_MODE3_D2 (BLOCK2)                             ??? ADC2 calibration 8                             = -440 R/W (0xee)
ADC1_MODE0_D1 (BLOCK2)                             ??? ADC1 calibration 9                             = -64 R/W (0b110000)
ADC1_MODE1_D1 (BLOCK2)                             ??? ADC1 calibration 10                            = -44 R/W (0b101011)
ADC1_MODE2_D1 (BLOCK2)                             ??? ADC1 calibration 11                            = 48 R/W (0b001100)
ADC1_MODE3_D1 (BLOCK2)                             ??? ADC1 calibration 12                            = -108 R/W (0b111011)
ADC2_MODE0_D1 (BLOCK2)                             ??? ADC2 calibration 13                            = -56 R/W (0b101110)
ADC2_MODE1_D1 (BLOCK2)                             ??? ADC2 calibration 14                            = 80 R/W (0b010100)
ADC2_MODE2_D1 (BLOCK2)                             ??? ADC2 calibration 15                            = -36 R/W (0b101001)
ADC2_MODE3_D1 (BLOCK2)                             ??? ADC2 calibration 16                            = 72 R/W (0b010010)

Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Disables USB-OTG download feature in UART download = True R/W (0b1)
                                                    boot mode
DISABLE_WAFER_VERSION_MAJOR (BLOCK0)               Disables check of wafer version major              = False R/W (0b0)
DISABLE_BLK_VERSION_MAJOR (BLOCK0)                 Disables check of blk version major                = False R/W (0b0)
BLOCK_USR_DATA (BLOCK3)                            User data
[... cut ...]
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD3P3_CPU R/W (0b0)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.
```
In that case its completely my fault and i believe the case is solved.

Thanks

closed 2023-03-04 by igrr