espressif/esp-idf #10907: What happened to Websockets?

physiii, 2023-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have several devices using websockets as a client and server - and not the client is broke

Why was the client removed from ESP-IDF examples and documentation in v5.0?

If it hasn't been removed, then were is the documentation like you have in 4.x here: https://docs.espressif.com/projects/esp-idf/en/v4.1/api-reference/protocols/esp_websocket_client.html 

You removed all the header files so we can not use the client libraries anymore. I am very curious to the rational for this?

igrr, 2023-03-04:
Websocket client was removed from IDF but can be [installed separately](https://components.espressif.com/components/espressif/esp_websocket_client), please refer to the [v5.0 migration guide](https://docs.espressif.com/projects/esp-idf/en/v5.0.1/esp32/migration-guides/release-5.x/removed-components.html#components-moved-to-idf-component-registry).

The code is now located [here](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client) and the docs [here](https://espressif.github.io/esp-protocols/esp_websocket_client/index.html).

As for the rationale behind the removal, it was primarily to decouple the release cycles of these components from the (long) release cycles of ESP-IDF, making it possible to release new features and bug fixes faster.

physiii, 2023-03-04:
Thank you Igrr - I appreciate what you do here.

closed 2023-03-04 by physiii