espressif/esp-idf #10908: HTTP客户端打开新连接失败

CYCHEN97, 2023-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

HTTP客户端从网络获取数据

### What is the actual behavior?

HTTP客户端打开连接失败

### Steps to reproduce.

使用了一个web server，进入100多次就会出现web server没办法访问，但是没有看到错误代码输出；这个时候HTTP客户端如果去获取数据，就会出现HTTP客户端打开连接失败

### Debug Logs.

```plain
[0;31mE (3497471) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)[0m
[0;31mE (3497481) TRANSPORT_BASE: Failed to open a new connection: 32770[0m
[0;31mE (3497481) HTTP_CLIENT: Connection failed, sock < 0[0m
[0;31mE (3497491) HTTP_CLIENT: HTTP GET request failed: ESP_ERR_HTTP_CONNECT[0m
[0;32mI (3497501) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED[0m
[0;32mI (3497501) HTTP_CLIENT: Last esp error code: 0x8002[0m
[0;32mI (3497511) HTTP_CLIENT: Last mbedtls failure: 0x0[0m
```

### More Information.

没有

CYCHEN97, 2023-03-04:
访问web server时偶尔出现下面的error
[0;33mW (418881) httpd_txrx: httpd_sock_err: error in send : 104[0m
[0;33mW (418881) httpd_txrx: httpd_sock_err: error in send : 104[0m
[0;33mW (418891) httpd_txrx: httpd_sock_err: error in recv : 104[0m

CYCHEN97, 2023-03-05:
@ESP-YJM

ESP-YJM, 2023-03-06:
@CYCHEN97 看起来 socket 分配失败了。你可以打开 LWIP 的 SOCKETS_DEBUG 和在打印 `esp-tls: Failed to create socket` 的地方打印下 `errno` 的值，看看是 conn 分配失败还是 socket 分配失败。另外，服务器端打印 `104` 是客户端发送了 RST。可能的原因是客户端关了 socket 链接，服务器端还有未发送完成的数据发往客户端，客户端回了 RST。这个可以忽略。

ESP-YJM, 2023-03-09:
@CYCHEN97 看起来是你 Http server 用光了 socket。你可以参考这个 https://github.com/espressif/esp-idf/issues/10519。主要的思路就是：1. 使能 tcp keep-alive 机制检测僵尸客户端来释放 socket；2. 使能 TCP so_linger，关闭 socket 时，直接发 RST 给对端，避免 TCP 进入 Time-wait 状态。

CYCHEN97, 2023-03-09:
@ESP-YJM 参考他的修改了之后还是有问题，这次连强制门户配网都不会自动弹出来，还有web server的网页也出现了问题，这是测试log
[SaveWindows2023_3_10_星期五_2-06-32.TXT](https://github.com/espressif/esp-idf/files/10934704/SaveWindows2023_3_10_._2-06-32.TXT)

ESP-YJM, 2023-03-10:
```
[0;32mI (1653) file_server: Starting HTTP Server on port: '80'[0m
[0;31mE (1663) httpd: Configuration option max_open_sockets is too large (max allowed 13)
	Either decrease this or configure LWIP_MAX_SOCKETS to a larger value[0m
[0;31mE (1673) file_server: Failed to start file server![0m
```
你这个有报错，max_open_sockets 配的太大了。

[image]
另外，你的 socket 都被消耗完了，有一个客户端一直在连接你自己创建的 TCP server，端口 80，对端并且没有关闭 socket 导致。你需要在你 TCP server 上开启 socket keep-alive option.

CYCHEN97, 2023-03-10:
https://blog.csdn.net/xh870189248/article/details/102892766
我参考了这个来做的门户强制认证用来自动打开浏览器，创建了udp服务器和tcp服务器，然后本来就有自己的web server，还有一些获取网络数据的http客户端，是不是这些都要开启这个 socket keep-alive option.

ESP-YJM, 2023-03-10:
使用 TCP 的就需要，要不然客户端不调用 close 关闭 socket，就会形成僵尸客户端，占用 服务端 socket 数量。

CYCHEN97, 2023-03-10:
好的，今晚回去试试，谢谢大佬

CYCHEN97, 2023-03-10:
@ESP-YJM tcp开启这个 socket keep-alive option.跟上面web server开启的方法是不是一样的？

ESP-YJM, 2023-03-10:
原理上都是一样的，针对 accpet 后返回的 socket 进行 setsocketopt。

CYCHEN97, 2023-03-10:
好的好的

CYCHEN97, 2023-03-10:
> 
@ESP-YJM 
max_open_sockets 配置为12后就可以访问了，但是我用Magic 4手机连接wifi时，强制门户的速度很慢，也没有自动进入网页，过了一会才会在手机消息栏提示登录到WLAN网络，然后我点击了进入时，网络竟然被自动断开了。下面这个是log
[SaveWindows2023_3_11_星期六_0-35-16.TXT](https://github.com/espressif/esp-idf/files/10944141/SaveWindows2023_3_11_._0-35-16.TXT)

但是同样的操作在一部旧手机上就可以实现流畅的强制门户自动弹出网页的操作，而且每次都能够自动弹窗，这让我很不理解。

CYCHEN97, 2023-03-10:
@ESP-YJM 
我用小米NOTE3和华为NOVE6都测试了100次连接wifi的强制门户，都是可以秒弹出网页的，荣耀magic4测试时，速度就很慢，经常性出现wifi连接断开，请问这个是手机问题还是ESP32这边的问题？
使用浏览器访问web server时，会请求"/favicon.ico"和"/"两个路径，如果我给"/favicon.ico"返回的是图片，结果在直接显示在正中间了，之前还没增加自动关闭 socket 时，传图片就是传图片，传网页就是传网页，现在的现象就是只传递了图片，没有传网页了，我现在暂时是直接给"/favicon.ico"传了我的网页过去显示网页的，请问这个怎么操作才能在浏览器正常传递图片和网页？

CYCHEN97, 2023-03-13:
@ESP-YJM 上班啦

ESP-YJM, 2023-03-13:
> @ESP-YJM max_open_sockets 配置为12后就可以访问了，但是我用Magic 4手机连接wifi时，强制门户的速度很慢，也没有自动进入网页，过了一会才会在手机消息栏提示登录到WLAN网络，然后我点击了进入时，网络竟然被自动断开了。下面这个是log [...]

从日志看，是手机主动断开 wifi 导致的。一般有些手机连上 wifi 后会判断是否有网络连接，如果没有，会断开 wifi。一般手机上会有选项，类似没有网络连接的话，还是保持当前 wifi 连接，你看看有没有。

ESP-YJM, 2023-03-13:
> @ESP-YJM 我用小米NOTE3和华为NOVE6都测试了100次连接wifi的强制门户，都是可以秒弹出网页的，荣耀magic4测试时，速度就很慢，经常性出现wifi连接断开，请问这个是手机问题还是ESP32这边的问题？ 使用浏览器访问web [...]

你可以参考这个示例 https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving/main

CYCHEN97, 2023-03-13:
@ESP-YJM 我就是参考这个示例的，但是现在我们不是增加了这个关闭socket的功能嘛，就出问题了

CYCHEN97, 2023-03-13:
> 从日志看，是手机主动断开 wifi 导致的。一般有些手机连上 wifi 后会判断是否有网络连接，如果没有，会断开 wifi。一般手机上会有选项，类似没有网络连接的话，还是保持当前 wifi 连接，你看看有没有。

有时候wifi连接没断开，这个手机的强制门户也是很慢的，这个是手机问题还是esp32这边的问题呢？

ESP-YJM, 2023-03-13:
> @ESP-YJM 我就是参考这个示例的，但是现在我们不是增加了这个关闭socket的功能嘛，就出问题了

你怎么关闭 socket 的？

ESP-YJM, 2023-03-13:
> 有时候wifi连接没断开，这个手机的强制门户也是很慢的，这个是手机问题还是esp32这边的问题呢？

这个不是很好确定，要抓空中包看下。

CYCHEN97, 2023-03-13:
> @CYCHEN97 看起来是你 Http server 用光了 socket。你可以参考这个 #10519 。 主要的思路就是：1. 使能 tcp keep-alive 机制检测僵尸客户端来释放 socket；2. 使能 TCP so_linger，关闭 socket 时，直接发 RST 给对端，避免 TCP 进入 Time-wait 状态。
@ESP-YJM 就是增加了这个就出现的

CYCHEN97, 2023-03-13:
是不是一个http请求完"/favicon.ico"后网页的socket就被关闭了

CYCHEN97, 2023-03-13:
@ESP-YJM

ESP-YJM, 2023-03-13:
@CYCHEN97 正常来说，加了 SO_LINGER 只是会在调用 close 后发 RST 断开连接，不应该会出现加了 SO_LINGER 后才会出现，如果有问题，不加 SO_LINGER 也会有问题。你抓个 HTTP  交互包看看呢。

CYCHEN97, 2023-04-02:
@ESP-YJM 大佬你好，我的esp32开启了http服务器和tcp服务器的 socket keep-alive option后，还是出现http客户端连接异常的现象，而且设备重新启动后还是异常，需要我清除整个flash后重新烧录才能正常运行了，请问这个是什么原因造成的？下面这个是log，麻烦大佬帮忙看看。
[SaveWindows2023_4_2_星期日_17-36-16.TXT](https://github.com/espressif/esp-idf/files/11131733/SaveWindows2023_4_2_._17-36-16.TXT)
在monitor里面看是这样的，但是我还是找不到原因
[image]

CYCHEN97, 2023-04-02:
这是monitor的完整log
[monitor.txt](https://github.com/espressif/esp-idf/files/11131745/monitor.txt)

CYCHEN97, 2023-04-03:
@ESP-YJM 麻烦大佬帮忙看下，谢谢大佬

CYCHEN97, 2023-04-03:
好的，我同时测试两台，然后都用http客户端去访问http服务器，测试了几次后，竟然在差不多同一时间出现了异常，就是这样子 的访问非法地址，不知道会不会是解析JSON出错了，然后把一些内存修改了，但是这样在差不多同一时间出现，这让我很匪夷所思，而且我清空flash后重新烧录，还是访问那个http服务器获取数据，又是正常的了，可以给我提供一些检查代码的思路吗？

CYCHEN97, 2023-04-03:
@ESP-YJM

CYCHEN97, 2023-04-03:
> @ESP-YJM 大佬你好，我的esp32开启了http服务器和tcp服务器的 socket keep-alive option后，还是出现http客户端连接异常的现象，而且设备重新启动后还是异常，需要我清除整个flash后重新烧录才能正常运行了，请问这个是什么原因造成的？下面这个是log，麻烦大佬帮忙看看。 [...]

@ESP-YJM 而且你看这个，他死掉的时候，异常都不在应用层这边呢

ESP-YJM, 2023-04-03:
应该是内存被破坏了，你可以先把你加的应用代码一点一点注释，看看是哪个导致的。

CYCHEN97, 2023-04-03:
好的，我测试下

CYCHEN97, 2023-04-05:
@ESP-YJM 你好，我测试了，问题应该是在http获取数据那里。我通过web服务器设置一些参数让它去做http请求获取数据，设置了20多次之后，就出现问题了，设置成功了，但是就是HTTP那里机器出现了重启现象。我用两部手机同时分别设置两台机器，设置次数是一样的，重启的次数也是一样的。
出现重启现象之后，可以看到http客户端请求不再成功，而且会在http请求时又继续出现重启现象，log显示
httpd_uri: httpd_uri: uri handler execution failed

[01:17:46.705]收←◆esp-tls: couldn't get hostname for :d1.weather.com.cn: getaddrinfo() returns 202, addrinfo=0x0[0m
[0;31mE (1942) TRANSPORT_BASE: Failed to open a new connection: 32769[0m
[0;31mE (1952) HTTP_CLIENT: Connection failed, sock < 0[0m
[0;31mE (1952) HTTP_CLIENT: HTTP GET request failed: ESP_ERR_HTTP_CONNECT[0m
[0;32mI (1962) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED[0m
[0;32mI (1962) HTTP_CLIENT: Last esp error code: 0x8001[0m
[0;32mI (1972) HTTP_CLIENT: Last mbedtls failure: 0x0

下面是其中一台的log
[SaveWindows2023_4_6_星期四_1-18-01.TXT](https://github.com/espressif/esp-idf/files/11161490/SaveWindows2023_4_6_._1-18-01.TXT)
以上，麻烦大佬再帮我看看，非常谢谢大佬

CYCHEN97, 2023-04-05:
出现重启现象后我用monitor看的log，都是在HTTP请求后重启的。
[image]

CYCHEN97, 2023-04-06:
@ESP-YJM

ESP-YJM, 2023-04-06:
你能提供复现的代码与步骤吗？

CYCHEN97, 2023-04-06:
@ESP-YJM 可以私密提供给你吗？我今晚把我整个工程打包给你看看

CYCHEN97, 2023-04-06:
@ESP-YJM 或者能不能有个微信或者啥的我直接发给你看下

CYCHEN97, 2023-04-06:
@ESP-YJM

ESP-YJM, 2023-04-06:
@CYCHEN97 你可以创建个私有仓库，然后邀请我就行。

CYCHEN97, 2023-04-06:
@ESP-YJM 好的，那我今晚@你看下哈，我下班才回家搞的

CYCHEN97, 2023-04-06:
@ESP-YJM 你好，我已经邀请你进入我的项目了，麻烦帮忙看下

closed 2023-05-22 by Alvin1Zhang