espressif/esp-idf #10909: RGB LCD Panel Example: unknown type name 'esp_lcd_rgb_panel_event_data_t'

w3llschmidt, 2023-03-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3710-gacac972

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should compile

### What is the actual behavior?

Breaks with error: **unknown type name 'esp_lcd_rgb_panel_event_data_t'**

### Steps to reproduce.

1. idf.py -p /dev/ttyUSB0 build flash monitor

### Build or installation Logs.

```plain
/home/h/Documents/rgb_panel/main/rgb_lcd_example_main.c:71:72: error: unknown type name 'esp_lcd_rgb_panel_event_data_t'
   71 | static bool example_on_vsync_event(esp_lcd_panel_handle_t panel, const esp_lcd_rgb_panel_event_data_t *event_data, void *user_data)
      |                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/h/Documents/rgb_panel/main/rgb_lcd_example_main.c: In function 'app_main':
/home/h/Documents/rgb_panel/main/rgb_lcd_example_main.c:128:5: error: unknown type name 'esp_lcd_rgb_panel_config_t'; did you mean 'esp_lcd_panel_handle_t'?
  128 |     esp_lcd_rgb_panel_config_t panel_config = {
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |     esp_lcd_panel_handle_t
```

### More Information.

This is my hardware: [https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-4-3-inch.html](url)

suda-morris, 2023-03-05:
Hi @w3llschmidt The RGB LCD example can only run on esp32s3, if you have read the README:

[image]

Maybe you forgot to run the `idf.py set-target esp32s3` before the `idf.py build`?  https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html#configure-your-project

w3llschmidt, 2023-03-05:
@suda-morris - yes, **esp32s3**, this is what i have here. I did also a **idf.py set-target esp32s3**

https://www.makerfabs.com/esp32-s3-parallel-tft-with-touch-4-3-inch.html

- Controller: ESP32-S3-WROOM-1, PCB Antenna, 16MB Flash, 8MB PSRAM

w3llschmidt, 2023-03-05:
I did a fresh install! It complies now!

Thx!!!

closed 2023-03-05 by w3llschmidt