espressif/esp-idf #10910: Memprot feature locked after the system reset

Qinka, 2023-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I write my program in Rust, I  reset the board with `esp_idf_hal::reset::restart`, which call [`esp_restart`](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0.1/esp32c3/api-reference/system/misc_system_api.html#_CPPv411esp_restartv).

But I got error:
```
E (2184) cpu_start: Memprot feature locked after the system reset! Potential safety corruption, rebooting.
```

I want to know, what should I do before and after reset call, to eliminate such a error.

I use esp-idf-sys and esp-idf-hal with esp-idf 5.0.1.

full log after restart
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
configspi:0x000000b9
SPIWP:0x00
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
load:0x403d0000,len:0x2ce0
entry 0x403ce000
I (4294967295) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (4294967295) boot: compile time 12:10:40
I (4294967295) boot: chip revision: 3
I (4294967295) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (4294967295) boot.esp32c3: SPI Speed      : 80MHz
I (4294967295) boot.esp32c3: SPI Mode       : DIO
I (4294967295) boot.esp32c3: SPI Flash Size : 4MB
I (4294967295) boot: Enabling RNG early entropy source...
I (4294967295) boot: Partition Table:
I (4294967295) boot: ## Label            Usage          Type ST Offset   Length
I (4294967295) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (4294967295) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (4294967295) boot:  2 factory          factory app      00 00 00010000 00200000
I (4294967295) boot:  3 config           Unknown data     01 82 00210000 00040000
I (4294967295) boot: End of partition table
I (4294967295) boot_comm: chip revision: 3, min. application chip revision: 0
I (4294967295) esp_image: segment 0: paddr=00010020 vaddr=3c0f0020 size=49c90h (302224) map
I (4294967295) esp_image: segment 1: paddr=00059cb8 vaddr=3fc90a00 size=024cch (  9420) load
I (4294967295) esp_image: segment 2: paddr=0005c18c vaddr=40380000 size=03e8ch ( 16012) load
I (4294967295) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=e7f3ch (950076) map
I (4294967295) esp_image: segment 4: paddr=00147f64 vaddr=40383e8c size=0c9c4h ( 51652) load
I (4294967295) boot: Loaded app from partition at offset 0x10000
I (4294967295) boot: Disabling RNG early entropy source...
I (4294967295) cpu_start: Pro cpu up.
E (2252) cpu_start: Memprot feature locked after the system reset! Potential safety corruption, rebooting.
```

mahavirj, 2023-03-08:
@Qinka 

This issue should not occur if you are calling `esp_restart` API. I would suggest that you please look out through following code for digital reset assertion in your setup once:

https://github.com/espressif/esp-idf/blob/afbdb0f3ef195ab51690a64e22bfb8a5cd487914/components/esp_system/esp_system.c#L67-L80

mahavirj, 2023-03-24:
Closing due to lack of update. Please feel free to share reproducible code and I will take a look..

pacucha42, 2023-03-24:
@mahavirj - your comment is absolutely right, the Memory protection setup is locked after being set during the system startup, and the only way to unlock is to call esp_restart_noos_dig(). This is handled  by esp_restart() internally so the reported issue should never happen unless somebody tinkered with the Memprot setup.

@Qinka - could you please supply Verbose log of your application? And possibly describe more of its background?

pacucha42, 2023-03-30:
Closing as there hasn't been any recent update. Please, reopen if needed

closed 2023-03-30 by pacucha42