espressif/esp-idf #10911: tud_cdc_n_read() can only read 64 bytes at a time

nopnop2002, 2023-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I change these CONFIG:
```
#CONFIG_TINYUSB_CDC_RX_BUFSIZE=64
#CONFIG_TINYUSB_CDC_TX_BUFSIZE=64
CONFIG_TINYUSB_CDC_RX_BUFSIZE=256
CONFIG_TINYUSB_CDC_TX_BUFSIZE=256
```

I sent 70 bytes at a time via USB.

However, tud_cdc_n_read() can only read 64 bytes at a time.

I added some logging in espressif__esp_tinyusb/additions/src/tusb_cdc_acm.c
```
        int read_res = tud_cdc_n_read(  itf,
                                        acm->rx_tfbuf,
                                        CONFIG_TINYUSB_CDC_RX_BUFSIZE );
        ESP_LOGI(__FUNCTION__, "tud_cdc_n_read CONFIG_TINYUSB_CDC_RX_BUFSIZE=%d read_res=%d", CONFIG_TINYUSB_CDC_RX_BUFSIZE, read_res);
        int res = xRingbufferSend(acm->rx_unread_buf,
                                  acm->rx_tfbuf,
                                  read_res, 0);
```

This was displayed.
```
I (20664) tud_cdc_rx_cb: tud_cdc_n_read CONFIG_TINYUSB_CDC_RX_BUFSIZE=256 read_res=64
```

tore-espressif, 2023-04-04:
Hello @nopnop2002 , thank you for the report, I could reproduce the issue.

Due to some historical reasons, there are multiple buffers in the CDC driver.

TinyUSB is always preparing an OUT transfer with size of MPS (64 bytes in this case) even when you configure larger `CONFIG_TINYUSB_CDC_RX_BUFSIZE`.  So, you will get a RX callback after every packet. https://github.com/hathach/tinyusb/blob/master/src/class/cdc/cdc_device.c#L100

I could get larger transfers done by defining `CFG_TUD_CDC_EP_BUFSIZE = 256` and setting `rx_unread_buf_sz = 256,` in `tinyusb_config_cdcacm_t`. You will have to be careful, because `CFG_TUD_CDC_EP_BUFSIZE` is also used in EP descriptor, where max value is 64 for FullSpeed devices.

We'll provide a fix in upstream TinyUSB repo

nopnop2002, 2023-04-04:
Thank you for contacting.
i will wait for a fix.

tore-espressif, 2023-07-11:
Hello @nopnop2002 I'm sorry for such a long delay in response.

After more testing I can offer this workaround and explanation:

Explanation: Not all OSes end USB transfers with zero-length-packet when the transfer_length==maximum_packet_size (64), which makes it difficult for TinyUSB to detect end of transfer. So, TinyUSB has a safe implementation where every packet reception is forwarded to the user, even when more data is expected. (e.g. transfer with length 65 bytes will trigger two interrupts: one with 64 bytes received and second with 1 byte received).

In case you read RX data from fifo in RX callback, you will read 64 bytes (the rest has not been received yet).

In case you read RX data in another task eg
``` c
uint8_t buf[CONFIG_TINYUSB_CDC_RX_BUFSIZE + 1];
while (true) {
    vTaskDelay(1);
    size_t rx_size = 0;
    ESP_ERROR_CHECK(tinyusb_cdcacm_read(0, buf, CONFIG_TINYUSB_CDC_RX_BUFSIZE, &rx_size));
    if (rx_size > 0) {
       printf("RX %d bytes\n", rx_size);
    }
}
```
you will read the full transfer, that is 65 bytes

nopnop2002, 2023-07-11:
Thanks for the suggestions.
I will try this suggestion.

closed 2023-09-25 by tore-espressif