espressif/esp-idf #10912: About example->Wifi->espnow

nopnop2002, 2023-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0.1-3-g886e98a2c1-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

espnow sample not working as expected.

### What is the actual behavior?

This is send data structure:
magic is uint32_t.

```
/* User defined field of ESPNOW data in this example. */
typedef struct {
    uint8_t type;                         //Broadcast or unicast ESPNOW data.
    uint8_t state;                        //Indicate that if has received broadcast ESPNOW data or not.
    uint16_t seq_num;                     //Sequence number of ESPNOW data.
    uint16_t crc;                         //CRC16 value of ESPNOW data.
    uint32_t magic;                       //Magic number which is used to determine which device to send unicast ESPNOW data.
    uint8_t payload[0];                   //Real payload of ESPNOW data.
} __attribute__((packed)) example_espnow_data_t;
```

```magic```  variable on the receiving side is an int, and the sent Magic number cannot be stored correctly.
https://github.com/espressif/esp-idf/blob/master/examples/wifi/espnow/main/espnow_example_main.c#L155

### Steps to reproduce.

1. Execute [this](https://github.com/espressif/esp-idf/tree/master/examples/wifi/espnow)

nopnop2002, 2023-03-06:
I found one more bug.
If we use a channel other than channel 1, you need the following code when initializing WiFi.
```
    /* Change WiFi primary channel */
    uint8_t primary;
    wifi_second_chan_t second;
    ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &second));
    ESP_LOGI(TAG, "current primary channel=%d", primary);
    primary = CONFIG_ESPNOW_CHANNEL;
    ESP_ERROR_CHECK(esp_wifi_set_channel(primary, WIFI_SECOND_CHAN_NONE));
    ESP_ERROR_CHECK(esp_wifi_get_channel(&primary, &second));
    ESP_LOGI(TAG, "new primary channel=%d", primary);
```

zhangyanjiaoesp, 2024-03-26:
@nopnop2002 thanks for reporting, about the `magic`, we will update the example. About the channel issue, the https://github.com/espressif/esp-idf/commit/62d4adb482bf4a63b21adaef394dcc60e0a529bb commit has solved the problem, please update your IDF version.

closed 2024-04-04 by espressif-bot