espressif/esp-idf #10913: FATFS uses buffers not optimally

BOBAH1248, 2023-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone!
I found issue with FF_MAX_SS

in [struct FIL](https://github.com/espressif/esp-idf/blob/698752684e3e15c790f887523d816010da96fd24/components/fatfs/src/ff.h#L217-L219) and probably in [struct FATFS](https://github.com/espressif/esp-idf/blob/698752684e3e15c790f887523d816010da96fd24/components/fatfs/src/ff.h#L173)

By default, the FATFS configured for worst use case, i.e.:
- different storages (SPI-flash, SD-card and so)
- few partitions per storage
- optimized for fast file IO (i.e. has buffers per "file")

This configuration also used for [arduino-esp32](https://github.com/espressif/arduino-esp32) repository.
So, FF_MIN_SS = 512, FF_MAX_SS = 4096, FF_FS_TINY = 0
And FATFS will allocate buffers on 4096, but will use dependent on [storage type](https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/components/fatfs/diskio/diskio_sdmmc.c#L101-L103), that is not optimal.
For example, for SD-card (i.e. with sector size is 512 bytes), buffers used on 12.5% (512 of 4096).

But when used number of files, the situation is worse (going back to `arduino-esp32`): when used 5 files, we get 20kB heap allocation instead of the expected 2.5kB (8 times more).

PS: To tell the truth, I don't know where exactly to address this issue (esp-idf, arduino-esp32 of [fatts](http://elm-chan.org/fsw/ff/bd/?show=3725)) - the fatfs library has the described issue, but it is associated with non-typical use (i.e. where this library is precompiled for the worst case, but only the simplest case is used).
So, I also posted description of issue on fatfs forum http://elm-chan.org/fsw/ff/bd/?show=3725

igrr, 2023-03-05:
Hi @BOBAH1248, 
This shouldn't be the case, as FATFS normally asks the diskio layer about the sector size of the concrete underlying implementation, and allocates the buffers accordingly. Could you please provide instructions for reproducing the issue you observe?

BOBAH1248, 2023-03-05:
Hello @igrr and thanks for prompt feedback, despite the weekend.

> Edit: my bad, i see now what you mean. The buffer inside FIL structure is indeed [fixed](https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/components/fatfs/src/ff.h#L218).

Yes, exactly =)

igrr, 2023-03-05:
The simplest solution for Arduino-esp32 would be to set the wear levelling library sector size to 512 bytes. However this would be a breaking change, as the existing partitions would become unreadable.

Another slightly more complex option is to provide a custom diskio implementation in Arduino-esp32, which would pretend that the SD sector size is 4kB. The heap usage would not improve, but at least the space would not be wasted.

A more flexible solution would indeed require changes in the upstream library...

BOBAH1248, 2023-03-05:
Agreed
For this reason, I wrote my "PS" in description - this issue has number of ways to solve on any of 3 levels: fatfs, arduino-esp32 and esp-idf
But only fix of fatfs should get the best result (adequate heap usage per storage)

chipweinberger, 2023-03-06:
The FATFS library author is pretty open to improvements from what I can tell.

BOBAH1248, 2023-03-19:
I did not have time to deal with this issue, so in my project I simply reduced the number of files from 5 (default) to 2 (its required minimum for my project).

Now I have finally taken the time to deal with this issue.
And what I was able to find:

### 1. the problem of wasted buffers is not related to the FatFS library

Yes, it declares structures with a fixed size of buffers, but it does so only in an "informative form" and does not deal with memory allocation (at all!)

### 2. The actual allocation of memory occurs in the `vfs_fat.c` of this (esp-idf) framework

Allocation made by single pass in [`esp_vfs_fat_register`](https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat.c#L155-L156)
This line not only allocates memory for the [file system](https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat.c#L24), but also buffers for the [files](https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat.c#L28) themselves.
After that, the calling code (`arduino-esp32` in my case) receives a pointer to the FATFS structure ("Filesystem object structure") and must prepare FatFS library for work on its own (the `arduino-esp32` library mounts and, if necessary, formats the partition).
In any case, subsequent file I\O operations occur [using the buffers](https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat.c#L341) allocated inside `esp_vfs_fat_register`.

Of course, the sector size is not passed to the `esp_vfs_fat_register` function and looks that impossible to calculate required buffer size. But the `ff_diskio_register` already called and pointer on the `ioctl` function is available. So, it can be called for the `GET_SECTOR_SIZE` command result.

### Thus,
there is no point in waiting for the FATFS library author to fix the problem - the library only defined/offered the maximal safe buffer size for any cases.
Allocation did this `esp-idf` framework - it allocated memory for all required structs.

igrr, 2023-03-19:
If the FIL structure contains a buffer declared as having MAX_SS size, i think it would be pretty dangerous to actually allocate less memory for the structure:

- copying the structure by an assignment would no longer be possible (or rather, it would silently corrupt heap which follows the structure.)
- if FATFS library ever accesses the part of the buffer beyond the actual sector size, it would be legal from the perspective of C language but would cause a heap buffer overrun.

So while yes, i do agree that it is technically possible to allocate less than sizeof(FIL) for FIL structure, i don't think it is a safe thing to do. The better solution would be if the file buffer inside struct FIL was changed into a pointer.

BOBAH1248, 2023-03-19:
To reduce the complexity of [`fatfs` isp-idf component](https://github.com/espressif/esp-idf/tree/master/components/fatfs), it will be better to replace the buffers array with pointers in `fatfs` library. And allocate memory, using multiple `ff_memalloc` calls.
But do it not only for [FIL](https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/ff.h#L218), but [FATFS](https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/ff.h#L173) structs (because the FATFS is in middle of the top-level [`vfs_fat_ctx_t`](https://github.com/espressif/esp-idf/blob/master/components/fatfs/vfs/vfs_fat.c#L24) structure).

BOBAH1248, 2023-03-19:
Any way, I made the workaround in my project without patching of esp-idf nor arduino-esp32 (because I use number of memory hungry libs that left around 5kB RAM for my code, while I must use `String` class that easily fragmented this free memory, the additional 20kB RAM is the same as find a treasure).

If someone wants to repeat my feat, the steps the following:
1. copy all `.c` files from [all subfolders](https://github.com/espressif/esp-idf/tree/master/components/fatfs) to the `src` folder of your project (exclude files "fatfs/src/diskio.c" and "fatfs/src/ffsystem.c")
2. copy all `.h` files as above, to the `include` folder of your project
3. patch the `ffconf.h` - [`#define FF_SS_WL          (FF_SS_SDCARD-1)      /* CONFIG_WL_SECTOR_SIZE */`](https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/ffconf.h#L214)
4. disable [`FF_MAX_SS` and `FF_MIN_SS` check](https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/ff.c#L259) in `ff.c` file
5. **Use this "workaround" in your project on your own risk**

Then build, flash the project and check the heap usage `heap_caps_get_info(&info, MALLOC_CAP_DEFAULT)` before and after this workaround.

PS: I use `heap_caps_get_info` with `MALLOC_CAP_DEFAULT` instead of `ESP.getFreeHeap` because `getFreeHeap` usually returns larger value, than can be allocated using `malloc` at this time.

_For those who are wondering why need to copy files to the `src` and `include` folder instead of creating new library (for example, just copy full folder as-is to `lib` folder in 'platformio' project):_
_- the library may not override another one (you may clearly see it in `.map` file) or will cause error at link-time (if partially replaced some functions)_
_- the `.o` (object) files has priority over `.a` (library/archive) files for linker (you may clearly see it in `.map` file)_

_Then `FF_SS_WL` defined less than `FF_SS_SDCARD`, to use the [`FF_MAX_SS != FF_MIN_SS`](https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/ff.h#L142-L144) and `ioctl(GET_SECTOR_SIZE)` as before._

AxelLin, 2024-03-26:
@adokitkat This has been in Reviewing state for a while, how is the status now? (Why the status becomes Awaiting Response?)

adokitkat, 2024-03-31:
@AxelLin Hi, I am no assignee of this issue (see our internal JIRA, the change was not reflected here). However there is a MR which should solve this issue linked there made by our colleague and is nearly finished.

closed 2024-04-23 by espressif-bot