espressif/esp-idf #10914: ESP32C6 code execution freeze

RuslanSpartak, 2023-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32c6 module

### Power Supply used.

External 5V

### What is the expected behavior?

code  execution freeze after startup.

### What is the actual behavior?

code execution freeze

### Steps to reproduce.

1. Use spi slave, blufi with network, mqtt and http clients
2. After startup code can freeze (if we don't use task watchdog)

### Debug Logs.

```plain
W (8568) wifi: [trc.c,1014] (trc)phytype:CBW20-SGI, snr:43, maxRate:144, highestRateIdx:0
W (8576) wifi: [trc.c,1104] (trc)rate(S-MCS7, schedIdx:0), ampdu(rate:S-MCS7, schedIdx(0, stop:8)), snr:43, ampduState:wait operational
W (8587) wifi: [wl_cnx.c,2909] ifidx:0, rssi:-52, nf:-95, phytype(0x3, CBW20-SGI), phymode(0x3, 11bgn), max_rate:1440, he:0
W (8598) wifi: [wl_cnx.c,2912] max ampdu length exponent:3(65535 bytes), mmss:6(8 us)
W (8614) wifi: [wl_cnx.c,3078] (extcap)mbssid:0, enhanced_mbssid_advertise:0, complete_nontxbssid_profiles:0
W (8615) wifi: [wl_cnx.c,3087] connected with Spinn_Office, aid = 1, channel 1, BW20, bssid = d8:b1:90:80:2f:20
W (8625) wifi: [wl_cnx.c,3095] cipher(pairwise:0x3, group:0x3), pmf:0, security:WPA2-PSK, phy:11bgn, rssi:-52
W (8635) wifi: [pm.c,3216] pm start, type: 1

W (8638) wifi: [pm.c,3219] pm start, type:1, aid:0x1, trans-BSSID:d8:b1:90:80:2f:20, BSSID[5]:0x20, mbssid(max-indicator:0, index:0), he:0
W (8651) wifi: [pm.c,3286] set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
W (8660) wifi: [ieee80211_ht.c,875] [ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
W (8673) wifi: [ieee80211_ht.c,875] [ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x30)
W (8685) wifi: [ieee80211_ht.c,875] [ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
W (8699) wifi: [hal_ampdu.c,94] <ba-add>idx:0, ifx:0, tid:0, TAHI:0x100202f, TALO:0x8090b1d8, (ssn:1218, win:64, cur_ssn:1218), CONF:0xc0000005
W (8711) wifi: [ieee80211_ht.c,971] [ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
W (8722) wifi: [ieee80211_ht.c,971] [ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
W (8733) wifi: [ieee80211_ht.c,971] [ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (8744) WiFi: SYSTEM_EVENT_STA_CONNECTED
```

Alvin1Zhang, 2023-03-06:
Thanks for reporting, seemingly a similar/same issue was created at https://github.com/espressif/esp-idf/issues/10898, will close that one and track the issue here. Feel free to reopen https://github.com/espressif/esp-idf/issues/10898 if that is a different issue. Thanks.

igrr, 2023-03-06:
Probably this wasn't closed intentionally, @Alvin1Zhang meant to close the other issue.

RuslanSpartak, 2023-03-06:
May I know how much time it can take? When we will have sdk 5.1 fully support for esp32c6?

igrr, 2023-03-06:
@RuslanSpartak For this specific issue, i'm not sure we have enough information to help you troubleshoot the problem. The log you have attached doesn't look abnormal, and we have not observed a similar issue in IDF tests or examples.

I would suggest making the issue more actionable by doing one or more of the following:
- Try to run a simpler version of your code, or one of the examples. See if adding or removing a particular feature makes the issue appear or go away.
- See if you can provide us with some minimal version of the code which we could run locally to reproduce the issue.
- Try to find what specifically is the issue that happens.
   - Is it just one task that doesn't run, or none of the tasks run?
   - If it's just one task, what is it blocked on?

Regarding the release timeline, IDF v5.1 is expected to be released in June.

RuslanSpartak, 2023-03-06:
I will ask my company about what I can share with you.
But examples that you test are as simple as they can be.
When we use few components together (mqtt, blufi with wifi, http client) I get that code just freeze at different points without any hard faults or anything specific things showing. I add tasks watchdog and I see that after freezing esp32c6 reboots. We have working code on ESP32-WROOM module. 
If it will help I can try to make small video. Will it help?

RuslanSpartak, 2023-03-06:
So copying settings from blufi project fix freezing code execution. Not sure what was wrong, but at least we can close that issue.

RuslanSpartak, 2023-03-06:
Copying settings helped

closed 2023-03-06 by RuslanSpartak

CarlosKiamy, 2023-04-21:
What do you mean by copying settings? My program also freezes randomly upon logging "W (25728) wifi: [hal_ampdu.c,45] <ba-add>idx:1 (ifx:0, MY_MAC_ADDRESS), tid:0, ssn:0, winSize:64"