espressif/esp-idf #10915: ESP32c3 Nimble (un)directed advertising with own_addr_t=rpa_pub

braiden, 2023-03-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-3778-g370e250072

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32c3

### Power Supply used.

USB

### What is the expected behavior?

When advertising with an RPA addr type my ota ADDR should be a random rotating RPA addr.

### What is the actual behavior?

The advertising address has the fixed value `7e:1e:1e:79:4f:f7`. I've verified this is the physically transmitted address with [nrf sniffer](https://www.nordicsemi.com/Products/Development-tools/nrf-sniffer-for-bluetooth-le)

There are several things wrong with this:

1. While the MSB to indicate this address is an RPA, it is not random. Specially the `prand` value is always `1e:1e:1e`. It does not change every 15 minutes despite nimble calling `BLE_HCI_OCF_LE_SET_RPA_TMO` [here](https://github.com/espressif/esp-nimble/blob/nimble-1.3.0-idf/nimble/host/src/ble_hs_pvcy.c#L182). Power-cycle, etc, also don't matter.

2. I don't know what IRK was used to hash this address, but its not using the irk returned by `ble_hs_pvcy_our_irk()` and shared by the controller when pairing. `7e:1e:1e:fc:8f:8b` would be the correctly hashed address for this IRK and prand. Default is hard-coded [here](https://github.com/espressif/esp-nimble/blob/nimble-1.3.0-idf/nimble/host/src/ble_hs_pvcy.c#L31).

3. Per [this](https://github.com/espressif/esp-nimble/blob/nimble-1.3.0-idf/nimble/host/src/ble_hs_pvcy.c#L254) the resolve list should already be setup such that advertising uses this irk.

### Steps to reproduce.

`examples/bluetooth/nimble/bleprph`. Change set `own_addr_type = BLE_OWN_ADDR_RPA_PUBLIC_DEFAULT.

I've tried both directed and undirect, `adv` and `ext_adv`.

Does the ESP32c3 actually support controller based RPA generated addresses?

chegewara, 2023-03-05:
From few years ago, on bluedroid stack, i remember you can set only `PUBLIC` or `PRIVATE` own address. You cant tell host you want it to be `RPA` or `STATIC`, because in `LL` packet there is only `TxAdd` bit flag, which means private or public.

braiden, 2023-03-06:
I don't think that can be case? When with privacy extensions enabled TxAdd=0 implies RPA? The physically captured broadcast package is absolutely and RPA address (i computed the address hash manually to verify).

_The Link Layer may use resolvable private addresses or non-resolvable private
addresses for the advertiser’s device address (AdvA field) when entering the
Advertising State and using connectable and scannable undirected events._

*BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 4, Part E, 7.8.5 LE Set Advertising Parameters command* provides 4 options for own address.

chegewara, 2023-03-06:
1. `TxAdd = 0` - means public address
2. I had to recolect what was the problem; well, beside of setting this value in bludroid i also had to call this function `esp_ble_gap_config_local_privacy()`
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gap_ble.html#_CPPv432esp_ble_gap_config_local_privacyb

Im not sure if similar function is used in nimble thou.

braiden, 2023-03-06:
Thanks for the suggestion. I'll take look. I'm assuming `esp_ble_gap_config_local_privacy` implies host mode privacy, but I'm specifically looking for controller based privacy on the esp32c3.

braiden, 2023-03-06:
Traced that bluedriod code path. It essentially does `HCI_LE_Set_Address_Resolution_-Enable`. The analogous call in NimBLE happens in  `ble_hs_pvcy_set_our_irk` → `ble_host_rpa_enabled`. I've already made sure this is happening. :(. Thanks for the tip though. I'm a a bluedroid user will find this thread.

braiden, 2023-03-10:
Since the controller is a binary blob I really need espressif's help for a fix.

- I've verified nimble enables address resolution, sets a timer for address rotation, and adds the correct irk to resolve list. 
- I've verified the controller generates a valid a valid RPA address with a correctly computed hash.

... but as I mention the RPA is neither random nor changes.

Vol 6, Part B, 6.1:

_The Link Layer shall set a timer determined by the Host. A new resolvable
private address shall be generated when the timer expires. If the Link Layer is
reset, a new resolvable private address shall be generated and the timer
started with any value in the allowed range._

As a terrible hack I've called `r_lld_rpa_renew()` symbol exported by the linker and implemented in rom(?), and have had some success getting RPA address to change, but this is clearly the wrong path.

braiden, 2023-03-10:
Most recent findings: three bugs one of them mine:

1. Most importantly: The RPA address changes only when advertising starts. After restarting adverting every 30 seconds the RPA began to rotate. I'm not able to decipher the BLE spec well enough to know for sure this is working-as-intended, but its at least good enough.
2. However the RPA rotates every 15 minutes ignoring `CONFIG_BT_NIMBLE_RPA_TIMEOUT` even though a timeout is set: `ble_hs_pvcy_ensure_started()` →  `ble_hs_pvcy_set_addr_timeout()` → `BLE_HCI_OCF_LE_SET_RPA_TMO`. Controller Bug?
3. The _random_ address always starts with a non-random fixed value. Controller Bug?

Oh, and I also alluded to the fact that nimble uses a hard coded IRK. I'd expect this will causing boning to more than one privacy enabled ESP device. More of a nimble issue, but I'll mention in here for anyone finding this thread. See micro-python for workaround to that issue. https://github.com/micropython/micropython/blob/cacc96d98c2a70dc7e5194331ea70746c39746ec/extmod/nimble/modbluetooth_nimble.c#L281

braiden, 2023-03-12:
> 2. However the RPA rotates every 15 minutes ignoring CONFIG_BT_NIMBLE_RPA_TIMEOUT even though a timeout is set: ble_hs_pvcy_ensure_started() → ble_hs_pvcy_set_addr_timeout() → BLE_HCI_OCF_LE_SET_RPA_TMO. Controller Bug?

Manually sending *HCI_LE_Set_Resolvable_Private_Address_-Timeout* (Vol 4, Pt E, 7.8.45) in the sync callback fixes this. Unclear why. NimBLE calls this too. Maybe something to do with order of the commands being issued (e.g. must enable resolve list first)? So could be either a NimBLE bug or a controller bug?

devcexx, 2023-06-11:
Hi,

This thread was the only result in Google after copy pasting the MAC address I was getting on my two ESP32-S3 devices (with different manufacturers and form factors), while advertising using Controller-based RPA Public MAC address with NimBLE. I'm glad that I'm not the only one having the same issue.

@braiden the workaround you've published for the ESP32 is just for host-based privacy right? Unfortunately, my use-case requires controller-based privacy since I'm using the whitelist feature, which doesn't work with host-based privacy.

So, I can see this issue while enabling GAP advertising with the following line: 

```C
rc = ble_gap_adv_start(BLE_OWN_ADDR_RPA_PUBLIC_DEFAULT, NULL, BLE_HS_FOREVER,
			 &adv_params, rfble_gap_event, NULL);
```

This is making the MAC address to be set on my two devices to `7e:1e:1e:79:4f:f7`. The resolution of the peer's RPA works perfectly though. I'm not very familiar with NimBLE stack nor Bluetooth in general, so I'm not sure if there's anything else that I need to do to setup the controller for using RPA or this is indeed a controller/host stack bug. Any guidance is appreciated.

braiden, 2023-06-11:
I'm using device based privacy.  I may not have updated the bug with my latest findings:

Do not use `ble_gap_adv_start` with `BLE_HS_FOREVER`. Instead I restart advertisement ever 5 minutes. After this change, the device will rotate the address every 15 minutes. Remaining issues are:

1. The initial address is always 7e:1e:1e:79:4f:f7, but a new random address will be assigned in ~15 minutes.
2. The API to change the timeout to something shorter than 15 minutes didn't work.

devcexx, 2023-06-11:
This is reproducible on my side while using Controller-based privacy. Restarting the advertisement every 5 minutes, after 15 minutes the device will start advertising a new RPA address, but the initial address is always 7e:1e:1e:79:4f:f7. For my case changing the setting `BT_NIMBLE_RPA_TIMEOUT` won't have any effect because it is only applicable to host-based privacy.

achugbGL, 2023-09-16:
We are seeing this issue on our ESP32-C3 devices as well with the same initial address: 7e:1e:1e:79:4f:f7.

zhp0406, 2023-09-18:
thanks for report the problem,you can set RPA timeout with menuconfig:
(Top) → Component config → Bluetooth → NimBLE Options ->(5) RPA timeout in seconds

zhp0406, 2023-09-18:
@achugbGL @devcexx 
This is an issue (with the same initial address). could you please specify how this issue affects your usage? We will fix it ASAP.

zhp0406, 2023-09-18:
> @zhp0406, thanks for looking into this. Like @braiden mentioned above, the RPA timeout setting seems to be getting ignored for some reason. Regarding the initial address issue, the biggest problem for us is that when multiple devices using our firmware boot around the same time, they have conflicting addresses.

sorry for that,it will take some time for us to solve the issue(The initial address is always 7e:1e:1e:79:4f:f7). In the meantime, to avoid affecting your usage, you can set a different Identity Resolving Key (IRK) for each device. This way, the addresses will be different during the first initialization.
Note:
You can use the API **int ble_hs_pvcy_set_our_irk(const uint8_t *irk)** to set the IRK for each device.

achugbGL, 2023-09-19:
Thank you! Appreciate your attention to this issue

rahult-github, 2023-10-30:
Hi @achugbGL ,

The attached patch has a change to generate a new IRK for each device on boot without the need for user to call any API. Please help check if it works for you. The patch should go in $IDF_PATH/components/bt/host/nimble/nimble path.

[0001-Added-change-to-generate-unique-IRK-based-on-chip-ma.txt](https://github.com/espressif/esp-idf/files/13201630/0001-Added-change-to-generate-unique-IRK-based-on-chip-ma.txt)

Indastri, 2023-12-07:
Since there is no response from the user, closing the issue, if there will be any questions or issues feel free to reopen or create a new issue. Cheers.

closed 2023-12-07 by Indastri

achugbGL, 2025-03-19:
Hi there. Sorry for missing this and for re-opening it. We had implemented our own IRK generation & storage method and I'm looking to migrate to the stock IDF implementation. One question I had was regarding how to reset the stored IRK and force generation of a new random IRK. Thanks!

esp-zhp, 2025-03-19:
@achugbGL 
We have modified the random number generation method so that prand is truly random each time, ensuring that the address is also random.
see：
BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 6, Part B page 2680

[image]

achugbGL, 2025-03-19:
Hi. Sorry, I should have been more clear. It seems like the ESP on boot/initializing BLE checks to see if there is an IRK in NVS and if not it generates a random IRK and saves it to NVS. I would like to be able to reset the saved IRK with a new random IRK so that previously bonded devices cannot track the device.

esp-zhp, 2025-03-19:
@achugbGL 
This is a new issue. Please create a separate issue to keep this one organized; otherwise, it may become confusing.  

Regarding your question, you can try enabling the configuration option **BT_BLE_SMP_ID_RESET_ENABLE** to see if it meets your requirements.  

See:  
**(Top) → Component config → Bluetooth → Bluedroid Options → Bluetooth Low Energy → Include BLE security module (SMP)**  
[ ] Reset device identity when all bonding records are deleted.

achugbGL, 2025-03-19:
Sorry, I thought it might be a quick question about the functionality that was added to resolve this issue. I can make a new one. 

I did see that config option but it seemed like that one was specific to Bluedroid where as this is for NimBLE

rahult-github, 2025-03-19:
> I did see that config option but it seemed like that one was specific to Bluedroid where as this is for NimBLE 

For NimBLE:

(Top) → Component config → Bluetooth → NimBLE Options
[ ] Reset device identity when all bonding records are deleted