espressif/esp-idf #10916: Question about TLS on TCP SERVER

Inverter-Jo, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to apply TLS in TCP/IP communication.
We are checking the related information through the link below.
There is only content about TCP CLIENT about TLS, is there anything related to TCP SERVER?

Alvin1Zhang, 2023-03-06:
@Inverter-Jo Thanks for reporting, would you please help share the link you mentioned in the description? Thanks.

Inverter-Jo, 2023-03-06:
@Alvin1Zhang

link is here.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html?highlight=tls
https://mbed-tls.readthedocs.io/en/latest/kb/how-to/mbedtls-tutorial/

Thanks for answer

mahavirj, 2023-03-06:
Seems same as https://github.com/espressif/esp-idf/issues/10893, closing as duplicate

closed 2023-03-06 by mahavirj