espressif/esp-idf #10917: ESP32-S3 - SECURE_SIGNED_APPS_NO_SECURE_BOOT apps will run with no signature, or with the wrong signature type

maskenfix, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using ESP32-S3 and IDF v5.1-dev-3797-gaf805df3cb

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/secure-boot-v2.html#signed-app-verification-without-hardware-secure-boot

> For this reason, it’s essential that the initial app flashed to the device is also signed. A check is run on app startup and the app will abort if no signatures are found. This is to try and prevent a situation where no update is possible. The app should have only one valid signature block in the first position. 

Building an image with the below options (ie without any signature) will run OK (no aborts), but any OTA update will fail.  So it seems that the documentation is not correct.

CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_SIGNED_ON_UPDATE_NO_SECURE_BOOT=y
CONFIG_SECURE_BOOT is not set
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES is not set

Additionally, if enabling BUILD_SIGNED_BINARIES but creating the signing key file as a v1 (ECDSA) type, the build system will happily sign this app and the factory image will run.  However, the OTA update mechanism can only use v2 (RSA) type and the digest check returns an error.  So no OTA update is possible if the signing key is the wrong type.

That means there are 2 different mistakes where Signed App Verification can lock you out of doing OTA updates.

If the factory and OTA images are all signed with v2 (RSA) key then it all works as expected.

Suggest improving the app startup check to make sure:

1. That there is a signature present (the documentation says the app will check, but it doesn't)
2. That the signature is the correct type

mahavirj, 2023-03-06:
@maskenfix If you want the signature to be verified during bootup (with hardware secure boot disabled), please consider enabling following option:

https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/components/bootloader/Kconfig.projbuild#L561-L571

mahavirj, 2023-03-06:
> Suggest improving the app startup check to make sure: That there is a signature present (the documentation says the app will check, but it doesn't)

Aha. Please disregard my earlier comment. You would like to ensure sanity of the signature block from app startup code itself.

Startup code hook for this part can be found below:

https://github.com/espressif/esp-idf/blob/af805df3cb6117cc74f0875831fbab6446824453/components/bootloader_support/src/secure_boot.c#L187-L189

Can you please check by putting some debug prints to verify the code flow once? Please share debug/verbose enabled log for the device startup part.

maskenfix, 2023-03-07:
The debug print indicated the signature check is running, and it found 1 valid signature after uploading an unsigned app.

Then I realised that the cause is on the flash erase side of things.  Because I flashed the signed app first, then the unsigned app, the unsigned app is smaller, and the flash is only erased as much as necessary to upload the new image.  So the old signature block hangs around and is found by the signature check.

If I do an idf.py erase-flash first then it works as expected (the app is in fact checking that a signature is present, and it will abort if I upload an unsigned app).  It's an extremely narrow edge case.

closed 2023-03-07 by maskenfix