espressif/esp-idf #10918: About Example bluetooth/bluedroid/ble/gatt_server_service_table

TarikAkyuz, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

According to table creation(esp_gatts_attr_db_t gatt_db[HRS_IDX_NB] ) All values connected to char_value(uint8_t char_value[4]).
Therefore, if I write a value to any character, I must read the same value in the other character.

### What is the actual behavior?

Since the value array will be updated when I write in any of the writable characters, 
I can't read the same value when I read any character. I always read the default value

### Steps to reproduce.

just use default example

zhp0406, 2023-11-17:
/**
 * @brief           This function is called to set the attribute value by the application
 *
 * @param[in]       attr_handle: the attribute handle which to be set
 * @param[in]       length: the value length
 * @param[in]       value: the pointer to the attribute value
 *
 * @return
 *                  - ESP_OK : success
 *                  - other  : failed
 *
 */
esp_err_t esp_ble_gatts_set_attr_value(uint16_t attr_handle, uint16_t length, const uint8_t *value);

/**
 * @brief       Retrieve attribute value
 *
 * @param[in]   attr_handle: Attribute handle.
 * @param[out]  length: pointer to the attribute value length
 * @param[out]  value:  Pointer to attribute value payload, the value cannot be modified by user
 *
 * @return
 *                  - ESP_GATT_OK : success
 *                  - other  : failed
 *
 */
esp_gatt_status_t esp_ble_gatts_get_attr_value(uint16_t attr_handle, uint16_t *length, const uint8_t **value);

zhp0406, 2023-11-17:
If you want to get and set the value of the server database, please use the following API.：
esp_err_t esp_ble_gatts_set_attr_value(uint16_t attr_handle, uint16_t length, const uint8_t *value);
esp_gatt_status_t esp_ble_gatts_get_attr_value(uint16_t attr_handle, uint16_t *length, const uint8_t **value);

closed 2024-03-01 by Alvin1Zhang