espressif/esp-idf #10919: [question] hci over usb on S3

chegewara, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,
sorry for noob-ish question, and probably i should ask on forum, but is it possible to implement HCI over USB or is there plan to add it?

Thanks

roma-jam, 2023-03-07:
@chegewara, 
Hi.
Isn't USB Host interface already available can fulfil your requirement? 
Have you checked it? 

https://github.com/espressif/esp-idf/blob/master/components/usb/include/usb/usb_host.h

Thanks.

chegewara, 2023-03-07:
Hi,
i just realized that my question was not precise.
What i am asking is BLE HCI over USB, just like in this example over UART:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart_esp32c3_and_esp32s3

Thanks

xiewenxiang, 2023-03-08:
Yes, you can refer to https://github.com/espressif/esp-iot-solution/tree/usb/add_usb_solutions/examples/usb/device/usb_dongle

closed 2024-02-23 by Alvin1Zhang