espressif/esp-idf #10924: REGRESSION v4.4.3 -> v4.4.4: W5500 Ethernet initialization breaks all network (WiFi+ETH) connectivity on Arduino-ESP32 2.0.7, worked OK on 2.0.6

avillacis, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

YUBOX One ESP32-S2

### Power Supply used.

USB

### What is the expected behavior?

The initialization of the W5500 Ethernet module should work correctly and not interfere with any operation of existing WiFi interfaces.

Under Arduino-ESP32 2.0.6 (using v4.4.3), the call to the eth_w5500_begin() function (implementation shown below) worked correctly, and did not interfere in any way with the WiFi support. After initialization is complete, both the WiFi and the Ethernet interfaces answer pings and are usable for network servers initialized for the project, mainly the HTTP server. Also, the softAP interface works as usual.

### What is the actual behavior?

Under recently-released Arduino-ESP32 2.0.7, the Ethernet initialization returns success on all API calls, as in 2.0.6. However, the Ethernet interface does not answer the pings directed to the address supposedly handed out through DHCP. Additionally, the WiFi interface appears to authenticate successfully to the network, since it received the STA_CONNECTED event, but never receives an IP. After a long time, the interface disconnects on its own (STA_DISCONNECTED received). The softAP interface also appears to accept clients but never answers to the default IP 192.168.4.1.

### Steps to reproduce.

Switch from Arduino-ESP32 2.0.6 (ESP-IDF v4.4.3) to Arduino-ESP32 2.0.7 (ESP-IDF v4.4.4). No other changes to projects.

The function eth_w5500_begin() is shown below:
```C++
#include "esp_system.h"

#include "esp_event.h"
#include "esp_eth.h"
#include "esp_eth_phy.h"
#include "esp_eth_mac.h"
#include "esp_eth_com.h"
#if CONFIG_IDF_TARGET_ESP32
#include "soc/emac_ext_struct.h"
#include "soc/rtc.h"
//#include "soc/io_mux_reg.h"
//#include "hal/gpio_hal.h"
#endif

#include "lwip/err.h"
#include "lwip/dns.h"

#include "driver/spi_master.h"

#include "esp32-hal.h"

#include <Arduino.h>

extern void tcpipInit();
[... 198 lines cut ...]
E (283778) w5500.mac: w5500_write(74): spi transmit failed
E (283778) w5500.mac: w5500_write_buffer(173): write TX buffer failed
E (283778) w5500.mac: emac_w5500_transmit(528): write frame failed
```

### More Information.

This report is based on bug report https://github.com/espressif/arduino-esp32/issues/7909 . I was asked to make a bug report here, because all the APIs are supposedly 100% ESP-IDF.

W5500 board connected to the following GPIO pins on ESP32-S2:

#define ETH_MISO GPIO_NUM_13
#define ETH_MOSI GPIO_NUM_11
#define ETH_SCLK GPIO_NUM_12
#define ETH_SS GPIO_NUM_10
#define ETH_RST GPIO_NUM_6
#define ETH_IRQ GPIO_NUM_7

I am using an ESP32-S2 board with a pluggable W5500 Ethernet module. This module uses the built-in WiFi support for both softAP and STA, and (due to a project requirement) also initializes the W5500 using the ESP-IDF built-in support, and attempts to attach it to the TCP/IP stack with custom code (shown below), since the Ethernet.h library (still) does not handle support for the W5500.

The project code calls eth_w5500_begin() from the setup() function, then leaves it initializing and scans for WiFi networks, checks in NVRAM for known networks with credentials, and uses them to join the WiFi network. Usually the Ethernet DHCP runs to completion before the WiFi scan completes.

If I compile the project using 2.0.7 but comment out the call to eth_w5500_begin(), the rest of the networking support behaves correctly as it did under 2.0.6.

The eth_w5500_begin() function is supposedly based on the Ethernet library shipped with Arduino-ESP32, but with added SPI initialization required for the W5500 module. This initialization, and the TCP/IP registration, has been tested to work under 2.0.6.

kostaond, 2023-03-10:
@avillacis thanks for reporting potential issue. There is practically no changes to `esp_eth` between v4.4.3 and v4.4.4. Could you please share some minimum code example to reproduce the issue? You nicely described it but having the code would help me a lot.

Did you observed the
 ```
E (283778) w5500.mac: w5500_write(74): spi transmit failed
```
before?
How is your W5500 module connected to ESP board? If just via jump wires, high SPI frequency may cause intermittent troubles. I am not saying this is the root cause of the reported issue, it's just a general case.

AxelLin, 2023-04-22:
@avillacis Any update regarding comments in https://github.com/espressif/esp-idf/issues/10924#issuecomment-1463882679

avillacis, 2023-04-22:
> @avillacis Any update regarding comments in [#10924 (comment)](https://github.com/espressif/esp-idf/issues/10924#issuecomment-1463882679)

The Arduino-ESP32 project recently released 2.0.8, but is unusable for me due to https://github.com/espressif/arduino-esp32/issues/8108 . 

The W5500 is soldered to a dedicated plug-in board designed for the YUBOX-One board, a board based on the ESP32-S2. However I do not think this is the issue, since the problem is persistent, not intermittent, when compiling under Arduino-ESP32 2.0.7, and likewise going back to Arduino-ESP32 2.0.6 fixes the bug reliably.

kostaond, 2023-04-23:
@avillacis Could you please share some minimum code example to reproduce the issue?

kostaond, 2023-07-25:
I'm closing the issue due to lack of feedback. I'm not able to reproduce. Feel free to reopen but please provide minimal code example to reproduce the issue.

closed 2023-07-25 by kostaond

avillacis, 2023-09-27:
Sorry for the lack of feedback.

I have been doing some tests with the following example code compiled with Arduino-ESP32 2.0.13:
```c++
#include <WiFi.h>

#include "esp_system.h"

#include "esp_event.h"
#include "esp_eth.h"
#include "esp_eth_phy.h"
#include "esp_eth_mac.h"
#include "esp_eth_com.h"
#if CONFIG_IDF_TARGET_ESP32
#include "soc/emac_ext_struct.h"
#include "soc/rtc.h"
#endif

#include "lwip/err.h"
#include "lwip/dns.h"

#include "driver/spi_master.h"

#include "esp32-hal.h"

#include <Arduino.h>

extern bool tcpipInit();
extern void add_esp_interface_netif(esp_interface_t interface, esp_netif_t* esp_netif); /* from WiFiGeneric */

#if CONFIG_IDF_TARGET_ESP32S2

#define ETH_MISO    GPIO_NUM_13
#define ETH_MOSI    GPIO_NUM_11
#define ETH_SCLK    GPIO_NUM_12
#define ETH_SS      GPIO_NUM_10
#define ETH_RST     GPIO_NUM_6
#define ETH_IRQ     GPIO_NUM_7

#define ETH_SPI_BUS FSPI_HOST

#elif CONFIG_IDF_TARGET_ESP32S3

#define ETH_MISO    GPIO_NUM_48
#define ETH_MOSI    GPIO_NUM_45
#define ETH_SCLK    GPIO_NUM_46
#define ETH_SS      GPIO_NUM_42
#define ETH_RST     GPIO_NUM_6
#define ETH_IRQ     GPIO_NUM_5

#define ETH_SPI_BUS SPI2_HOST

#else

#error Pines de control no definidos para board objetivo!

#endif

#define ETH_PHY_ADDR 0
#define ETH_SPI_CLOCK_MHZ 27    // 5..80, 12 para ESP32,ESP32C3, 36 para ESP32S2,ESP32S3

#define ETH_IF_KEY "ETH_SPI_0"

static spi_device_handle_t spi_handle = NULL;

static esp_eth_mac_t *mac_spi;
static esp_eth_phy_t *phy_spi;
static esp_eth_handle_t eth_handle_spi = NULL;

typedef struct {
[... 264 lines cut ...]
17:22:02.551 -> E (10350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:02.551 -> E (10352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:02.551 -> ..E (12350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:04.544 -> E (12350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:04.544 -> E (12352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:04.544 -> ..E (14350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:06.537 -> E (14350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:06.537 -> E (14352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:06.537 -> ..E (16350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:08.532 -> E (16350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:08.532 -> E (16352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:08.564 -> ..E (18350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:10.528 -> E (18350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:10.528 -> E (18352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:10.560 -> ..E (20350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:12.521 -> E (20350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:12.521 -> E (20352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:12.552 -> ..E (22350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:14.547 -> E (22350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:14.547 -> E (22352) w5500.phy: w5500_get_link(108): update link duplex speed failed
17:22:14.580 -> ..E (24350) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
17:22:16.543 -> E (24350) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
17:22:16.543 -> E (24352) w5500.phy: w5500_get_link(108): update link duplex speed failed
```
Sorry for not (yet) testing with pure ESP-IDF but the project requires Arduino-ESP32.

avillacis, 2023-09-27:
I have noticed closed bug https://github.com/espressif/esp-idf/issues/11845 which is also about the W5500 code and has a similar behavior to my own. However, disabling PSRAM seems to have no effect on my results.

kostaond, 2023-11-10:
@avillacis thank you very much for the code! I tried to run it. I made only marginal changes as described below:
```c
#if CONFIG_IDF_TARGET_ESP32S2

#define ETH_MISO    GPIO_NUM_1
#define ETH_MOSI    GPIO_NUM_9
#define ETH_SCLK    GPIO_NUM_8
#define ETH_SS      GPIO_NUM_39
#define ETH_RST     -1
#define ETH_IRQ     GPIO_NUM_3

#define ETH_SPI_BUS FSPI_HOST
```
```c
#define ETH_SPI_CLOCK_MHZ 20 
```
I've used PlatformIO with the following configuration:
```
[env:esp32dev]
platform = espressif32
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.13
board = esp32-s2-saola-1
framework = arduino
monitor_speed = 115200
```
Unfortunately, I was not able to reproduce the issue. The WiFi and Ethernet works at the same time. I'm able to ping both interfaces.