espressif/esp-idf #10925: Virtual Efuses Don't Work with PSRAM Enabled

sidwarkd, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Pico Devkit M-22 v1.0

### Power Supply used.

USB

### What is the expected behavior?

Expect application to boot when virtual efuses and psram are both enabled.

### What is the actual behavior?

Application does not boot and resets due to watchdog.

### Steps to reproduce.

To easily reproduce this issue I have created a repo. 

  1. Clone the repo at https://github.com/sidwarkd/virtual_efuses_bug
  2. Ensure v4.4.4 of IDF is installed
  3. Build and run project with `idf.py flash monitor`
  4. Notice efuses will be read during boot and application will run normally.
  5. Modify sdkconfig.defaults file by uncommenting lines 62-70 (SPIRAM config)
  6. Run `idf.py fullclean`
  7. Build and run project with `idf.py flash monitor`
  8. Notice application no longer boots but hangs and restarts due to watchdog timer.

### Debug Logs.

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff00b8,len:7644
ho 0 tail 12 room 4
load:0x40078000,len:16532
load:0x40080400,len:3832
0x40080400: _init at ??:?

entry 0x4008069c
W (66) boot.esp32: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (44) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (44) boot: compile time 15:43:08
I (44) boot: chip revision: v3.0
I (47) boot.esp32: SPI Speed      : 80MHz
I (52) boot.esp32: SPI Mode       : DIO
I (56) boot.esp32: SPI Flash Size : 8MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 0000c000 00002000
I (85) boot:  1 nvs_custom       WiFi data        01 02 0000e000 00003000
I (92) boot:  2 otadata          OTA data         01 00 00011000 00002000
I (100) boot:  3 nvs_key          NVS keys         01 04 00013000 00001000
I (107) boot:  4 custom_key       WiFi data        01 02 00014000 0000a000
W (115) efuse: Loading virtual efuse blocks from flash
EFUSE_BLKx:
0) 0x00000000 0x6d28fedc 0x005de89f 0x0000ad00 0x00001038 0x0015de26 0x00000004 
1) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 
2) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 
3) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 

I (153) boot:  5 efuse_em         efuse            01 05 0001e000 00002000
I (161) boot:  6 ota_0            OTA app          00 10 00020000 00390000
I (168) boot:  7 ota_1            OTA app          00 11 003b0000 00390000
I (176) boot:  8 storage          Unknown data     01 81 00740000 00080000
I (183) boot:  9 coredump         Unknown data     01 03 007c0000 00010000
I (191) boot: End of partition table
I (195) boot: No factory image, trying OTA 0
I (200) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=0a280h ( 41600) map
I (221) esp_image: segment 1: paddr=0002a2a8 vaddr=3ffb0000 size=021fch (  8700) load
I (225) esp_image: segment 2: paddr=0002c4ac vaddr=40080000 size=03b6ch ( 15212) load
I (233) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=15a94h ( 88724) map
I (263) esp_image: segment 4: paddr=00045abc vaddr=40083b6c size=06e80h ( 28288) load
I (279) boot: Loaded app from partition at offset 0x20000
I (279) boot: Set actual ota_seq=1 in otadata[0]
I (279) boot: Disabling RNG early entropy source...
I (296) psram: This chip is ESP32-D0WD
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff00b8,len:7644
ho 0 tail 12 room 4
load:0x40078000,len:16532
load:0x40080400,len:3832
0x40080400: _init at ??:?

entry 0x4008069c
W (67) boot.esp32: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (44) boot: ESP-IDF v4.4.4 2nd stage bootloader
I (45) boot: compile time 15:43:08
```

chipweinberger, 2023-03-06:
In my own application, I do not have a problem with virtual effuses and PS ram

sidwarkd, 2023-03-06:
@chipweinberger Would you be able to try to reproduce with the steps I provided above? This is a dead simple test app using an off-the-shelf dev kit. Perhaps some other combination of settings in your sdkconfig causes it to work :man_shrugging: ? In the sample I provided it is very easy to repro the issue by just turning on PSRAM. Also, are you on IDF 4.4.4 or a different version?

chipweinberger, 2023-03-06:
I'm away from my device for awhile.

Did you erase your flash before enabling virtual efuses?

sidwarkd, 2023-03-06:
@chipweinberger Not immediately but I had tried erasing the entire flash after it was not working. Same result.

KonstantinKondrashov, 2023-03-07:
Hi @sidwarkd !
I run your example on v4.4.4 with and without PSRAM. I do not see any issue at all. I see the log `This is a test of virtual efuses and psram`.
I used ESP32 v1.0 (ECO1), and FLASHSIZE_4MB (I reduced the partition table).

sidwarkd, 2023-03-07:
@KonstantinKondrashov :thinking: I've reproduced this on two separate ESP32 Pico v3 chips. One was the Pico Devkit M-2 and the other is our custom hardware that uses the Pico V3-02. Do you have access to  Pico V3 that you could also try it on.

KonstantinKondrashov, 2023-03-07:
@sidwarkd This happens because when the virt efuse mode is on then the efuse buffer should be filled by real efuse before to start using it. There is [a efuse constructor]( https://github.com/espressif/esp-idf/blob/master/components/efuse/src/esp_efuse_utility.c#L30) that is called at the startup code but apparently, later than the psram init function. So there is a solution just to use efuse directly from the hal component. You can try it, should help.
FYI, it is already fixed on the master.

```diff
diff --git a/components/esp_hw_support/port/esp32/spiram_psram.c b/components/esp_hw_support/port/esp32/spiram_psram.c
index 4bbd239..a4f36cd 100644
--- a/components/esp_hw_support/port/esp32/spiram_psram.c
+++ b/components/esp_hw_support/port/esp32/spiram_psram.c
@@ -26,6 +26,7 @@
 #include "soc/soc_caps.h"
 #include "driver/gpio.h"
 #include "hal/efuse_hal.h"
+#include "hal/efuse_ll.h"
 #include "hal/gpio_hal.h"
 #include "driver/spi_common_internal.h"
 #include "driver/periph_ctrl.h"
@@ -822,7 +823,7 @@ bool psram_is_32mbit_ver0(void)
 esp_err_t IRAM_ATTR psram_enable(psram_cache_mode_t mode, psram_vaddr_mode_t vaddrmode)   //psram init
 {
     psram_io_t psram_io={0};
-    uint32_t pkg_ver = esp_efuse_get_pkg_ver();
+    uint32_t pkg_ver = efuse_ll_get_chip_ver_pkg();
     if (pkg_ver == EFUSE_RD_CHIP_VER_PKG_ESP32D2WDQ5) {
         ESP_EARLY_LOGI(TAG, "This chip is ESP32-D2WD");
         rtc_vddsdio_config_t cfg = rtc_vddsdio_get_config();
```

sidwarkd, 2023-03-07:
@KonstantinKondrashov I confirmed this solution works! Thanks for your help.

closed 2023-03-07 by sidwarkd