espressif/esp-idf #10927: Flash Encryption is cannot be enabled - SecureBoot Enabled before

DerPicknicker, 2023-03-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-3462-g045163a2ec-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32

### Power Supply used.

USB

### What is the expected behavior?

I expected that after enabling SecureBoot and then enable FlashEncryption in the IDF-Menuconfig. That the ESP with the newly flashed Bootloader will start the encrpytion process as described in the Manuals. 

### What is the actual behavior?

The Flash-Encryption is not working. If I try to flash the Image with the --encrypted Option I will get this Log: 

```
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash encryption key is not programmed

A fatal error occurred: Can't perform encrypted flash write, consult Flash Encryption documentation for more information
CMake Error at run_serial_tool.cmake:63 (message):
  
 /Users/USERNAME/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/USERNAME/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32
  failed.

FAILED: CMakeFiles/encrypted-flash /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example/build/CMakeFiles/encrypted-flash 
cd /Users/USERNAME/esp/esp-idf/components/esptool_py && /Users/USERNAME/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin/cmake -D IDF_PATH=/Users/USERNAME/esp/esp-idf -D "SERIAL_TOOL=/Users/USERNAME/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/USERNAME/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=no_reset;write_flash;@encrypted_flash_args" -D WORKING_DIRECTORY=/Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example/build -P /Users/USERNAME/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed
```

Now then I try to flash it without the --encryped flag it seems to work but in the end its not the case as you can see in the Monitor Log: 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[... 80 lines cut ...]
CHIP_PACKAGE_4BIT (BLOCK0):                        Chip package identifier #4bit                      = 0 R/W (0b0)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)
WAFER_VERSION_MAJOR (BLOCK0):                      calc WAFER VERSION MAJOR from CHIP_VER_REV1 and CH = 1 R/W (0b001)
                                                   IP_VER_REV2 and apb_ctl_date (read only)          
PKG_VERSION (BLOCK0):                              calc Chip package = CHIP_PACKAGE_4BIT << 3 + CHIP_ = 0 R/W (0x0)
                                                   PACKAGE (read only)                               

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 1 R/W (0b0000001)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = True R/W (0b1)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                                   Flash encryption key                              
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLOCK2 (BLOCK2):                                   Secure boot key                                   
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
BLOCK3 (BLOCK3):                                   Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
```

mahavirj, 2023-03-08:
@DerPicknicker 

> FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 1 R/W (0b0000001)

This suggest that the device has flash encryption enabled (odd count of set bits)

But there is no flash encryption key programmed in the BLOCK1. Probably you enabled or set the flash crypt cnt field externally and then flashed the bootloader with flash encryption enabled configuration.

Couple of ways to proceed now:

1. Please increase the `FLASH_CRYPT_CNT` field so that flash encryption gets disabled. Please refer to the command from documentation section [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#disabling-flash-encryption). After this, the plaintext bootloader should enable the flash encryption workflow and correctly program the key in the EFuse.
2. Burn the flash encryption key externally. Please refer to the documentation section [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#using-host-generated-key). Once you burn the key, `encrypted-*` commands should work to encrypt artifact before flashing.

Hope this helps!

mahavirj, 2023-03-08:
> If I get you correctly I should increase the field to disable the flash-encryption. Then the Encryption process will start automatically, right?

Yes

> In option 2) you mentioned flashing the key externally is this a real security issue then I go to release later? I want the highest possible security to protect my Code and prevent flashing unofficial firmware.

If you generate per device unique key and carefully destroy it once its programmed in the device then it should not result in any security issue. Of-course, enabling flash encryption release mode through bootloader is more convenient option but here we are trying to recover this device first.

DerPicknicker, 2023-03-08:
I will do it this evening and comeback to you.

DerPicknicker, 2023-03-08:
Hi, I did it. But the result is the same. Flash Encryption is not starting. The idf.py Monitor shows the same as before (first Post) What should I do @mahavirj ..? 

thank you for your time :) 

```
FUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1107 R/W (0b00001)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 57 R/W (0x39)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 256 R/W (0x0100)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 2 R/W (0x2)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme                
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address                               
   = 58:bf:25:36:1e:dc (CRC 0x95 OK) R/W 
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 149 R/W (0x95)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = False R/W (0b0)
WAFER_VERSION_MINOR (BLOCK0):                      WAFER VERSION MINOR                                = 0 R/W (0b00)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 0 R/W (0b000)
CHIP_PACKAGE_4BIT (BLOCK0):                        Chip package identifier #4bit                      = 0 R/W (0b0)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)
WAFER_VERSION_MAJOR (BLOCK0):                      calc WAFER VERSION MAJOR from CHIP_VER_REV1 and CH = 1 R/W (0b001)
                                                   IP_VER_REV2 and apb_ctl_date (read only)          
PKG_VERSION (BLOCK0):                              calc Chip package = CHIP_PACKAGE_4BIT << 3 + CHIP_ = 0 R/W (0x0)
                                                   PACKAGE (read only)                               

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 3 R/W (0b0000011)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = True R/W (0b1)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                                   Flash encryption key                              
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLOCK2 (BLOCK2):                                   Secure boot key                                   
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
BLOCK3 (BLOCK3):                                   Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Flash voltage (VDD_SDIO) determined by GPIO12 on reset 

```

mahavirj, 2023-03-08:
> The idf.py Monitor shows the same as before (first Post)

I hope you are flashing signed but plaintext bootloader now. You can also try with default bootloader generated without any security features enabled, it shall fail for the signature verification part but at-least it should not give `flash read err, 1000`.

DerPicknicker, 2023-03-08:
> I hope you are flashing signed but plaintext bootloader now. You can also try with default bootloader generated without any security features enabled, it shall fail for the signature verification [...]

I will double check that. Plaintext == disabled inside IDF - Menuconfig?

mahavirj, 2023-03-08:
Interim, I noticed that you are using "secure boot v1" scheme here. If you are using ESP32 ECO3 and above revision then you should consider enabling "secure boot v2" scheme. 

You can check the revision using: `esptool.py chip_id` command.

For "secure boot v1" case, if you had generated the key through the bootloader then reflashing the bootloader is not possible. It can potentially brick the device. Please refer to this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#re-flashable-software-bootloader)

DerPicknicker, 2023-03-08:
Flash Encryption Never worked. And yes I use V1 - After enabling, flashing bootloader, it displays me the mentioned logs. After disabling, re-flash new bootloader the esp32 works as expected. I will double check your ideas ;)

DerPicknicker, 2023-03-08:
> Interim, I noticed that you are using "secure boot v1" scheme here. If you are using ESP32 ECO3 and above revision then you should consider enabling "secure boot v2" scheme.
>
> You can check the [...]

Sorry for my late Reply. But Indeed I made a mistake after setting the Fuse, as you mentioned. 
The Output differs as from the first Post: 
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

Now I checked and disabled the Flash-Encryption in the Security-Menu of Menuconfig. Build a new Bootloader and flash it to the Device using this Command: 
`esptool.py --before default_reset --after hard_reset --chip esp32 write_flash --erase-all --force --flash_mode dio --flash_freq 40m --flash_size detect  0x0 build/bootloader/bootloader-reflash-digest.bin`

After the Bootloader is flashed. The Error is not the same 

```
secure boot check fail
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff00b8,len:11144
load:0x40078000,len:25104
load:0x40080400,len:3864
0x40080400: _init at ??:?
```

Now I flashed the Chip again. But now it will not come back. Is it now Broken? The Command to flash the Bootloader was also used before. But now it seems that I cannot recover my Device. 

Here are my Sec-Settings: 
[image]

Hope this Helps.

mahavirj, 2023-03-09:
Yes, if you had the secure boot v1 key generated by the bootloader (and not externally programmed) then you can not re-flash the bootloader. In-fact, we have some measures in the `esptool` to prevent re-flashing bootloader image for this case, please see [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#bootloader-protection).

> But now it seems that I cannot recover my Device.

Since, you reflashed the bootloader (using `--force`) and the secure boot v1 key is not accessible outside (read-protected), there is no way to regenerate the digest again.

For future devices, please consider using "secure boot v2" scheme, provided you are using ESP32 with ECO3 or higher revision.

DerPicknicker, 2023-03-09:
Hello @mahavirj , 

Thank you for the explanations. Before you close it. I already have an ESP-S3 (due the non-fixable secrurity issues on the esp32-board). 

I don’t want to break that board again - how are the steps to enable both. Should I do it at the same time?

1. Enable SecureBoot
2. Enable FlashEncryption
3. Generate key for bootloader (how to select public key and not private key?)
4. Generate Bootloader 
5. Flash bootloader 
But then how should I proceed? Burn Fueses?

Could you make a short list (please don’t refer to the Documentation - I read them several times…) how the right order is to get SecureBoot as FlashEncryption running.

best regards

DerPicknicker, 2023-03-12:
@mahavirj .. Any Updates on your Side? 

I bought a new ESP32-S3. Is my Process described here correct? Or should I proceed in a different way? 

> 
> 1. Enable SecureBoot
> 2. Enable FlashEncryption
> 3. Generate key for bootloader (how to select public key and not private key?)
> 4. Generate Bootloader
> 5. Flash bootloader
>    But then how should I proceed? Burn Fueses?
> 
>

Best regards :)

mahavirj, 2023-03-13:
@DerPicknicker 

1. Please enable required security features through `idf.py menuconfig -> Security features`
2. You may keep flash encryption in DEVELOPMENT mode for testing purpose
3. You may also keep some potentially insecure options enabled, like `SECURE_INSECURE_ALLOW_DL_MODE` enabled. This is to keep UART DL mode enabled and hence all esptool commands to be supported. Same thing for other options like JTAG etc.
4. Modify partition table offset, with security features bootloader size increases and hence partition table offset must be increased. Typically `0xd000` offset should work well here.
5. Build the project
6. Build system would notify regarding missing signing key and command to generate one. Execute the command to generate the signing key. This is private key which will be used to sign the artifacts but only public key (its digest) would be programmed in the EFuse by bootloader on very 1st boot.
7. If secure boot is enabled, then bootloader must be flashed separately. Please flash all other artifacts like partition table, application and then bootloader at the very last. Flashing commands would be displayed during the build process.
8. On the 1st boot, bootloader would verify, configure EFuses and encrypt all partitions. This process may take some time. Once this is over, device should restart and have all security features enabled in the EFuse.

Additionally, we also have a qemu (emulator) fork for ESP32 chip, you may use that to verify the security features workflow. Instructions are available [here](https://github.com/espressif/qemu/wiki). Process would then be similar for other chips like ESP32-S3.

Hope this helps!

DerPicknicker, 2023-03-13:
Hello @mahavirj ... 

Thank you for the explanation. I made some screenshots of my current settings. I still face some problems. But I am not sure how to fix them. 
Here are my Settings: 
[image]
[image]

After I build the Project - I created a new Key. Then the flashing works fine. With the idf.py flash I flashed the entire project beside the Bootloader. Before I did this it seems that I am on a good way. Now I built a new bootloader.. Flashed it with the given command. And I got an issue regarding NVS - the app is not starting at this stage. I tried to re-flash the app with the idf.py command. But after flashing the boot loader I have some strange behaviour. 

Here you see the Flash-Output, it fails at different Percentages. I tried different Baudrates (down to 1000...) If I use 9600 It goes up to around 70%.. 
<details>
  <summary>FLASH-LOG</summary>

```
[915/917] Generating signed binary image
espsecure.py v4.5.1
1 signing key(s) found.
Signed 983040 bytes of data from /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build/simple_ota-unsigned.bin. Signature sector now has 1 signature blocks.
Generated signed binary image /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build/simple_ota.bin from /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build/simple_ota-unsigned.bin
[916/917] cd /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build/esp-idf/esptool_py && /Users/USERNAME/.espressif/python_env/idf5.1_py3.10_env/bin/python /Users/USERNAME/esp/esp-idf/components/partition_table/check_sizes.py --offset 0xd000 partition --type app /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build/partition_table/partition-table.bin /Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build/simple_ota.bin
simple_ota.bin binary size 0xf1000 bytes. Smallest app partition is 0x100000 bytes. 0xf000 bytes (6%) free.
[916/917] cd /Users/USERNAME/esp/esp-idf/components/esptool_py && /Users/USERNAME/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin/cmake -D IDF_PATH=/Users/USERNAME/esp/esp-idf -D "SERIAL_TOOL=/Users/USERNAME/.espressif/python_env/idf5.1_py3.10_env/bin/python;;/Users/USERNAME/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=no_reset;--no-stub;write_flash;@flash_args" -D WORKING_DIRECTORY=/Users/USERNAME/esp/esp-idf/examples/system/ota/simple_ota_example.nosync/build -P /Users/USERNAME/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool esp32s3 -p /dev/cu.usbmodem14301 -b 460800 --before=default_reset --after=no_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size keep 0x20000 simple_ota.bin 0xd000 partition_table/partition-table.bin 0x12000 ota_data_initial.bin
esptool.py v4.5.1
Serial port /dev/cu.usbmodem14301
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:98:15:f0
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00020000 to 0x00110fff...
Flash will be erased from 0x0000d000 to 0x0000dfff...
Flash will be erased from 0x00012000 to 0x00013fff...
Erasing flash...
Took 1.29s to erase flash block
Writing at 0x00020000... (0 %)
Writing at 0x00020400... (0 %)
Writing at 0x00020800... (0 %)
Writing at 0x00020c00... (0 %)
Writing at 0x00021000... (0 %)
Writing at 0x00021400... (0 %)
Writing at 0x00021800... (0 %)
Writing at 0x00021c00... (0 %)
Writing at 0x00022000... (0 %)
Writing at 0x00022400... (1 %)
Writing at 0x00022800... (1 %)
Writing at 0x00022c00... (1 %)
Writing at 0x00023000... (1 %)
Writing at 0x00023400... (1 %)
Writing at 0x00023800... (1 %)
Writing at 0x00023c00... (1 %)
Writing at 0x00024000... (1 %)
Writing at 0x00024400... (1 %)
Writing at 0x00024800... (1 %)
Writing at 0x00024c00... (2 %)
Writing at 0x00025000... (2 %)
Writing at 0x00025400... (2 %)
Writing at 0x00025800... (2 %)
Writing at 0x00025c00... (2 %)
Writing at 0x00026000... (2 %)
Writing at 0x00026400... (2 %)
Writing at 0x00026800... (2 %)
Writing at 0x00026c00... (2 %)
Writing at 0x00027000... (3 %)
Writing at 0x00027400... (3 %)
Writing at 0x00027800... (3 %)
[... 734 lines cut ...]
```

After I checked the EFUSEs wanted to clarify that the Flash-Size is correct: 
```
esptool.py --p /dev/cu.usbmodem14301 chip_id
esptool.py v4.5
Serial port /dev/cu.usbmodem14301
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:98:15:f0
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x45): Possible serial noise or corruption.
```

This are all steps which I have done to troubleshoot the Behaviour. Now I don't have any Idea what's caused the Problem. Did I destroyed another Board? 
I also tried the following: 
- Restarted my Macbook 
- Installed latest Version of ESPTOOL and Pyserial
- Use a different USB-C Port on my Machine 
Nothing works in the end - please help me to find out why I am stuck again ;)

DerPicknicker, 2023-03-14:
@mahavirj .. Should I try to flash it with an windows machine  instead?

mahavirj, 2023-03-14:
@DerPicknicker 

As per the logs you shared above, I think all security features have been correctly enabled. Just a note, please do take a backup of the generated secure signing key.

> A fatal error occurred: Invalid head of packet (0x45): Possible serial noise or corruption.

Troubleshooting [guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#invalid-head-of-packet-0xxx-possible-serial-noise-or-corruption) has some tips, primarily faulty USB cable. I would recommend that you try replacing it with new one..

DerPicknicker, 2023-03-14:
@mahavirj .. 

I will try that - just for my information:
If I lost the key I cannot flash the code to the device due the fact that the bootloader requires my Signature, right? 

Can I change the Signature after I flashed it to the device? Now I develop my code and test a lot. Now I change the key to a new one. Can I update the signature? 

Best regards.

mahavirj, 2023-03-14:
>  If I lost the key I cannot flash the code to the device due the fact that the bootloader requires my Signature, right?

Yes. Both bootloader and  the application are signed.

> Can I change the Signature after I flashed it to the device? Now I develop my code and test a lot. Now I change the key to a new one. Can I update the signature?

No. The hash of the public key goes inside the device EFuse and hence for that device bootloader/application MUST be signed by the corresponding private key. Please see this [technical](https://blog.espressif.com/understanding-esp32s-security-features-14483e465724) write-up.

DerPicknicker, 2023-03-14:
@mahavirj .. 

I tried now a different USB-Cable but the behaviour is still the same. Until 54% and then it breaks up. I don't know why.. 

Here is an Image of my Board, it supports also flashing via UART, should I try that? Currently I use the USB-Connection built-in the esp32-s3..
I use the USB-C port on the left... 
[image]

mahavirj, 2023-03-15:
@DerPicknicker 

Can you please try with esptool v4.5.1 once? Please see https://github.com/espressif/esptool/releases/tag/v4.5.1

You can install it using `pip install esptool==4.5.1`

DerPicknicker, 2023-03-15:
I double checked it @mahavirj .. I already use the Version 4.5.1 ..

dobairoland, 2023-03-17:
The chip cannot be put into download mode on this port:
```
Serial port /dev/cu.wchusbserial51850136391
Connecting......................................
```

It could help to hold the BOOT button on the board until flashing is started. It is also possible that the port is wrong and the device is not connected to that port.

As regarding the previous flashing attempts through the other port: I don't have another idea. The flashing stops randomly if the serial connection is not stable. Lower baud rate, different USB cable, port or driver could help. Sometimes adding a USB hub between the computer and the board also helps.

What is the version of your macOS and device?

DerPicknicker, 2023-03-19:
Hi @dobairoland .. 

First of all thank you for your help. The board has two ports:
- One Port directly connected to the esp32-s3 Chips
I tried: different cable usb-a to usb-c and usb-c to usb-c. The USB-A cable was connected to the MacBook via an USB-C hub. 
I also lowered the baudrate down to 1000… 

-  One port which connects the esp32-s3 via an Serial to USB chip 
For that I tried pressing the boot-button => Still no luck.

I use an 2017 MacBook Pro with macOS Ventura. I already ordered a new board. Should I try to flash this board without secure boot / flash encryption? And after several times, flash the firmware with all security features enabled to see if the issue is reproducible?

best regards

dobairoland, 2023-03-20:
> Should I try to flash this board without secure boot / flash encryption?

Yes. Don't enable secure boot or flash encryption until you are able to reliable flash an unencrypted application.

DerPicknicker, 2023-03-22:
Hi @dobairoland ... , 

I tried now a new ESP32-Board. With this flashing works without any problems... I tried it several times. It seems that it's not a problem with my Machine or the USB-Port. 

Now I tried the other Board with security Features enabled - flashing doesn't work and still doesn't work properly..  It still doesn't finish successfully... 

Writing at 0x0009cc00... (51 %)
Writing at 0x0009d000... (51 %)

A serial exception error occurred: Could not configure port: (6, 'Device not configured')
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:63 (message):

And yes: I tried a different Cable, Port, lowering the baud rate. I don't have any other Ideas to fix that problem.

DerPicknicker, 2023-03-23:
One day later.. I tried the same Board but flashing via the Serial Adapter on the Board works. But the Flashing via the integrated USB-Function of the S3 Board is totally broken until now. 

But now I found out how I can re-flash the Board successfully.. Pressing the BOOT and RST together the output via Serial-interface stops and the IDF.py flash works fine. 
It seems that the Board requires some pressed HW to set the board again into download mode. But this is only needed if your Serial Connection is heavly used. 

But the ESP32-S3 still doesnt boot successfully. could you Help me with that again @mahavirj and @dobairoland ..? 

Here is the current output: 

98 size=00dd0h (  3536) load[0m
[0m
, continuing..[0m
[0;32mI (405) boot: Checking flash encryption...[0m
[0;32mI (417) boot: Disabling RNG early entropy source...[0m
[0;32mI (434) cpu_start: Starting app cpu, entry point is 0x403[0;32mI (457) cpu_start: App version:      v5.1-dev-3462-g04516[0;32mI (464) cpu_start: Compile time:     Mar 23 2023 21:19:19[0;32mI (476) cpu_start: ESP-IDF:          v5.1-dev-3462-g045163a2ec-dirty[0m
[0;32mI (483) cpu_start: Min chip rev:     v0.0[0m
c allocation:[0m
[0;32mI (524) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[0m
[0;32mI (531) spi_flash: detected chip: gd[0m
[0;33mW (538) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m
[0;33mW (552) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)[0m
[0;32mI (560) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (566) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (573) app_start: Starting scheduler on CPU0[0m
[0;32mI (578) app_start: Starting scheduler on CPU1[0m
[0;32mI (578) main_task: Started on CPU0[0m
[0;32mI (588) main_task: Calling app_main()[0m
[0;32mI (588) simple_ota_example: OTA example app_main start[0m
tition with subtype nvs_keys found in the partition table.[0m
file: "/IDF/examples/system/ota/simple_ota_example.nosync/main/simple_ota_example.c" line 180
func: app_main
expression: err
0 0x4037cdeb:0x3fca2e40 0x420093df:0x3fca2e70 0x42098817:0x3fca2ea0 0x4037f7c5:0x3fca2ed0

ELF file SHA256: 83f8adcf9eb2f38b

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
Saved PC:0x40375788
Valid secure boot key blocks: 0
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x528c
[0;32mI (69) boot: ESP-IDF v5.1-dev-3462-g045163a2ec-dirty 2nd [0;32mI (71) boot: chip revision: v0.1[0m
[0;32mI (75) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (79) boot.esp32s3: SPI Mode       : DIO[0m
[0;32mI (84) boot.esp32s3: SPI Flash Size : 4MB[0m
[0;32mI (89) boot: Enabling RNG early entropy source...[0m
[0;32mI (94) boot: Partition Table:[0m
[0;32mI (98) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (105) boot:  0 nvs              WiFi data        01 02 0000e000 00004000[0m
[0;32mI (113) boot:  1 otadata          OTA data         01 00 00012000 00002000[0m
00020000 00100000[0m
[0;32mI (135) boot:  4 ota_0            OTA app          00 10 00120000 00100000[0m
[0;32mI (159) esp_image: segment 0: paddr=00020020 vaddr=3c0a0020 size=3cb58h (248664) map[0m
98 size=00dd0h (  3536) load[0m
00 size=12d18h ( 77080) load[0m
[0;32mI (372) esp_image: segment 5: paddr=0010cbf8 vaddr=000000[0;32mI (376) secure_boot_v2: Verifying with RSA-PSS...[0m
[0;32mI (383) secure_boot_v2: Signature verified successfully![0;32mI (394) secure_boot_v2: enabling secure boot v2...[0m
[0;32mI (397) secure_boot_v2: secure boot v2 is already enabled, continuing..[0m
[0;32mI (449) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (449) cpu_start: Application information:[0m
[0;32mI (452) cpu_start: Project name:     simple_ota[0m
[0;32mI (457) cpu_start: App version:      v5.1-dev-3462-g045163a2ec-dirty[0m
[0;32mI (464) cpu_start: Compile time:     Mar 23 2023 21:19:19[0m
[0;32mI (470) cpu_start: ELF file SHA256:  83f8adcf9eb2f38b...[0m
[0;32mI (476) cpu_start: ESP-IDF:          v5.1-dev-3462-g045163a2ec-dirty[0m
[0;32mI (483) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (488) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (493) cpu_start: Chip rev:         v0.1[0m
[... 1077 lines cut ...]
[0;32mI (375) esp_image: Verifying image signature...[0m
[0;32mI (383) secure_boot_v2: Signature verified successfully![0m
[0;32mI (394) boot: Loaded app from partition at offset 0x20000[0m
[0;32mI (394) secure_boot_v2: enabling secure boot v2...[0m
[0;32mI (409) flash_encrypt: flash encryption is enabled (1 plaintext flashes left)[0m
[0;32mI (434) cpu_start: Pro cpu up.[0m
7546c[0m
[0;32mI (449) cpu_start: Pro cpu start user code[0m
[0;32mI (449) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (449) cpu_start: Application information:[0m
[0;32mI (452) cpu_start: Project name:     simple_ota[0m
[0;32mI (457) cpu_start: App version:      v5.1-dev-3462-g045163a2ec-dirty[0m
[0;32mI (464) cpu_start: Compile time:     Mar 23 2023 21:19:19[0m
[0;32mI (476) cpu_start: ESP-IDF:          v5.1-dev-3462-g045163a2ec-dirty[0m
[0;32mI (483) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (488) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (498) heap_init: Initializing. RAM available for dynami[0;32mI (511) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM[0m
[0;32mI (518) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (524) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM[0m
[0;32mI (531) spi_flash: detected chip: gd[0m
[0;32mI (566) sleep: Enable automatic switching of GPIO sleep configuration[0m
[0;32mI (573) app_start: Starting scheduler on CPU0[0m
[0;32mI (578) app_start: Starting scheduler on CPU1[0m
[0;32mI (578) main_task: Started on CPU0[0m
[0;32mI (588) main_task: Calling

DerPicknicker, 2023-03-23:
Here is my E-Fuse Summary: 
etecting chip type... ESP32-S3

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                           ??? Temperature calibration                        = -10.600000000000001 R/W (0b101101010)
ADC1_MODE0_D2 (BLOCK2)                             ??? ADC1 calibration 1                             = -432 R/W (0xec)
ADC1_MODE1_D2 (BLOCK2)                             ??? ADC1 calibration 2                             = 188 R/W (0x2f)
ADC1_MODE2_D2 (BLOCK2)                             ??? ADC1 calibration 3                             = -368 R/W (0xdc)
ADC1_MODE3_D2 (BLOCK2)                             ??? ADC1 calibration 4                             = -232 R/W (0xba)
ADC2_MODE0_D2 (BLOCK2)                             ??? ADC2 calibration 5                             = -312 R/W (0xce)
ADC2_MODE1_D2 (BLOCK2)                             ??? ADC2 calibration 6                             = 64 R/W (0x10)
ADC2_MODE2_D2 (BLOCK2)                             ??? ADC2 calibration 7                             = -328 R/W (0xd2)
ADC2_MODE3_D2 (BLOCK2)                             ??? ADC2 calibration 8                             = -256 R/W (0xc0)
ADC1_MODE0_D1 (BLOCK2)                             ??? ADC1 calibration 9                             = 0 R/W (0b100000)
ADC1_MODE1_D1 (BLOCK2)                             ??? ADC1 calibration 10                            = -12 R/W (0b100011)
ADC1_MODE2_D1 (BLOCK2)                             ??? ADC1 calibration 11                            = 44 R/W (0b001011)
ADC1_MODE3_D1 (BLOCK2)                             ??? ADC1 calibration 12                            = -92 R/W (0b110111)
ADC2_MODE0_D1 (BLOCK2)                             ??? ADC2 calibration 13                            = -80 R/W (0b110100)
ADC2_MODE1_D1 (BLOCK2)                             ??? ADC2 calibration 14                            = -44 R/W (0b101011)
ADC2_MODE2_D1 (BLOCK2)                             ??? ADC2 calibration 15                            = -96 R/W (0b111000)
ADC2_MODE3_D1 (BLOCK2)                             ??? ADC2 calibration 16                            = 68 R/W (0b010001)

Config fuses:
DIS_ICACHE (BLOCK0)                                Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                                Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)                       Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)                       Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)                        Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                                   Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_APP_CPU (BLOCK0)                               Disables APP CPU                                   = False R/W (0b0)
FLASH_TPUW (BLOCK0)                                Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                                    unit is (ms/2). When the value is 15, delay is 7.
                                                   5 ms                                              
DIS_DIRECT_BOOT (BLOCK0)                           Disables direct boot mode                          = True R/W (0b1)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             Disables USB-Serial-JTAG ROM printing              = False R/W (0b0)
FLASH_ECC_MODE (BLOCK0)                            Configures the ECC mode for SPI flash             
   = 16-byte to 18-byte mode R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disables USB-Serial-JTAG download feature in UART  = False R/W (0b0)
                                                   download boot mode                                
UART_PRINT_CONTROL (BLOCK0)                        Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                                Selects SPI flash type                             = 4 data lines R/W (0b0)
FLASH_PAGE_SIZE (BLOCK0)                           Sets the size of flash page                        = 0 R/W (0b00)
FLASH_ECC_EN (BLOCK0)                              Enables ECC in Flash boot mode                     = False R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)                         Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                                   d during SPI boot                                 
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Disables USB-OTG download feature in UART download = False R/W (0b0)
                                                    boot mode                                        
DISABLE_WAFER_VERSION_MAJOR (BLOCK0)               Disables check of wafer version major              = False R/W (0b0)
DISABLE_BLK_VERSION_MAJOR (BLOCK0)                 Disables check of blk version major                = False R/W (0b0)
BLOCK_USR_DATA (BLOCK3)                            User data                                         
[... cut ...]
VDD_SPI_FORCE (BLOCK0)                             Force using VDD_SPI_XPD and VDD_SPI_TIEH to config = True R/W (0b1)
                                                   ure VDD_SPI LDO                                   
PIN_POWER_SELECTION (BLOCK0)                       Sets default power supply for GPIO33..37           = VDD_SPI R/W (0b1)

Wdt Config fuses:
WDT_DELAY_SEL (BLOCK0)                             Selects RTC WDT timeout threshold at startup       = 0 R/W (0b00)

Flash voltage (VDD_SPI) set to 3.3V by efuse.

DerPicknicker, 2023-03-23:
EDIT: I think now it works FINE.. NO HELP NEEDED AT THIS STAGE. I forgot to enable the ENCRYPTED NVS Partition in the MENUCONFIG. Thank you for YOUR SUPPORT.

mahavirj, 2023-03-27:
@DerPicknicker Is there anything else that we can help you with? or we can close the issue for now?

DerPicknicker, 2023-03-27:
Hi, yes everything works now. Thank you for your help

Alvin1Zhang, 2023-03-27:
Thanks for reporting and sharing the updates, feel free to reopen.

closed 2023-03-27 by Alvin1Zhang

hdpklm, 2023-04-17:
Hi @DerPicknicker 

Can you list step by step how to do the encryption?
and what port did you use to program it, usb or uart?

thank you.

DerPicknicker, 2023-04-17:
Which board and where do you stuck?

hdpklm, 2023-04-18:
hi @DerPicknicker 
ESP32-S3

I receiver the same error like this  
```sh
A fatal error occurred: Invalid head of packet (0x45): Possible serial noise or corruption.
```

I thing is because I enabled the SECURE_BOOT_EN.
so I want to know step by step how to do encryption. the document its so confuse

thank you

DerPicknicker, 2023-04-18:
This is not caused by securit features. You need to press the boot and reset btn at the same time to enable the download mode (all LEDs should not blink anymore). 
You need to be more specific if you need help. What did you tried, which comment did you use.

hdpklm, 2023-04-18:
yes I press que boot and reset, and I sure its in upload mode.
but its the first time I encrypt the esp32-s3

i use this commands
```sh
set PORT=COM24

espsecure generate_flash_encryption_key key.bin
espefuse --chip esp32s3 --port %PORT% burn_key BLOCK_KEY0 key.bin XTS_AES_128_KEY
espsecure encrypt_flash_data --keyfile ./key.bin --address 0x10000 --output app.bin ./build/main.ino.bin
espefuse --chip esp32s3 --port %PORT% burn_efuse SPI_BOOT_CRYPT_CNT 1
espefuse --chip esp32s3 --port %PORT% burn_efuse SECURE_BOOT_EN 1
esptool.exe --chip esp32-s3 --port COM24 --baud 921600 write_flash 0x1000 ./app.bin
```

please pass me the commands you used to encrypt the ESP32-S3

DerPicknicker, 2023-04-18:
And what’s the output?

hdpklm, 2023-04-18:
this is the output for  `esptool.exe --chip esp32-s3 --port COM24 --baud 921600 write_flash 0x1000 ./app.bin`

```sh
C:\hassan\tools\esp32-encrypt>esptool.exe --chip esp32-s3 --port COM24 --baud 921600 write_flash 0x1000 ./app.bin
esptool.py v4.5.1
Serial port COM24
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:95:a9:78
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x45): Possible serial noise or corruption.
```

DerPicknicker, 2023-04-18:
Try a different cable. 

What do you want to achieve? Flach encryption + secure boot ?

hdpklm, 2023-04-18:
I thing the problem its not in the cable, I upload a lot of firmware with this cable without any problem.
I try to upload the firmware, and the encrypted firmware, and the bootloader. bot nothing work

DerPicknicker, 2023-04-18:
It might happen that you broke your board… 

In which order did you program your board? 
1. you have to create a key on your pc for signing your firmware 
2. Then you can program that key to your board 
3. Now you need to enable secure boot inside the menuconfig (allow insecure option if not you’re unable to flash it via usb)
4. Then you’re able to build the bootloader (idf.py bootloader)
5. DONT FLASH IT 
6. Enable flash encryption and update the partition details to enable the encryption to all partitions!!
7. Then you’re able to build the project and flash it with idf.py flash

hdpklm, 2023-04-18:
ok thank you. your steps are work properly

this is all the code I use

```sh
# create project
idf.py create-project test-encrypt-1
cd test-encrypt-1

# configure project
set IDF_TARGET=esp32s3
idf.py set-target esp32s3
idf.py menuconfig

# changes in menuconfig
# 1. security features -> Enable flash encryption boot -> Enable usage node -> developer
# 2. security features -> Potentially insecure options -> Leave unused digest slots
# 3. partition table -> Custom partition table CSV -> partition_table.csv
# 4. partition table -> offset -> 0xD000

# secure boot
espsecure.py generate_signing_key --version 2 secure_boot_signing_key.pem
idf.py build
idf.py flash

# upload bootloader after uploading app and partition table
esptool --chip esp32s3 -p COM30 --no-stub write_flash 0x0 build/bootloader/bootloader.bin --force
```

what I have to do to lock the chip in deploy mode ??

DerPicknicker, 2023-04-18:
Google..? Disable all insecure options and change from development to production and do all steps again.