espressif/esp-idf #10936: GDBGUI hard requirement makes installation difficult

andy31415, 2023-03-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESPIDF requires gdbgui which in turn requires gevent. Apparently this dependency is not updated often for new systems and currently causes compile problems for projects such as https://github.com/project-chip/connectedhomeip/issues/25385 (TLDR: newer debian fails to install gdbgui). This happens at least on newer debian and arch as far as I can tell.

Overall, even if the dependency is updated, compiling gevent from source is slow . Part of the project chip CI we do a bootstrap for our compiler environment and having native dependencies such as this slows down CI.

We would like to have a way to not enforce all dependencies - in particular gdbgui is not needed at all for code compilation and it should not be a hard dependency. Please allow us to bypass it somehow (do not require it strictly or set some environment variable of IDF_COMPILE_ENV_ONLY or similar).

dobairoland, 2023-03-07:
@andy31415 We have already taken care of this. gdbgui is no longer a mandatory requirement since ESP-IDF v5.0. It can be installed as optional feature but it is not part of the default installation.

The linked issue is with Python 3.11 if I'm correct. We have also took care of it for ESP-IDF v4.4 (in the 4th bugfix revision) and earlier releases as well: https://github.com/espressif/esp-idf/blob/release/v4.4/requirements.txt#L22

Perhaps connectedhomeip needs to update to the latest bugfix version of ESP-IDF.

Am I missing something?

andy31415, 2023-03-07:
Great to head it is resolved. I would assume then project chip needs to update to the latest SDK - we will have to follow up on that. Thank you!