espressif/esp-idf #10939: SoftAP WPA3 crashes

0xFEEDC0DE64, 2023-03-07:
I am running my softAP with WPA3_PSK now and since then I experience crashes, as soon as I connect my phone to the ESP32.

```
W (354673) ESPCPPUTILS schedulertask.cpp:60 loop(): task delta_cloud hang for 103ms (heap8=59032) iterating: 83ms (max-key: etop 9ms) serializing: 1ms send: 16ms                                                                                                                                                  [289/6216]
W (427633) ESPCPPUTILS schedulertask.cpp:60 loop(): task delta_cloud hang for 103ms (heap8=58268) iterating: 94ms (max-key: tsks 7ms) serializing: 1ms send: 6ms
W (493533) wifi: [ieee80211_ht.c,2089] new:<13,0>, old:<13,0>, ap:<13,2>, sta:<13,0>, prof:1
W (493533) wifi: [wl_cnx.c,2721] station: 4a:26:c7:f6:75:39 join, AID=1, bgn, 20
W (494113) wifi: [ieee80211_ht.c,2089] new:<13,0>, old:<13,0>, ap:<13,2>, sta:<13,0>, prof:1
W (494113) wifi: [wl_cnx.c,2721] station: 4a:26:c7:f6:75:39 join, AID=1, bgn, 20
W (494123) wifi: [wl_cnx.c,2539] station: 4a:26:c7:f6:75:39 leave, AID = 1, bss_flags is 8391746, bss:0x3fff4a94
W (494123) wifi: [ieee80211_ht.c,2089] new:<13,0>, old:<13,0>, ap:<13,2>, sta:<13,0>, prof:1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008a4dd  PS      : 0x00060030  A0      : 0x800859a4  A1      : 0x3ffd0070
0x4008a4dd: xQueueTakeMutexRecursive at esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:727

A2      : 0xfefefefe  A3      : 0xffffffff  A4      : 0x3ffd0050  A5      : 0x0000000c
A6      : 0x3ffb5674  A7      : 0x3f405691  A8      : 0x8027a604  A9      : 0x3ffcfcc0
A10     : 0x0000005d  A11     : 0xffffffff  A12     : 0x0000005d  A13     : 0x3ffcfed0
A14     : 0x3ffcfcd0  A15     : 0x0000000d  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0xfefeff06  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

Backtrace: 0x4008a4da:0x3ffd0070 0x400859a1:0x3ffd0090 0x4023f617:0x3ffd00b0 0x4022cbf5:0x3ffd00d0 0x4022d70f:0x3ffd0100 0x4021dcb9:0x3ffd0160 0x4021a9d8:0x3ffd0190 0x4021df82:0x3ffd01c0 0x4022937f:0x3ffd01f0 0x4023ea15:0x3ffd0210 0x4023f882:0x3ffd0230 0x4023fcad:0x3ffd0260 0x4023fd7a:0x3ffd0280 0x40239661:0x3ffd02b
0 0x4022d9dc:0x3ffd0300 0x40216711:0x3ffd0380 0x40216eed:0x3ffd0490 0x40214da7:0x3ffd04f0 0x40095995:0x3ffd0510 0x40093630:0x3ffd0530 0x4008c5dd:0x3ffd0560
0x4008a4da: xQueueTakeMutexRecursive at esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:720 (discriminator 1)
0x400859a1: mutex_lock_wrapper at esp-idf/components/esp_wifi/esp32/esp_adapter.c:247
0x4023f617: wpa_ap_remove at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2592
0x4022cbf5: cnx_node_remove at ??:?
0x4022d70f: cnx_node_leave at ??:?
0x4021dcb9: wifi_ap_remove_sta_node_process at ??:?
0x4021a9d8: ieee80211_ioctl_process at ??:?
0x4021df82: ieee80211_ioctl at ??:?
0x4022937f: esp_wifi_ap_deauth_internal at ??:?
0x4023ea15: wpa_sta_disconnect at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:223
 (inlined by) sm_WPA_PTK_DISCONNECT_Enter at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:1357
0x4023f882: sm_WPA_PTK_Step at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2014
 (inlined by) wpa_sm_step at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2474
0x4023fcad: wpa_auth_sta_associated at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:461
 (inlined by) wpa_auth_sta_associated at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:433
0x4023fd7a: wpa_ap_join at esp-idf/components/wpa_supplicant/src/ap/wpa_auth.c:2559
0x40239661: hostap_sta_join at esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpa_main.c:302
0x4022d9dc: cnx_node_join at ??:?
0x40216711: hostap_recv_mgmt at ??:?
0x40216eed: hostap_input at ??:?
0x40214da7: ap_rx_cb at ??:?
0x40095995: ppRxPkt at ??:?
0x40093630: ppTask at ??:?
0x4008c5dd: vPortTaskWrapper at esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: bc5211708470fd69
```

0xFEEDC0DE64, 2023-03-07:
esp-idf is newest master version

0xFEEDC0DE64, 2023-03-07:
Also it doesnt crash every time, only sometimes. So WPA3 is definately working

kapilkedawat, 2023-04-21:
please try with latest master. This is fixed in latest master.

0xFEEDC0DE64, 2023-04-27:
Verified, works now flawless, I can start and stop the AP, change the way it's authentication works and it never crashed again. Thanks!

0xFEEDC0DE64, 2023-04-27:
Out of curiousity, what did you have to change on the closed source blob to fix this issue? was there some nullptr dereference issue?

closed 2023-04-27 by 0xFEEDC0DE64

kapilkedawat, 2023-04-28:
This issue was fixed in https://github.com/espressif/esp-idf/commit/c797146f43e54b1ae01ce340fbdb02d5399bc1c9 . After that also we had done some more fixes related to softAP here https://github.com/espressif/esp-idf/commit/888b909e793517157d98fd8e5d3a335cf24c979c as well as some PMF optimizations in closed source libs.