espressif/esp-idf #10943: gpio_install_isr_service generates wdt on v4.4.4

william-ferguson-au, 2023-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-148-g4c2afac355

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32UE

### Power Supply used.

Battery

### What is the expected behavior?

I expect to gpio_install_isr_service() to return without tripping the watch dog timer.

### What is the actual behavior?

When it reaches the call to gpio_install_isr_service() it crashes.

### Steps to reproduce.

1. Build, Flash, Monitor
2. Watch crash, reboot, crash ..

```
#include <esp_log.h>
#include <esp_intr_alloc.h>
#include <driver/gpio.h>

extern "C" {
void app_main();
}

static const char *TAG = "main";

void app_main(void) {
    ESP_LOGI(TAG, "Logger initialised");

    // Install the ISR service so that we can set up individual ISRs for each pin using gpio_isr_handler_add().
    ESP_ERROR_CHECK(gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1));
    ESP_LOGI(TAG, "Installed the ISR Service ...");

    .. more code that is never reached
}
```

### Debug Logs.

```plain
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14944
load:0x40080400,len:3816
0x40080400: _init at ??:?

entry 0x40080698
I (27) boot: ESP-IDF v4.4.4-148-g4c2afac355 2nd stage bootloader
I (27) boot: compile time 15:40:32
I (27) boot: chip revision: v3.0
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0

...

I (1486) main: Logger initialised
V (1504) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1510) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40082af2  PS      : 0x00050034  A0      : 0x3ffaf510  A1      : 0x3ffb0bb0
0x40082af2: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

A2      : 0x00002040  A3      : 0x80089005  A4      : 0x000637ff  A5      : 0x3ffb0b80
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffb7cd0  A9      : 0x40086026
0x40086026: _frxt_int_enter at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:119

A10     : 0x00000000  A11     : 0x3ffb7d0c  A12     : 0x3ffb7de4  A13     : 0x00000000
A14     : 0x00000010  A15     : 0x0000000c  SAR     : 0x0000001c  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x400d3fcf  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082af2
0x400d3fcf: uart_hal_write_txfifo at C:/Dev/espressif/esp-idf-v4.4/components/hal/uart_hal_iram.c:35

0x40082af2: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

Backtrace: 0x40082aef:0x3ffb0bb0 0x3ffaf50d:0x3ffaf5d0 |<-CORRUPTED
0x40082aef: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

Core  1 register dump:
PC      : 0x40134cc6  PS      : 0x00060334  A0      : 0x800d3b8e  A1      : 0x3ffb7a70
0x40134cc6: esp_pm_impl_waiti at C:/Dev/espressif/esp-idf-v4.4/components/esp_pm/pm_impl.c:832

A2      : 0x00000000  A3      : 0x10000000  A4      : 0x80089005  A5      : 0x3ffafc80
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x800edf1a  A9      : 0x3ffb7a40
A10     : 0x00000000  A11     : 0x0b1845f4  A12     : 0x00000019  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffe7cc4  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40134cc3:0x3ffb7a70 0x400d3b8b:0x3ffb7a90 0x4008764d:0x3ffb7ab0 0x40088d51:0x3ffb7ad0
0x40134cc3: cpu_ll_waiti at C:/Dev/espressif/esp-idf-v4.4/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Dev/espressif/esp-idf-v4.4/components/esp_pm/pm_impl.c:830

0x400d3b8b: esp_vApplicationIdleHook at C:/Dev/espressif/esp-idf-v4.4/components/esp_system/freertos_hooks.c:63

0x4008764d: prvIdleTask at C:/Dev/espressif/esp-idf-v4.4/components/freertos/tasks.c:3987 (discriminator 1)

0x40088d51: vPortTaskWrapper at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:142
```

### More Information.

If I comment out all the lines after gpio_install_isr_service() then it doesn't crash.
How can code that is never executed affect the behaviour of gpio_install_isr_service()?

william-ferguson-au, 2023-03-08:
If I add some logging into gpio_install_isr_service and gpio_isr_register then it still fails but you actually get a stacktrace into the failure. Looks like it is getting stuck on vPortExitCritical

```
I (1485) maxima_main: Logger initialised
I (1490) gpio_install_isr_service: Allocated space for the gpio_isr_func_t.
I (1497) gpio_isr_register: About to get isr_core_id
I (1503) gpio_isr_register: Got isr_core_id
V (1508) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1514) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x2
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40082b06  PS      : 0x00050034  A0      : 0x4000bff0  A1      : 0x3ffb0bb0
0x40082b06: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

A2      : 0x00002040  A3      : 0x3ffaf5d0  A4      : 0xc0000000  A5      : 0x3ffb0b80
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x3ffb7cd0  A9      : 0x40086026
0x40086026: _frxt_int_enter at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:119

A10     : 0x00000000  A11     : 0x3ffb7d0c  A12     : 0x3ffb7de4  A13     : 0x00000000
A14     : 0x00000010  A15     : 0x0000000c  SAR     : 0x0000001c  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core  0 was running in ISR context:
EPC1    : 0x400d3fa7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082b06
0x400d3fa7: uart_hal_write_txfifo at C:/Dev/espressif/esp-idf-v4.4/components/hal/uart_hal_iram.c:35

0x40082b06: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

Backtrace: 0x40082b03:0x3ffb0bb0 0x4000bfed:0x3ffaf5d0 0x40089002:0x3ffaf5e0 0x400d708f:0x3ffaf600 0x400d710d:0x3ffaf640 0x400e66a5:0x3ffaf670 0x400830cd:0x3ffaf690 0x40088d51:0x3ffaf6b0
0x40082b03: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x40089002: vPortClearInterruptMaskFromISR at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:332

0x400d708f: esp_intr_alloc_intrstatus at C:/Dev/espressif/esp-idf-v4.4/components/esp_hw_support/intr_alloc.c:584

0x400d710d: esp_intr_alloc at C:/Dev/espressif/esp-idf-v4.4/components/esp_hw_support/intr_alloc.c:604

0x400e66a5: gpio_isr_register_on_core_static at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:518

0x400830cd: ipc_task at C:/Dev/espressif/esp-idf-v4.4/components/esp_ipc/src/esp_ipc.c:78

0x40088d51: vPortTaskWrapper at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:142

Core  1 register dump:
PC      : 0x40134aae  PS      : 0x00060334  A0      : 0x800d3b66  A1      : 0x3ffb7a70
0x40134aae: esp_pm_impl_waiti at C:/Dev/espressif/esp-idf-v4.4/components/esp_pm/pm_impl.c:832

A2      : 0x00000000  A3      : 0x10000000  A4      : 0x80089005  A5      : 0x3ffafc80
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x800edd0e  A9      : 0x3ffb7a40
A10     : 0x00000000  A11     : 0x0b180f6b  A12     : 0x00000019  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffe7cc4  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40134aab:0x3ffb7a70 0x400d3b63:0x3ffb7a90 0x4008764d:0x3ffb7ab0 0x40088d51:0x3ffb7ad0
0x40134aab: cpu_ll_waiti at C:/Dev/espressif/esp-idf-v4.4/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Dev/espressif/esp-idf-v4.4/components/esp_pm/pm_impl.c:830

0x400d3b63: esp_vApplicationIdleHook at C:/Dev/espressif/esp-idf-v4.4/components/esp_system/freertos_hooks.c:63

0x4008764d: prvIdleTask at C:/Dev/espressif/esp-idf-v4.4/components/freertos/tasks.c:3987 (discriminator 1)

0x40088d51: vPortTaskWrapper at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:142

```

igrr, 2023-03-08:
The function where the interrupt watchdog happens is `_xt_lowint1`, which is the interrupt vector for level 1 interrupts. Usually this means that there is an ISR which repeatedly enters, exits, and enters again. The most common cause is that the ISR doesn't clear the interrupt status bit, so the ISR keeps getting called.

This could be a bug in the GPIO interrupt handler. We would need some extra information to reproduce this: aside from gpio_install_isr_service, are you also calling gpio_isr_handler_add? Are you configuring GPIOs to enable interrupts before or after the call to gpio_install_isr_service? Could you try to reduce your program to a minimal example which reproduces the issue and which you can share with us?

william-ferguson-au, 2023-03-08:
In that last log, I haven't had a chance to set any ISRs as it hasn't yet returned from gpio_install_isr_service (note the absence of any "Installed the ISR Service ..." log.

I am calling gpio_isr_handler_add, but a long time after gpio_install_isr_service, so as you can see gpio_isr_handler_add doesn't have a chance to get called.

All calls to configure GPIOs are also well after gpio_install_isr_service and hence are not getting called.

Note also that the failure in the log above occurs when

1. I am logging at verbose. Changing to debug and gpio_install_isr_service succeeds (but gpio_isr_handler_add fails with a WDT later).
2. If don't comment out the code after gpio_install_isr_service. If I do comment out the code AFTER gpio_install_isr_service, then gpio_install_isr_service succeeds even though that code is never invoked when gpio_install_isr_service fails.

This change of behaviour depending upon whether some code is commented in/out, even though it is never invoked makes providing you with a simple program showing failure extremely challenging. 

Why would gpio_install_isr_service succeed just because code after it has been commented out?

william-ferguson-au, 2023-03-08:
If I switch to DEBUG (instead of VERBOSE), then gpio_install_isr_service succeeds, but gpio_isr_handler_add fails later on with WDT.

Code flow is:
- gpio_install_isr_service 
- configure gpio pins
- gpio_isr_handler_add 

```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x400832e4  PS      : 0x00050034  A0      : 0x40082b24  A1      : 0x3ffb0b80
0x400832e4: gpio_isr_loop at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:412
 (inlined by) gpio_intr_service at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:433

0x40082b24: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

A2      : 0x00000010  A3      : 0x00000010  A4      : 0x3ff44090  A5      : 0x40085ff6
0x40085ff6: _frxt_int_enter at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/portasm.S:119

A6      : 0x00000280  A7      : 0x00013ffc  A8      : 0x00000004  A9      : 0x00000001
A10     : 0x00000000  A11     : 0x3ffb6950  A12     : 0x80088fd5  A13     : 0x3ffb6a40
A14     : 0x00000003  A15     : 0x00060e23  SAR     : 0x0000001c  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb
Core  0 was running in ISR context:
EPC1    : 0x400d3e67  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400832e4
0x400d3e67: uart_hal_write_txfifo at C:/Dev/espressif/esp-idf-v4.4/components/hal/uart_hal_iram.c:35

0x400832e4: gpio_isr_loop at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:412
 (inlined by) gpio_intr_service at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:433

Backtrace: 0x400832e1:0x3ffb0b80 0x40082b21:0x3ffb0bb0 0x4000bfed:0x3ffb6aa0 0x40088fd2:0x3ffb6ab0 0x400e6da6:0x3ffb6ad0 0x400d816e:0x3ffb6b00 0x400d76bf:0x3ffb6b20 0x40136bbc:0x3ffb6b40 0x40088d21:0x3ffb6b60
0x400832e1: gpio_isr_loop at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:412
 (inlined by) gpio_intr_service at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:433

0x40082b21: _xt_lowint1 at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/xtensa_vectors.S:1114

0x40088fd2: vPortClearInterruptMaskFromISR at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:332

0x400e6da6: gpio_isr_handler_add at C:/Dev/espressif/esp-idf-v4.4/components/driver/gpio.c:475

0x400d816e: init_display() at C:/Source/wylas/maxima/main/src/maxima/display_ra8875.cpp:697

0x400d76bf: app_main at C:/Source/wylas/maxima/main/src/main.cpp:120 (discriminator 11)

0x40136bbc: main_task at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/port_common.c:141 (discriminator 2)

0x40088d21: vPortTaskWrapper at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:142

Core  1 register dump:
PC      : 0x401343e6  PS      : 0x00060734  A0      : 0x800d3a26  A1      : 0x3ffb79f0
0x401343e6: esp_pm_impl_waiti at C:/Dev/espressif/esp-idf-v4.4/components/esp_pm/pm_impl.c:832

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x80088fd5  A5      : 0x3ffb7920
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800ed846  A9      : 0x3ffb79c0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x80088fd5  A13     : 0x3ffb7900
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x401343e3:0x3ffb79f0 0x400d3a23:0x3ffb7a10 0x4008761d:0x3ffb7a30 0x40088d21:0x3ffb7a50
0x401343e3: cpu_ll_waiti at C:/Dev/espressif/esp-idf-v4.4/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Dev/espressif/esp-idf-v4.4/components/esp_pm/pm_impl.c:830

0x400d3a23: esp_vApplicationIdleHook at C:/Dev/espressif/esp-idf-v4.4/components/esp_system/freertos_hooks.c:63

0x4008761d: prvIdleTask at C:/Dev/espressif/esp-idf-v4.4/components/freertos/tasks.c:3987 (discriminator 1)

0x40088d21: vPortTaskWrapper at C:/Dev/espressif/esp-idf-v4.4/components/freertos/port/xtensa/port.c:142

```

william-ferguson-au, 2023-03-09:
OK, more data.

I have the firmware configured to reboot on crash, so I missed the fact that the original error and crash is caused by a WDT on gpio_isr_handler_add, and when it reboots we get a WDT on gpio_install_isr_service which then reboots getting further WDTs on  gpio_install_isr_service.

So I'll investigate what is causing the WDT on the gpio_isr_handler_add.

But why would gpio_install_isr_service generates a WDT on rebooot?
Shouldn't everything have been cleared?
How is the reboot different from the initial startup?

william-ferguson-au, 2023-03-09:
OK, so the ISR was tripping constantly (becauuse it had ben configured with the wrong gpio_int_type_t.

@igrr I'm still curious about why gpio_isr_install_service  would WDT on the reboot after the crash.

KaeLL, 2023-03-09:
@william-ferguson-au Would you reconsider reopening the issue, and provide us with a reproducible example? If confirmed, this issue is too severe to risk leaving it unattended.

william-ferguson-au, 2023-03-09:
Here is a simple reproduction.

```
#include <esp_log.h>
#include <esp_intr_alloc.h>
#include <driver/gpio.h>

extern "C" {
    void app_main();
}

static const uint8_t PIN_HIGH = 1;

// Screen
const gpio_num_t PIN_SCREEN_POWER       = GPIO_NUM_17;
const gpio_num_t PIN_SCREEN_INT         = GPIO_NUM_4;

static const char *TAG = "main";

void IRAM_ATTR screen_pressed(void*) {
}

void app_main(void) {
    ESP_LOGI(TAG, "Logger initialised");

    ESP_ERROR_CHECK(gpio_install_isr_service(ESP_INTR_FLAG_LEVEL1));
    ESP_LOGI(TAG, "Installed the ISR Service ...");

    // Turn the screen on which will drive PIN_SCREEN_INT HIGH
    ESP_ERROR_CHECK(gpio_set_direction(PIN_SCREEN_POWER, GPIO_MODE_OUTPUT));
    ESP_ERROR_CHECK(gpio_set_level(PIN_SCREEN_POWER, PIN_HIGH));

    ESP_ERROR_CHECK(gpio_set_direction(PIN_SCREEN_INT, GPIO_MODE_INPUT));
    ESP_ERROR_CHECK(gpio_set_pull_mode(PIN_SCREEN_INT, GPIO_PULLDOWN_ONLY));

    ESP_ERROR_CHECK(gpio_set_intr_type(PIN_SCREEN_INT, GPIO_INTR_HIGH_LEVEL));
    ESP_ERROR_CHECK(gpio_isr_handler_add(PIN_SCREEN_INT, screen_pressed, nullptr));

    ESP_LOGI(TAG, "Finished setup");
}
```

kriegste, 2023-03-09:
I have not tried your example, but what happens if you do
`gpio_install_isr_service(ESP_INTR_FLAG_IRAM);`
instead?

william-ferguson-au, 2023-03-10:
Exactly the same.

KaeLL, 2023-03-10:
@william-ferguson-au you're using a level-triggered interrupt which you're not disabling after the trigger on a GPIO already asserted. igrr was right.
Set it to low before enabling the interrupt and you should probably switch to edge-triggered interrupts (GPIO_INTR_POSEDGE instead of GPIO_INTR_HIGH_LEVEL).

william-ferguson-au, 2023-03-10:
Setting the PIN_SCREEN_INT to low before enabling the interrupt does not change the behaviour.

I'd love to use edge trigger interrupts but they don't trip in this scenario.

arjanmels, 2023-03-19:
@william-ferguson-au 
I had a similar problem wityh level interrupts (on esp-idf 4.4.4; I have not tested other versions).
I think I found the problem: a SW-RESET (esp_restart), does NOT reset the GPIO registers. Therefore the interrupt is still enabled when you call ```gpio_install_isr_service```. As it is a level interrupt it will start looping the interrupt handler (and the interrupt handler has not been installed yet to disable the interrupt).

I solved it by doing an explicit gpio_intr_disable before ```gpio_install_isr_service``` and only do a ```gpio_intr_enable``` after ```gpio_isr_handler_add```.

(You can check if the interrupt is enabled via e.g. ```    ESP_EARLY_LOGW("GPIO", "Pin Reg: %08X", REG_READ(GPIO_PIN17_REG));```)

@KaeLL 
1) I see from the documentation of the esp_restart function that this is intentional. However maybe good to call it out in the first paragraph on page https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html as well?
2) How to do a full reset including all peripherals from software?

william-ferguson-au, 2023-03-19:
Thanks @arjanmels that gels exactly with what I have been seeing.
Note to self: disable all level interrupts before calling `gpio_install_isr_service`.

Note to all: I would in general much prefer to use edge interrupts but with ESP32 I am finding that more often than not:
1. An edge interrupt doesn't trip
2. An edge interrupt trip multiple times for one transition from low to high (or high to low).

And I haven't been able to determine any pattern as to why an edge interrupt will work in one scenario but not in others. 
Hence the fallback to level interrupts.

kriegste, 2023-03-19:
That's because of the hardware bug in ESP32:
https://www.espressif.com/sites/default/files/documentation/esp32_errata_en.pdf
Section 3.14.
Use level interrupts instead.

william-ferguson-au, 2023-03-19:
@kriegste ??what?? oh wow - thanks. No wonder the edge interrupts have been driving me nuts.

KaeLL, 2023-03-20:
@arjanmels Good catch! And apparently [that's intended](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#_CPPv411esp_restartv).
@william-ferguson-au check out #9746

william-ferguson-au, 2023-03-23:
I understand the need for ISRs to remain across a restart to support wake interrupts.
But surely when you invoke ```gpio_install_isr_service``` the first thing it should do is to remove all existing interrupts (not potentially cause them to fire).

Having to manually disable all interrupts that will be added later prior to calling ```gpio_install_isr_service``` means that you have to be have full knowledge of all interrupts ahead of time. That makes it very tricky when you are using libraries that don't readily expose whether they will use an interrupt.

benhansen-io, 2025-01-27:
I just hit this. I was surprised to have to call gpio_reset_pin only after a SW reset but not other times. I didn't see it mentioned at https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/gpio.html  Ideally this should resolve itself when gpio_install_isr_service is called again.