espressif/esp-idf #10944: (HELP ME!(2)) TLS ON TCP SERVER,CLIENT

Inverter-Jo, 2023-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

DEVKIT : ESP32
SDK VERSION : 5.0.0

When I connected tls after tcp connection, I was able to check client hello and server hello through wireshark.
However, seeing that the above message is frequently invoked, I think there is a problem with my code.
In addition, the client side called the following error message.
esp-tls: couldn't get hostname for :: getaddrinfo() returns 202

There are two questions.

Is it right to connect tls after tcp connection?
How do I set the hostname of the esp_tls_conn_new_sync() function? Should I write the local netif hostname of the server if it is a client terminal?
Below is a summary of my codes.

client//

  sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
  if (sock < 0) {
      ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
      vTaskDelay(250);
  }
  struct esp_tls *tcp_tls = esp_tls_init();
  
  const char *host_name = "server";
  esp_tls_cfg_t tls_cfg = {
      .clientcert_buf = (const unsigned char *)test_cacert_pem,//servercert_start,
      .clientcert_bytes = strlen(test_cacert_pem) + 1,//servercert_end - servercert_start + 1
      .clientkey_buf = (const unsigned char *)test_public_pem,//prvtkey_pem_start,
      .clientkey_bytes = strlen(test_public_pem) + 1,//prvtkey_pem_start - prvtkey_pem_end + 1, 
  };		
  int err = connect(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
  if (err != 0) { 
      ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);            
      vTaskDelay(250);  
  }
  else {
      ESP_LOGI(TAG, "Successfully connected");	
      int tls_conn = esp_tls_conn_new_async(NULL,0,443,&tls_cfg, tcp_tls);
      if(tls_conn>0){
        esp_tls_conn_write(tcp_tls, wifi_disconnect_noti, UART_PACKET_SIZE);
        vTaskResume(tcp_handle);
      }
      ESP_LOG_BUFFER_CHAR(TAG,wifi_config.sta.ssid,32);         
      vTaskDelete(NULL);
  }
server//

  listen_sock = socket(addr_family, SOCK_STREAM, ip_protocol);
  ...
  int opt = 1;
  setsockopt(listen_sock, SOL_SOCKET, SO_REUSEADDR, &opt, sizeof(opt));//
  int err = bind(listen_sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
  ...
  ESP_LOGI(TAG, "Socket bound, port %d", PORT);	
  err = listen(listen_sock, 1);
  ...
  ...
  socklen_t addr_len = sizeof(source_addr);
  sock = accept(listen_sock, (struct sockaddr *)&source_addr, &addr_len);
  ....	
  setsockopt(sock, SOL_SOCKET, SO_KEEPALIVE, &keepAlive, sizeof(int));
  ...
  ...
  ...
  
  if (source_addr.ss_family == PF_INET) {
      inet_ntoa_r(((struct sockaddr_in *)&source_addr)->sin_addr, addr_str, sizeof(addr_str) - 1);
  }				
  struct esp_tls *temp_tcp_tls = esp_tls_init();				
  tcp_tls = temp_tcp_tls;
  esp_tls_cfg_server_t tls_cfg = {
      .servercert_buf = (const unsigned char *)test_cacert_pem,
      .servercert_bytes = strlen(test_cacert_pem) + 1,
      .serverkey_buf = (const unsigned char *)test_prvtkey_pem,
      .serverkey_bytes = strlen(test_prvtkey_pem) + 1,
  };
  esp_tls_server_session_create(&tls_cfg,sock, tcp_tls);

Inverter-Jo, 2023-03-09:
I checked, both server and client try handshake, but client hello and server hello are not running.
I think I forgot something about the ssl structure in the tls structure.

AdityaHPatwardhan, 2023-03-13:
Hi @Inverter-Jo Please refer to the tcp related examples at [this link](https://github.com/espressif/esp-idf/tree/47852846d3e89580ef4da28210b6ffc1cb5eaa9d/examples/protocols/sockets/).
Please follow the [README](https://github.com/espressif/esp-idf/tree/47852846d3/examples/protocols/sockets#bsd-socket-api-examples) to know more about the standard BSD socket examples. 
You may find appropriate documentation about https_server [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_https_server.html)

Closing this issue since it does not represent any bug/feature request in ESP-IDF.
Please feel free to re-open the issue if you still face a problem with any IDF implantation/documentation.

closed 2023-03-13 by AdityaHPatwardhan