espressif/esp-idf #10946: BLE server throughput example compilation error

kloovt, 2023-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.01

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

I expected a piece of example code with 0 alterations to build first time without issues. The client example builds without any issue.

### What is the actual behavior?

I got a couple of compilation errors all related to line 647 in example_ble_server_throughput.c
The errors complain that the Macro ESP_LOGI() from esp_log.h expects an int, but the argument provided is a uint32_t.

### Steps to reproduce.

1. install esp-idf on vscode
2. create project from throughput server example
3. try to build as-is

### Build or installation Logs.

```plain
C:/Users/jorri/esp/esp-idf/components/log/include/esp_log.h:265:27: error: format '%d' expects argument of type 'int', but argument 7 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=] 265 | #define LOG_COLOR(COLOR) "\033[0;" COLOR "m"
```

### More Information.

I fixed it by rewriting line 647 in example_ble_server_throughput.c 

existing:
            ESP_LOGI(GATTS_TAG, "GATTC write Bit rate = %d Byte/s, = %d bit/s, time = %ds",

fixed:
            ESP_LOGI(GATTS_TAG, "GATTC write Bit rate = %" PRIu32 " Byte/s, = %" PRIu32 " bit/s, time = %ds",

zhp0406, 2023-11-17:
Is this issue still present on the latest branch?

jorgie0, 2024-02-11:
This issue is still present in the following ESP IDF
`esp-idf-v5.1.2-4`
I am using the following configuration:
`Windows 11
VS Code ( ESP IDF )`

Code being used:
[https://github.com/ESP32Tutorials/esp32-esp-idf-mqtt-bme280](url)

Compilation output:
`-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Projects/esp32-esp-idf-mqtt-bme280-main/sdkconfig
Compiler supported targets: xtensa-esp32-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "BME280_MQTT" version: 1
-- Adding linker script C:/Projects/esp32-esp-idf-mqtt-bme280-main/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update bme280 bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/app_trace C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/app_update C:/Projects/esp32-esp-idf-mqtt-bme280-main/components/bme280 C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/bootloader C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/bt C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/cmock C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/console C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/cxx C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/driver C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/efuse C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp-tls C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_adc C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_coex C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_common C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_eth C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_event C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_hid C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_http_client C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_http_server C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_https_server C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_lcd C:/Espressif/frameworks/esp-idf-v5.1.2-4/components/esp_local_ctrl C:/Espressif/f [... cut ...]
C:/Projects/esp32-esp-idf-mqtt-bme280-main/main/main.c:205:9: warning: unused variable 'msg_id' [-Wunused-variable]
  205 |     int msg_id;
      |         ^~~~~~
C:/Projects/esp32-esp-idf-mqtt-bme280-main/main/main.c:204:30: warning: unused variable 'client' [-Wunused-variable]
  204 |     esp_mqtt_client_handle_t client = event->client;
      |                              ^~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja " terminated with exit code: 1. `

zhp0406, 2024-02-19:
@jorgie0 
Hi,thanks for report the issue, I build throughput_server example using v5.1.2,and build success:

~/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server/build
Executing "ninja all"...
[1/4] cd /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/b...hroughput/throughput_server/build/throughput_server_demo.binthroughput_server_demo.bin binary size 0xbfb70 bytes. Smallest app partition is 0x100000 bytes. 0x40490 bytes (25%) free.
[1/1] cd /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/b...throughput/throughput_server/build/bootloader/bootloader.binBootloader binary size 0x6820 bytes. 0x7e0 bytes (7%) free.

Project build complete. To flash, run this command:
/home/zhanghaipeng/.espressif/python_env/idf5.1_py3.11_env/bin/python ../../../../../../components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/throughput_server_demo.bin
or run 'idf.py -p (PORT) flash'
~/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_throughput/throughput_server$ git log
commit 482a8fb2d78e3b58eb21b26da8a5bedf90623213 (HEAD, tag: v5.1.2, origin/test_esp32c6/v5.1.2, origin/test/coex_test_v5.1.2_20231113_karl)
Author: Aditya Patwardhan <aditya.patwardhan@espressif.com>
Date:   Fri Nov 10 08:08:53 2023 +0550

    change(version): Update version to 5.1.2

esp-cjh, 2024-02-26:
@jorgie0 The compilation error you report was caused by the code of the esp32-esp-idf-mqtt-bme280 example. We found no compilation errors in the BLE server throughput example.

closed 2024-02-26 by Alvin1Zhang