espressif/esp-idf #10949: Esp32 failed to send one to many using espnow protocol

lebossejames, 2023-03-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I use Arduino IDE with ESP32-C3 (library 2.0.7) and use ESP-NOW but ESP failed to send one to many times using espnow protocol.
`13:09:30.201 -> Last Packet Send Status: Delivery Fail
13:09:31.795 -> isAfterBLE_FRONT
13:09:31.795 -> Ask to open all relay 2CH
13:09:32.825 -> Sent with success
13:09:32.825 ->
13:09:32.825 -> Last Packet Send Status: Delivery Fail
13:09:34.452 -> isAfterBLE_FRONT`

I try to set this on my code but doesn't resolved:
`WiFi.useStaticBuffers(true);
WiFi.mode(WIFI_STA);
`

lebossejames, 2023-09-09:
Hello,

News of this issue?

zhangyanjiaoesp, 2023-09-19:
@lebossejames 

Do you use BLE when using ESP-NOW? Can you provide a demo for us to reproduce the issue? Can't tell from your log what happened .

lebossejames, 2023-09-20:
Hello,

This is my project.
On line 57: String mode_selected=modes[0]; 

if you set modes[0]  Client  ; modes[1]  Server

On line 71 : broadcastServerESPNow : mac address of wifi Server
you can flash on your esp
[ManageBluetoothCameraByESP.zip](https://github.com/espressif/esp-idf/files/12675300/ManageBluetoothCameraByESP.zip)

zhangyanjiaoesp, 2023-09-22:
We will try to test it on our side.

zhangyanjiaoesp, 2023-09-22:
@lebossejames 
When I was compiling the project you provided, there were many errors.
Can you describe what exactly your project does?
1. What the BLE do?
2. What the ESP-NOW do?
3. Can you open the debug log when testing? And provide the complete logs for both devices.

lebossejames, 2023-09-30:
@zhangyanjiaoesp I have one code contains BLE and ESP-NOW functions, i use only ESP-NOW

Client ESP-NOW send message when the level of input is High and other message when it's low to Server ESP-NOW.

What do you mean of debug log? I use Arduino IDE.

lebossejames, 2023-10-14:
@zhangyanjiaoesp any news?

zhangyanjiaoesp, 2023-10-17:
@lebossejames 
In case of Arduino, to open the debug log, besides enabling the core level (set the `Tools -> Core debug level -> debug`), you also need to `Serial.setDebugOutput(true);`, where Serial can be any serial port or USB.

lebossejames, 2023-10-17:
[Client_traces.txt](https://github.com/espressif/esp-idf/files/12925831/Client_traces.txt)

zhangyanjiaoesp, 2023-10-17:
@lebossejames 
can't get more infomation from the log. Is it the log of the device that send ESP-NOW data? 
When I try to test your code locally, there is the compile error:

zhangyanjiaoesp, 2023-10-17:
```
ManageBluetoothCameraByESP/LCD_Manage.cpp:9:9: fatal error: LiquidCrystal_I2C.h: No such file or directory
  #include<LiquidCrystal_I2C.h>
          ^~~~~~~~~~~~~~~~~~~~~
 compilation terminated.

 exit status 1
 
Compilation error: LiquidCrystal_I2C.h: No such file or directory
```

lebossejames, 2023-10-17:
Are you install: 
https://gitlab.com/painlessMesh/painlessMesh 
and 
[image]

zhangyanjiaoesp, 2023-10-17:
Just installed `LiquidCrystal I2C` in Arduino IDE. How to install painlessMesh? Clone it and put it under the same folder as the [ManageBluetoothCameraByESP.zip](https://github.com/espressif/esp-idf/files/12675300/ManageBluetoothCameraByESP.zip) project?

zhangyanjiaoesp, 2023-10-17:
Installed `painlessMesh`:

[image]

now the compile error is:
```
In file included from /home/Arduino/libraries/Painless_Mesh/src/painlessMesh.h:10,
                 from /home//ManageBluetoothCameraByESP/ManageBluetoothCameraByESP.ino:26:
/home/Arduino/libraries/Painless_Mesh/src/painlessmesh/configuration.hpp:31:10: fatal error: AsyncTCP.h: No such file or directory
 #include <AsyncTCP.h>
          ^~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

zhangyanjiaoesp, 2023-10-17:
Installed `AsyncTCP`:
[image]

now the log is:
```
ARNING: library LiquidCrystal I2C claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
Sketch uses 1511540 bytes (115%) of program storage space. Maximum is 1310720 bytes.
Global variables use 64476 bytes (19%) of dynamic memory, leaving 263204 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board

Compilation error: text section exceeds available space in board
```

zhangyanjiaoesp, 2023-10-18:
> i copy theses files on Painless folder library : [AsyncTCP.zip](https://github.com/espressif/esp-idf/files/12930233/AsyncTCP.zip)
> 
> I don't install AsyncTCP library !! I use ESP32-C3 partitions schemes 1.9APP/190kb SPIFFS

OK, I will try again.

zhangyanjiaoesp, 2023-10-18:
> @zhangyanjiaoesp
> 
> Hello,
> 
> This is my project. On line 57: String mode_selected=modes[0];
> 
> if you set modes[0] Client ; modes[1] Server
> 
> On line 71 : broadcastServerESPNow : mac address of wifi Server you can flash on your esp [ManageBluetoothCameraByESP.zip](https://github.com/espressif/esp-idf/files/12675300/ManageBluetoothCameraByESP.zip)

Now the project can run on my side. I have some questions :
1. If the device want to send ESP-NOW, set `String mode_selected=modes[0]`, and if the device want to receive ESP-NOW set `String mode_selected=modes[1]`, right?
2. there is comment on line 70:

> // REPLACE WITH YOUR RECEIVER MAC Address Wifi
uint8_t broadcastServerESPNow[] = {0x7C,0xDF,0xA1,0xC2,0x50,0xEC}; 

So the `broadcastServerESPNow` is the mac address which will receive the ESP-NOW data?

3. How does your code implement one device sends packets to multiple devices?

lebossejames, 2023-10-18:
1) right
2) right
3) i implement code to send only to One device, because i have only 1 server

zhangyanjiaoesp, 2023-10-19:
There is crash on the receive device, I think that is why the ESP-NOW can't be received.
Do you see a similar phenomenon on your device? @lebossejames 

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x91c
load:0x403ce710,len:0x25b0
entry 0x403cc710
setup---begin
This chip has 1 cores
                     I2C: SDA 4 SCL 5
setup---middle1
68:B6:B3:A4:61:30
setup---end
setup---disableCore0WDT
Bytes received: 16
String: A1

Ask to close relay-2CH n�°1
[2023-10-19 10:30:15] Bytes received: 64
[2023-10-19 10:30:15] String: Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[2023-10-19 10:30:15] 
[2023-10-19 10:30:15] Core  0 register dump:
[2023-10-19 10:30:15] MEPC    : 0x42034378  RA      : 0x42028c80  SP      : 0x3fcad940  GP      : 0x3fc92400  
[2023-10-19 10:30:15] TP      : 0x3fc78a08  T0      : 0x420f37fc  T1      : 0x4201fe00  T2      : 0xc9000001  
[2023-10-19 10:30:15] S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x60000000  A1      : 0x004904de  
[2023-10-19 10:30:15] A2      : 0x00000066  A3      : 0x3fcad95c  A4      : 0x004904de  A5      : 0x00000000  
[2023-10-19 10:30:15] A6      : 0x00000000  A7      : 0x00000009  S2      : 0x6e001000  S3      : 0x3fc9c558  
[2023-10-19 10:30:15] S4      : 0x6e001000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x004904de  
[2023-10-19 10:30:15] S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000001  S11     : 0x3fcb40a4  
[2023-10-19 10:30:15] T3      : 0xdfffffff  T4      : 0xbfffffff  T5      : 0x80000000  T6      : 0x00000009  
[2023-10-19 10:30:15] MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x004904de  
[2023-10-19 10:30:15] MHARTID : 0x00000000  
[2023-10-19 10:30:15] 
[2023-10-19 10:30:15] Stack memory:
[2023-10-19 10:30:15] 3fcad940: 0x00000001 0x3fcad9a6 0x6e001000 0x004904de 0x00000000 0x00000000 0x00000008 0x00000066
[2023-10-19 10:30:15] 3fcad960: 0x6e001000 0x00000000 0x3fc9c558 0x42029c70 0x00000000 0x00000000 0x00000000 0xffffffff
[2023-10-19 10:30:15] 3fcad980: 0x3fcb4134 0x00000000 0x3fc998cc 0xffffffdd 0x3fcb414c 0x3fcb40a4 0x00000045 0x00000040
[2023-10-19 10:30:15] 3fcad9a0: 0x00000001 0x004904de 0x3fc91f20 0x42023a8e 0x00000001 0x3c131890 0x3fc91f20 0x6e001000
[2023-10-19 10:30:15] 3fcad9c0: 0x3fc9c000 0x3fc9747c 0x6e001000 0x4201fe0e 0x3fc9c000 0x3fc97000 0x3fc97000 0x4202079a
[2023-10-19 10:30:15] 3fcad9e0: 0x00000002 0x3fc97000 0x3fc97000 0x4200255e 0x3fcb37b4 0x00000000 0x4203ed4c 0x4003ee24
[2023-10-19 10:30:15] 3fcada00: 0x00000001 0x3fce0000 0x00000000 0xd2fc49ee 0x00000002 0x00000001 0xffffffc0 0x3fc998ac
[2023-10-19 10:30:15] 3fcada20: 0x3fcb4150 0x3fcb414c 0x3fc9a000 0x420dd68c 0x3fcb4134 0x00000000 0x8b5260fe 0xffff20c0
[2023-10-19 10:30:15] 3fcada40: 0x00000080 0x3fcb40a4 0x3fcb414c 0x3fcb37b4 0x3fcb4279 0x3fcb39ec 0x3fcb4134 0x420f397a
[2023-10-19 10:30:15] 3fcada60: 0x3fca17f0 0x00000003 0x4957454e 0x00000000 0x3fcdf934 0x3fce0000 0x3ff1b000 0xa4b3b668
[2023-10-19 10:30:15] 3fcada80: 0x3fc93061 0x00000003 0x3fc9ac9c 0x3fc9d000 0x00060000 0x00000000 0xc0380000 0x00000000
[2023-10-19 10:30:15] 3fcadaa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadae0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadb00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadb20: 0x3fcadb78 0x00000000 0x3fc998cc 0x00000018 0x00000149 0x00000000 0x3fcb37b4 0x3fc99bb8
[2023-10-19 10:30:15] 3fcadb40: 0x000000d0 0x3fcb39ec 0x00000000 0x421235be 0x00000001 0x3fce0000 0x3fce0000 0x3fcabb34
[2023-10-19 10:30:15] 3fcadb60: 0x00000011 0xffffffdd 0x00000000 0x3fcb413e 0x3fce0000 0x3fce0000 0x3fce0000 0x00000000
[2023-10-19 10:30:15] 3fcadb80: 0x3fcdf934 0x3fce0000 0x3ff1b000 0x3fce0000 0x3fce0000 0x3fce0000 0x000000f5 0x3fce0000
[2023-10-19 10:30:15] 3fcadba0: 0x3fcb4104 0x3fce0000 0x3fcb40a4 0x400415ba 0x3fce0000 0x3fce0000 0x3fce0000 0x3ff1b5a8
[2023-10-19 10:30:15] 3fcadbc0: 0x00000000 0x3fcdf918 0x3fce0000 0x40040902 0x00000000 0x00000000 0x00000011 0x00000000
[2023-10-19 10:30:15] 3fcadbe0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadc00: 0x00000000 0x00000000 0x00000000 0x4038d314 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadc20: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
[2023-10-19 10:30:15] 3fcadc40: 0x00000160 0xabba1234 0x00000154 0x3fcad8f0 0x00004702 0x3fc98f74 0x3fc98f74 0x3fcadc4c
[2023-10-19 10:30:15] 3fcadc60: 0x3fc98f6c 0x00000002 0x3fca86bc 0x3fca86bc 0x3fcadc4c 0x00000000 0x00000017 0x3fcac23c
[2023-10-19 10:30:15] 3fcadc80: 0x69666977 0x5a7acb00 0xf752ab23 0x000f4c9c 0x00000000 0x3fcadc30 0x00000017 0x00000002
[2023-10-19 10:30:15] 3fcadca0: 0x00000000 0x00000000 0x00000000 0x3fca1eb4 0x3fca1f1c 0x3fca1f84 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadcc0: 0x00000001 0x00000000 0x00000000 0x00000000 0x4210e7a4 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadce0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadd00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[2023-10-19 10:30:15] 3fcadd20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

lebossejames, 2023-10-19:
I have theses traces:

`07:29:50.166 -> !⸮⸮ ⸮⸮⸮ʣN⸮⸮⸮ESP-ROM:esp32c3-api1-20210207
07:29:50.190 -> Build:Feb  7 2021
07:29:50.190 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
07:29:50.190 -> SPIWP:0xee
07:29:50.190 -> mode:DIO, clock div:1
07:29:50.190 -> load:0x3fcd5810,len:0x438
07:29:50.190 -> load:0x403cc710,len:0x918
07:29:50.190 -> load:0x403ce710,len:0x25f4
07:29:50.190 -> entry 0x403cc710
07:29:50.456 -> setup---begin
07:29:50.456 -> This chip has 1 cores
07:29:50.456 -> I2C: SDA 4 SCL 5
07:29:50.954 -> setup---middle1
07:29:50.954 -> [   793][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
07:29:51.020 -> [   827][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
07:29:51.020 -> 7C:DF:A1:C2:50:EC
07:29:53.013 -> setup---end
07:29:53.013 -> setup---disableCore0WDT
07:29:53.411 -> [  3238][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:29:53.411 -> [  3238][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:29:53.445 -> [  3241][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
07:29:55.835 -> [  5659][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:29:55.835 -> [  5660][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:29:55.835 -> [  5663][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:29:58.262 -> [  8082][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:29:58.262 -> [  8083][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:29:58.262 -> [  8086][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:00.686 -> [ 10511][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:00.686 -> [ 10511][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:00.686 -> [ 10515][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:03.122 -> [ 12934][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:03.122 -> [ 12934][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:03.122 -> [ 12937][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:05.534 -> [ 15358][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:05.534 -> [ 15359][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:05.534 -> [ 15362][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:07.959 -> [ 17783][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:07.959 -> [ 17783][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:07.959 -> [ 17786][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:10.384 -> [ 20205][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:10.384 -> [ 20205][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:10.384 -> [ 20208][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:12.808 -> [ 22629][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:12.808 -> [ 22629][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:12.808 -> [ 22633][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:15.233 -> [ 25052][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:15.233 -> [ 25052][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:15.233 -> [ 25056][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:17.657 -> [ 27476][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:17.657 -> [ 27476][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:17.657 -> [ 27480][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:20.081 -> [ 29900][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:20.081 -> [ 29900][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:20.081 -> [ 29904][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:22.506 -> [ 32325][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:22.506 -> [ 32326][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:22.506 -> [ 32329][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:24.930 -> [ 34747][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:24.930 -> [ 34748][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:24.930 -> [ 34751][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:27.354 -> [ 37170][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:27.354 -> [ 37170][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:30:27.354 -> [ 37174][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:30:29.779 -> [ 39593][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:30:29.779 -> [ 39593][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
[... cut ...]
07:31:33.299 -> [103118][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:31:33.299 -> [103118][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:31:33.299 -> [103121][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
07:31:35.724 -> [105542][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
07:31:35.724 -> [105542][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND
07:31:35.724 -> [105545][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
`

no crash.

zhangyanjiaoesp, 2023-10-19:
@lebossejames 
After comment  line 391, the crash disappear.
```
Serial.println(myDataESPNow.msgNow);
```

And I have captured the packets between the two ESP32C3 device, there are too many RTS and probe request packets.

[image]

After comment line 1039~1042, the ESP-NOW send callback can return success.
```
  //if(mode_selected.equals(modes[1])){
  //  WiFi.begin(WIFI_SSID_JM, JM_PASSWORD);
  //  delay(1000);
  //}
```

[image]

So suggest you begin to send ESP-NOW packets after the STA connected.

zhangyanjiaoesp, 2023-10-19:
> I have theses traces @zhangyanjiaoesp :
> 
> `07:29:50.166 -> !⸮⸮ ⸮⸮⸮�ʣ�N⸮⸮⸮ESP-ROM:esp32c3-api1-20210207 07:29:50.190 -> Build:Feb 7 2021 07:29:50.190 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT) 07:29:50.190 -> SPIWP:0xee 07:29:50.190 -> mode:DIO, clock div:1 07:29:50.190 -> load:0x3fcd5810,len:0x438 07:29:50.190 -> load:0x403cc710,len:0x918 07:29:50.190 -> load:0x403ce710,len:0x25f4 07:29:50.190 -> entry 0x403cc710 07:29:50.456 -> setup---begin 07:29:50.456 -> This chip has 1 cores 07:29:50.456 -> I2C: SDA 4 SCL 5 07:29:50.954 -> setup---middle1 07:29:50.954 -> [ 793][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY 07:29:51.020 -> [ 827][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START 07:29:51.020 -> 7C:DF:A1:C2:50:EC 07:29:53.013 -> setup---end 07:29:53.013 -> setup---disableCore0WDT 07:29:53.411 -> [ 3238][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:29:53.411 -> [ 3238][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:29:53.445 -> [ 3241][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running 07:29:55.835 -> [ 5659][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:29:55.835 -> [ 5660][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:29:55.835 -> [ 5663][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:29:58.262 -> [ 8082][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:29:58.262 -> [ 8083][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:29:58.262 -> [ 8086][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:00.686 -> [ 10511][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:00.686 -> [ 10511][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:00.686 -> [ 10515][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:03.122 -> [ 12934][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:03.122 -> [ 12934][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:03.122 -> [ 12937][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:05.534 -> [ 15358][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:05.534 -> [ 15359][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:05.534 -> [ 15362][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:07.959 -> [ 17783][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:07.959 -> [ 17783][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:07.959 -> [ 17786][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:10.384 -> [ 20205][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:10.384 -> [ 20205][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:10.384 -> [ 20208][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:12.808 -> [ 22629][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:12.808 -> [ 22629][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:12.808 -> [ 22633][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:15.233 -> [ 25052][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:15.233 -> [ 25052][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:15.233 -> [ 25056][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:17.657 -> [ 27476][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:17.657 -> [ 27476][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:17.657 -> [ 27480][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:20.081 -> [ 29900][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:20.081 -> [ 29900][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:20.081 -> [ 29904][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:22.506 -> [ 32325][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:22.506 -> [ 32326][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:22.506 -> [ 32329][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:24.930 -> [ 34747][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:24.930 -> [ 34748][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:24.930 -> [ 34751][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:27.354 -> [ 37170][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:27.354 -> [ 37170][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:27.354 -> [ 37174][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:30:29.779 -> [ 39593][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:30:29.779 -> [ 39593][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:30:29.779 -> [ 39596][D][WiFiGeneric.cpp:1085] _eventCallbac [... cut ...] :1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:31:33.299 -> [103121][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running 07:31:35.724 -> [105542][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED 07:31:35.724 -> [105542][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 201 - NO_AP_FOUND 07:31:35.724 -> [105545][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running `
> 
> no crash.

This log show the device can't find the AP.

lebossejames, 2023-10-19:
@zhangyanjiaoesp So there is a problem or not?

How works if i want to use Client-Server with this code (create a personal network with this informations network)
`if(mode_selected.equals(modes[1])){
  //  WiFi.begin(WIFI_SSID_JM, JM_PASSWORD);
  //  delay(1000);
  //}`
?

zhangyanjiaoesp, 2023-10-20:
Have suggested in https://github.com/espressif/esp-idf/issues/10949#issuecomment-1770208930 .

> So suggest you begin to send ESP-NOW packets after the STA connected. And the AP's channel should be same to the ESP-NOW sender.

That means you can call 
```
if(mode_selected.equals(modes[1])){
    WiFi.begin(WIFI_SSID_JM, JM_PASSWORD);
    delay(1000);
}
```
And init ESP-NOW and send ESP-NOW data after you get the `WIFI_EVENT_STA_CONNECTED` event. 

However, if the STA and AP have a lot of data interaction, it will still affect the sending and receiving of ESP-NOW packets.

lebossejames, 2023-10-20:
@zhangyanjiaoesp  how can i use ESP-NOW with client-Server with informations network?

zhangyanjiaoesp, 2023-10-24:
> @zhangyanjiaoesp how can i use ESP-NOW and connect Wifi AP (Server ESP32) without mac adress server?

@lebossejames I'm sorry I don't see what you mean. Could you describe it in more detail？

Sherry616, 2024-03-04:
Hi @lebossejames, do you still have this problem? Could you share the latest update? Thanks.

Sherry616, 2024-04-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2024-04-23 by Sherry616